openrazer/openrazer #2058: Support for [Razer Ornata V3 X]

Evonder, 2023-05-02:
### Device Info

* Name: Razer Ornata V3 X
* Product Number: RZ03-04470200-R3U1
* Model Number: RZ03-0447

### Device Information
```
Bus 004 Device 004: ID 1532:02a2 Razer USA, Ltd Razer Ornata V3 X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02a2 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata V3 X
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
```

```
total 0
drwxr-xr-x 2 root root 220 May  1 20:36 .
drwxr-xr-x 4 root root 720 May  1 20:36 ..
lrwxrwxrwx 1 root root   9 May  1 20:36 usb-Razer_Razer_Basilisk_V3-event-mouse -> ../event8
lrwxrwxrwx 1 root root  10 May  1 20:36 usb-Razer_Razer_Basilisk_V3-if01-event-kbd -> ../event24
lrwxrwxrwx 1 root root  10 May  1 20:36 usb-Razer_Razer_Basilisk_V3-if02-event-kbd -> ../event28
lrwxrwxrwx 1 root root   9 May  1 20:36 usb-Razer_Razer_Basilisk_V3-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 May  1 20:36 usb-Razer_Razer_Ornata_V3_X-event-if01 -> ../event6
lrwxrwxrwx 1 root root   9 May  1 20:36 usb-Razer_Razer_Ornata_V3_X-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 May  1 20:36 usb-Razer_Razer_Ornata_V3_X-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 May  1 20:36 usb-Razer_Razer_Ornata_V3_X-if02-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 May  1 20:35 usb-Razer_Razer_Ornata_V3_X-if02-mouse -> ../mouse0
```

<details>

```
004:005:003:DESCRIPTOR         1682999460.439620
 05 0C 09 01 A1 01 06 00 FF 09 02 15 00 25 01 75
 08 95 5A B1 01 C0

004:005:002:DESCRIPTOR         1682999460.439779
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

004:005:001:DESCRIPTOR         1682999460.440191
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

004:005:000:DESCRIPTOR         1682999460.440657
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0

004:004:002:DESCRIPTOR         1682999460.441072
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

004:004:001:DESCRIPTOR         1682999460.441660
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

004:004:000:DESCRIPTOR         1682999460.442157
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
```

</details>

### Packet Captures

### Synapse Screenshots

I do not have a Windows environment to acquire the other outputs.

Thank you.

armakunizen, 2023-07-18:
It seems that there are at least two different versions of the Razer Ornata V3 X, looking at the Product IDs:

0x0294 (which is already in the code available) and 0x02A2 (currently unsupported)

I tried the code and simple replaced alle occurences of these IDs as far I could find. Compiling and installing went well, but after reboot I was unable to enter anything on the keyboard (mental notice here: always have a spare keyboard at hand ;-) )

Polychromatic always says "unsupported device", even after my changes above.
Maybe I missed some step and I'm quite unsure if these two versions of the Ornata V3 X are identical (despite the IDs) or there are small differences...

VlaK0r, 2023-08-08:
[image]
Hello!
Thank you for your work
Also does not see the keyboard, please fix it.

w4t3rf, 2023-08-09:
Hello, 

Same here with the same keyboard (Ornata V3 X)  running it from/for a Linux Mint environment. 

Running the Openrazer troubleshooter, I got 2 failed results from the Kernel module(1) and the device compatibility check(2) ( giving 1532:02A2 as an Idproduct while it should be 1532:0294) 

I was asked to run "sudo modprobe razerkbd" : it gave me no result(null) from the terminal.

Is there a command I missed out ? I can provide more details to help to solve the issue there.

Thanks for your help, support and dedicated work, all the best.

dmo9, 2023-08-18:
I am also suffering (greatly) with incompatibility issues pm my Ornata V3X with ID 1532:02a2

`
Bus 003 Device 006: ID 1532:02a2 Razer USA, Ltd Razer Ornata V3 X
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02a2 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Ornata V3 X
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1`

ShinkarenkoMaxim, 2023-08-25:
Hello,

I faced the same issue. I bought it now and was surprised that the driver is installed and the software does not recognize it. Polychromatic gives an error with device 1532:02A2 as unsupported device. 

How do I set the keyboard to static mode? Linux Mint 21

phoerselmann, 2023-08-28:
Maybe my changes in the PR  (#2119 ) will help. The Ornata V3 X with the PID 02A2 is now recognized without errors and can finally be used in the same way as the Ornata V3 X with the PID 0294.

z3ntu, 2023-09-02:
Fixed by #2119

closed 2023-09-02 by z3ntu

Natedogwoooof, 2023-10-27:
not working

Natedogwoooof, 2023-10-27:
fix please

z3ntu, 2023-10-27:
@Natedogwoooof Thank you for your useful comment \s

You are probably not running OpenRazer from git checkout, the changes with this patch are not yet released. Please be patient or follow https://github.com/openrazer/openrazer/wiki/Building-a-package to build from source.

If you have any problems please join our community chat as that makes it easier to help you https://github.com/openrazer/openrazer/wiki/Community-chat (or otherwise open a new support issue)