openrazer/openrazer #2060: Support for Mouse Dock Pro

lowgos, 2023-05-02:
### Device Info

* Name: Razer Mouse Dock Pro
* Product Number: `RZ81-01990100`
* Model Number: `RC21-019901`

### Device Information
```
Bus 001 Device 009: ID 1532:00a4 Razer USA, Ltd Razer Mouse Dock Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00a4 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Mouse Dock Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1

```

```
total 0
drwxr-xr-x 2 root root 440 May  2 10:33 ./
drwxr-xr-x 4 root root 860 May  2 10:33 ../
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-CX_2.4G_Wireless_Receiver-event-if01 -> ../event20
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-CX_2.4G_Wireless_Receiver-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-CX_2.4G_Wireless_Receiver-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-CX_2.4G_Wireless_Receiver-if01-event-mouse -> ../event19
lrwxrwxrwx 1 root root   9 May  2 10:25 usb-CX_2.4G_Wireless_Receiver-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-Razer_Razer_Mouse_Dock_Pro-event-if01 -> ../event30
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-Razer_Razer_Mouse_Dock_Pro-event-mouse -> ../event27
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-Razer_Razer_Mouse_Dock_Pro-if01-event-kbd -> ../event28
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-Razer_Razer_Mouse_Dock_Pro-if01-event-mouse -> ../event29
lrwxrwxrwx 1 root root   9 May  2 10:25 usb-Razer_Razer_Mouse_Dock_Pro-if01-mouse -> ../mouse3
lrwxrwxrwx 1 root root  10 May  2 10:25 usb-Razer_Razer_Mouse_Dock_Pro-if02-event-kbd -> ../event31
lrwxrwxrwx 1 root root   9 May  2 10:25 usb-Razer_Razer_Mouse_Dock_Pro-mouse -> ../mouse2
...
```

<details>

I also have a Nommo Chroma I removed from previous outputs above. I don't know which part is related to the Mouse Dock Pro below.

```
003:005:000:DESCRIPTOR         1683017860.443568
 05 0C 09 01 A1 01 85 01 19 00 2A 3C 02 15 00 26
 3C 02 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05
 01 09 00 A1 01 85 05 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:009:002:DESCRIPTOR         1683017860.445278
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:009:001:DESCRIPTOR         1683017860.452277
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:009:000:DESCRIPTOR         1683017860.456275
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

```

</details>

### Packet Captures

I have to postpone this part to later this month. This seems complicated and I do not have Windows machine laying around atm. I understand this is useful on your end but this is a lot to ask for now.

If somebody can produce the packet captures for the Mouse Dock Pro feel free to look into it and post it below.

### Synapse Screenshots
Coming today, sorry for the incomplete issue. I have to get a Windows machine.

### Nota Bene

The Mouse Dock Pro can also act as a Hyperpolling dongle (at least for the Razer Basilisk v3 Pro), I had to configure this through Razer Synapse once. This then made the mouse 2.4GHz dongle useless as the mouse now connects to the Mouse Dock Pro. It also means that the mouse doesn't appear in open-razer anymore :cry:

warofsin, 2023-05-19:
I have Mouse Dock Pro with Basilisk V3 Pro mouse connected, Virtual Box with Windows 11 and Razer Synapse. I also have  working Wireshark on OpenSUSe Tumbleweed as a host. The capture shows permanent transfer URB_CONTROL packets (about 25 packets/sec from host to dock and instant reply) even if the mouse itself is turned off. This is not the case if Virtual Box is not running, so I attribute these packets to Synapse. Just now I have no idea how to extract useful information about the protocol, but if there is some idea how to solve this problem, I will do my best.

z3ntu, 2023-05-29:
@warofsin If you can share the pcap with me I can take a look.

warofsin, 2023-06-08:
z3nyu, please, look at these files 
https://drive.google.com/drive/folders/1gcp8WIQ6QDZbdCVm8Mtouhjx7jUhUkKe?usp=drive_link 
here is lsusb output:
Bus 001 Device 006: ID 1532:00a4 Razer USA, Ltd Razer Mouse Dock Pro

0zxw0, 2023-06-27:
no progress

lvctr, 2023-07-13:
I have my hands on the Mouse Dock Pro now and have access to a Windows PC as well. I will supply captures as soon as possible.

z3ntu, 2023-07-16:
See also https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md how you can add support to OpenRazer for this device

lvctr, 2023-07-16:
Thank you, I might give it a shot myself.

NayxYann, 2023-09-18:
Hello everyone, I want to know if everything is moving forward with Mouse Dock Pro support for Openrazer.

Drill-N-Bass, 2023-10-11:
any news in this thread?

lvctr, 2023-10-13:
I got really busy and couldn't really do as much, but I'll try continue over the weekend.

anselale, 2024-05-01:
Has there been any progress on this by any chance?

mrusme, 2024-06-14:
Is this sufficient for further development?

> z3nyu, please, look at these files https://drive.google.com/drive/folders/1gcp8WIQ6QDZbdCVm8Mtouhjx7jUhUkKe?usp=drive_link

Otherwise, I could provide more information if it helps, as I got my hands on the Dock as well.

dougmeek, 2024-07-01:
I'm actively working on the implementation of this one. That said, I only have the Basilisk V3 Pro to test with it, so it will be difficult for me to implement support for the Cobra Pro and Naga V2 Pro devices that are also supported by the Dock Pro.

dougmeek, 2024-07-02:
@z3ntu, I'm hoping you can provide some guidance on this implementation. I am not a master of C or Python, but I'm doing what I can.

I spent a while this evening decomposing the USB commands for the dock and mouse. Essentially, Razer is using the same USB PID for the dock and the paired mouse. Commands sent to 0xFF go to the dock, whereas commands sent to 0x1F go to the mouse.

I've been able to get the dock functions (RGB only) working correctly as an accessory. The mouse is mostly functional as well, through the paired dock. I have a couple of things left to finish up on the mouse functionality, but want to hold until there's a path forward.

The problem I've run into is with having both the dock and the mouse at the same time, since they use the same USB PID. If I try to create an accessory class for the dock and a mouse class for the mouse in the daemon, neither will show up as available for any of the clients. I don't believe this is unexpected behavior, but I'm at an impasse on how to best implement this.

In the official Razer app, each device shows up as a separate entity, which is what I was trying to do here. If I try to combine both devices into a single mouse device, I'm not sure how I'll be able to implement RGB control for the dock as well as the mouse.

Items of note:

- Mouse has 12 LEDs
- Mouse is capable of 125Hz, 500Hz, and 1000Hz polling rate when connected using it's dongle
- Mouse is capable of up to 8000Hz polling rate when using the dock
- Dock has 8 LEDs
- Dock requires a pairing function implemented to be able to connect devices

Any assistance would be appreciated. Thanks!

musk-1024, 2024-07-02:
> @z3ntu, I'm hoping you can provide some guidance on this implementation. I am not a master of C or Python, but I'm doing what I can.
> 
> I spent a while this evening deconstructing the USB commands for the dock and mouse. Essentially, Razer is using the same USB PID for the dock and the paired mouse. Commands sent to 0xFF go to the dock, whereas commands sent to 0x1F go to the mouse.
> 
> I've been able to get the dock functions (RGB only) working correctly as an accessory. The mouse is mostly functional as well, through the paired dock. I have a couple of things left to finish up on the mouse functionality, but want to hold until there's a path forward.
> 
> The problem I've run into is with having both the dock and the mouse at the same time, since they use the same USB PID. If I try to create an accessory class for the dock and a mouse class for the mouse in the daemon, neither will show up as available for any of the clients. I don't believe this is unexpected behavior, but I'm at an impasse on how to best implement this.
> 
> In the official Razer app, each device shows up as a separate entity, which is what I was trying to do here. If I try to combine both devices into a single mouse device, I'm not sure how I'll be able to implement RGB control for the dock as well as the mouse.
> 
> Items of note:
> 
> * Mouse has 12 LEDs
> * Mouse is capable of 125Hz, 500Hz, and 1000Hz polling rate when connected using it's dongle
> * Mouse is capable of up to 8000Hz polling rate when using the dock
> * Dock has 8 LEDs
> * Dock requires a pairing function implemented to be able to connect devices
> 
> Any assistance would be appreciated. Thanks!

So You means You can let both of the dock and mice turn off the rub when the pc is turn off? Synapse offers this function, but only few times it can work, If you can just use a script to turn off the rgb instead of using the synapse, it will be much more stable

gabriel-chojnacki, 2024-07-02:
> @z3ntu, I'm hoping you can provide some guidance on this implementation. I am not a master of C or Python, but I'm doing what I can.
>
> I spent a while this evening deconstructing the USB commands for [...]

Thank you so much for the work! It seems like very important info.

@musk-1024 what do you mean by "turn off the rub"?

lowgos, 2024-07-02:
> @z3ntu, I'm hoping you can provide some guidance on this implementation. I am not a master of C or Python, but I'm doing what I can.
>
> I spent a while this evening deconstructing the USB commands for [...]

Thank you so much for the work! It seems like very important info. Thanks again everyone

@musk-1024 what do you mean by "turn off the rub"?

lowgos, 2024-07-02:
No problem,

I hope we can manage to turn off the RGB as well 👌

dougmeek, 2024-07-02:
OpenRazer is a driver to interface with the hardware and a daemon to provide an interface to other applications. The current implementation of OpenRazer relies on hardware effects as far as I can deduce. This is why some devices don't support all the same features Synapse 3 supports. Synapse 3 uses software effects. If you watch the packet stream coming from Synapse you'll see that it constantly tells the RGB what to do, even on effects that are also implemented in hardware. One of the hardware effects supported by OpenRazer is "none" which essentially turns off the RGB effects. So yes, you should be able to turn off RGB on both devices.

musk-1024, 2024-07-02:
> No problem,
> 
> I hope we can manage to turn off the RGB as well 👌

To be honest, I do figure the problem, If we need to turn off the RGB when turning off the pc, we need to keep the fast boot on, while It will only works at the first time, but It can't be worked anymore later unless we totally restart the system, so I have to use the script to totally turn off the synapse and run it again each time when the pc wake up or turn on.

z3ntu, 2024-07-02:
@dougmeek Thanks for working on this!

I'm honestly not sure what we can do short-term about controlling both dock and connected mouse.. Everything in OpenRazer currently assumes one USB device = one Razer device and I think this assumption is there for a lot of things.

We also don't have the simple flexibility of targetting a different device (using transaction id - where btw the first 3 bits are meant as "command steering" to the correct device, so Razer designed it like that a long time ago) from the kernel driver as we don't really have any way to tell the kernel driver part that.

Very long term we want to move most of the packet handling into user space via hidraw, then we can easily put a second device on the same USB device and control everything more directly and easier. But again, shorter term I can't think of much apart from hacks all over the place (e.g. add sysfs file to driver that changes the target device for the following commands, and somehow get the daemon to I guess enumerate which devices might be attached to the dock and then instantiate devices based on that or who knows how)

dougmeek, 2024-07-02:
@z3ntu, thanks for your reply.

I'll continue to work on this and see what I can come up with. If you have any more tips, I would appreciate it.

z3ntu, 2024-07-02:
Nothing really for now but I guess as first step just get the dock working with the leds there, get a PR with that open and integrated. Because anything more will probably take a while and if just the dock is simple-ish to support then we should do that first.

KrAloe, 2024-08-27:
I have 2 dock pro and 2 naga v2 pro. if you need something then tell me.

wbonbon, 2024-09-16:
I came here looking for a way to know the battery level of Basilisk V3 Pro when using Mouse Dock Pro.
I created an environment to capture packets using wireshark 4.2.7 and USBPcap.
(For some reason, the USBPcap plugin does not work in wireshark 4.4)

However, I don't know the appropriate filter, so it's difficult to analyze.....
Any advice?
[basiliskv3pro_dock_pro.zip](https://github.com/user-attachments/files/17008625/basiliskv3pro_dock_pro.zip)

plexecutor, 2024-10-10:
I have a Mouse Dock Pro with a Basilisk V3 Pro and I am willing to provide any information needed to assist with this effort

kathampy, 2024-10-26:
Mouse Dock Pro has a hardware effect for changing the LED colour based on battery level. This effect persists once set and continues working after uninstalling Synapse.

Fr0g-Not-Found, 2025-01-03:
I mean If I could just set polling rate I would be really happy.

dougmeek, 2025-01-24:
Apologies for the delay on this, folks. I actually do not have this mouse/dock combination anymore. I started having issues with the combination ~3 months ago where the mouse would stop working and I'd have to unplug and replug the dock to get the mouse to work again (in both Linux and Windows). This janky behavior became annoying enough that I replaced it with something else entirely.

As for this specific issue, the details for interacting with the devices are listed in my comment above if anyone wants to pick up the torch. That said, based on @z3ntu's input, it will likely be difficult to fully implement functionality for this device configuration due to the architecture of openrazer. It may only be possible to implement functionality with the dock and not the mouse until which point some architectural changes are made.

As I no longer have this hardware, I will no longer be actively developing for this issue.

Dragoon91786, 2025-01-30:
> Apologies for the delay on this, folks. I actually do not have this mouse/dock combination anymore. I started having issues with the combination ~3 months ago where the mouse would stop working and [...]

Thanks for your assistance. I also noticed the issue. Though, I figured out it's very likely related to which port you plug in and the latest firmware from Razer. Unfortunately, Razer makes downgrading firmware f-ing difficult to impossible without special hardware or know-how. That said, As someone with the white model of Basilisk v3 PRO and the dongle, I was hoping to get the 8,000 polling rate. For anyone curious, I found that with the latest firmware, this particular dock seems to have the mouse freeze up when the mouse is not directly connected through a USB-C to USB-A cable and port directly to the motherboard or which has its own dedicated controller. I am not sure why that is, but on my particular laptop [2023 mobo of ASUS ROG Strix Scar 17"], the mouse tended to freeze up whenever I used my USB-C docks. If I avoided them and directly connected the mouse via USB-C to USB-A it didn't have that issue (or it happened far less often). If I used the USB-C to USB-A cables it likewise didn't happen as often. My guess is that the issue is a firmware one that Razer's bios engineer(s)/programmer(s) f-ed up and refuse to fix cuz it isn't worth their time/money.

Maybe, we can spam their Social Media and get them to look into the issue?

Either way, thanks in any event, @dougmeek for what work you've done so far. Even if I can't adequately control the polling rate or have the dameon recognized by the app, at least I can get the mouse to work with the dongle. My biggest qualm is that at lower polling rates and higher DPI, the mouse jerks, and while that isn't typically noticeable, I find it SUPER annoying for general usage. I'm trying to see if 1,000 will make any improvements, but I am not particularly confident, as I find I have similar issues in Windows proper.

Thanks again to everyone working on this project! <3 You're all awesome! <3

ckefalianou, 2025-05-14:
Hi guys. Trying to make the mouse dock pro work as an accessory (rgb etc) but no luck. My device recognizes the dock but commands are somehow rejected from the dock and no changes take place. ideas? @dougmeek I know it's been a while but if you could provide some working branch for me to study could be great

thank you!

dougmeek, 2025-05-14:
Apologies @ckefalianou, but my dev VM for this is long gone. Most of the info you need is above in comments. Sorry I can't be more help.

dougmeek, 2025-05-14:
> [@z3ntu](https://github.com/z3ntu), I'm hoping you can provide some guidance on this implementation. I am not a master of C or Python, but I'm doing what I can.
>
> I spent a while this evening [...]

This is really all the info I recall from this one. Your issue is likely with sending commands to the wrong endpoint of the PID. As explained in comments above, this one is going to be particularly challenging because OpenRazer isn't really set up to allow multiple endpoints on the same PID. Good luck!

ckefalianou, 2025-05-14:
> This is really all the info I recall from this one. Your issue is likely with sending commands to the wrong endpoint of the PID. As explained in comments above, this one is going to be particularly [...]

Oh great... 🥲

Anyway, thank you again for your help ☺️

mpkossen, 2025-12-30:
> Any update on this?

If you just want to control the LEDs install OpenRGB, that works just fine with the Mouse Dock Pro.

welshbloke70, 2026-02-25:
Has there been an update to the Mouse Dock Pro? Mines suddenly stopped working, in that I get no mouse response through it suddenly. Mouse (Basilisk pro 35K) connects fine through blue tooth or the usb dongle. Also is still paired with the dock if I plug it all into a windows machine.
Its a bit of a pain as I cant even use the dock to charge the mouse now as if I plug the dock in the mouse stops working!

z3ntu, 2026-02-26:
Yeah, that's the daily repo. Does the dock show up in e.g. polychromatic or RazerGenie actually? Does any of the behavior change when you stop the daemon (`systemctl --user stop openrazer-daemon.service`)?