openrazer/openrazer #2061: Open Build Service key expired for project

term1n4l, 2023-05-03:
to whom it concerns:

Your SUSE Open Build Service project key is expired and requires updating for the Fedora yum repos

Thanks

``` bash
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
      because: The primary key is not live
      because: Expired on 2021-03-28T09:55:39Z
  2. Key 062F9FD4D6D11CE4 invalid: key is not alive
      because: The primary key is not live
      because: Expired on 2021-03-28T09:55:39Z
```

z3ntu, 2023-05-03:
This key should still be valid for a few more weeks until 2023-05-23
```
$ curl -s http://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key | gpg
gpg: WARNING: no command supplied.  Trying to guess what you mean ...
pub   rsa2048 2014-09-16 [SC] [expires: 2023-05-23]
      E96C496E4E77C159C2FCA053062F9FD4D6D11CE4
uid           hardware OBS Project <hardware@build.opensuse.org>
```
Do you have this problem since 2021 or did you just recently get this issue? Assuming you had openrazer installed for longer already.

term1n4l, 2023-05-04:
only just recently started since I upgraded to Fedora 38 (well technically since openrazer-3.6 released), I imported the key from the repo metadata and it says the same as what you have there:

``` bash
pub   rsa2048 2014-09-16 [SC] [expires: 2023-05-23]
      E96C496E4E77C159C2FCA053062F9FD4D6D11CE4
uid           [ unknown] hardware OBS Project <hardware@build.opensuse.org>
```

but rpm/dnf says the private key has expired.

z3ntu, 2023-05-04:
Just tried in a container, works fine here.
```
[root@5e56e7271498 /]# dnf config-manager --add-repo https://download.opensuse.org/repositories/hardware:razer/Fedora_$(rpm -E %fedora)/hardware:razer.repo
Adding repo from: https://download.opensuse.org/repositories/hardware:razer/Fedora_38/hardware:razer.repo
[root@5e56e7271498 /]# dnf install openrazer-meta
<snip>
Importing GPG key 0xD6D11CE4:
 Userid     : "hardware OBS Project <hardware@build.opensuse.org>"
 Fingerprint: E96C 496E 4E77 C159 C2FC A053 062F 9FD4 D6D1 1CE4
 From       : https://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key
Is this ok [y/N]: y
Key imported successfully
<snip>
```

term1n4l, 2023-05-04:
sure so it's probably a cached key somewhere, but it should have updated the key in any case and I already tried delete and re-import to rpm

z3ntu, 2023-05-04:
Unfortunately I'm not experienced with dnf/rpm enough to be able to help here. Please check in Fedora forums or somewhere, and if you figure it out please post it here!

cfeller, 2023-05-06:
I figured out the issue. Like the OP, my Fedora system has been upgraded through many Fedora releases. I was also able to confirm that it works successfully in a Fedora 38 container. I was able to narrow it down to an RPM issue.  

Checking signatures on, say polychromatic

`rpm --checksig $(dnf download --url polychromatic -q)`

yielded the following on the affected system:

`digests SIGNATURES NOT OK`
 
But the same command inside the container:
 
`digests signatures OK`
 
 
Comparing the GPG keys between the affected system and the container show the same key, however on the affected system it shows that the key was imported with a much older version of RPM.  

The solution here is to remove the old GPG key so that it can be re-added with the current RPM version:

`rpm -e gpg-pubkey-d6d11ce4`

Then dnf upgrade/install works with the openrazer packages:

```
~> dnf upgrade
...
Downloading Packages:
[SKIPPED] openrazer-daemon-3.6.0-2.1.noarch.rpm: Already downloaded                                                                                                                       
[SKIPPED] openrazer-kernel-modules-dkms-3.6.0-2.1.noarch.rpm: Already downloaded                                                                                                          
[SKIPPED] openrazer-meta-3.6.0-2.1.noarch.rpm: Already downloaded                                                                                                                         
[SKIPPED] polychromatic-0.8.1-1.1.noarch.rpm: Already downloaded                                                                                                                          
[SKIPPED] python3-openrazer-3.6.0-2.1.noarch.rpm: Already downloaded                                                                                                                      
hardware:razer (Fedora_38)                                                                                                                                 1.5 kB/s | 1.1 kB     00:00    
Importing GPG key 0xD6D11CE4:
 Userid     : "hardware OBS Project <hardware@build.opensuse.org>"
 Fingerprint: E96C 496E 4E77 C159 C2FC A053 062F 9FD4 D6D1 1CE4
 From       : https://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key
Is this ok [y/N]: y
Key imported successfully

```

I'm not 100% sure of the root cause here, however I know that Fedora significantly tightened security policies with this release (38) which caused problems with other 3rd party repos as well. I suspect that may have been related as to what was going on here.

term1n4l, 2023-05-14:
hmmm so I did what was suggested above, now I'm getting a different error:

```bash
Downloading Packages:
[SKIPPED] openrazer-daemon-3.6.1-1.1.noarch.rpm: Already downloaded                                                                                        
[SKIPPED] openrazer-kernel-modules-dkms-3.6.1-1.1.noarch.rpm: Already downloaded                                                                           
[SKIPPED] python3-openrazer-3.6.1-1.1.noarch.rpm: Already downloaded                                                                                       
Running transaction check
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2022-12-16T20:35:28Z
error: rpmdbNextIterator: skipping h#   16491 
Header V3 RSA/SHA256 Signature, key ID d6d11ce4: BAD
Header SHA256 digest: OK
Header SHA1 digest: OK
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2022-12-16T20:35:28Z
error: rpmdbNextIterator: skipping h#   16491 
Header V3 RSA/SHA256 Signature, key ID d6d11ce4: BAD
Header SHA256 digest: OK
Header SHA1 digest: OK
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: An rpm exception occurred: package not installed
```

term1n4l, 2023-05-14:
looks like I found a partial solution. I had to perform the following:

`sudo rpm -e --nodeps --nodigest --nosignature python3-openrazer libopenrazer openrazer-daemon razergenie openrazer-kernel-modules-dkms`

and then after this I was able to install openrazer itself. However it looks like razergenie and libopenrazer may be still problematic:

```bash
$ sudo dnf install razergenie
Last metadata expiration check: 1:12:51 ago on Sun 14 May 2023 13:32:31.
Install  2 Packages

error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:48:56Z
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:48:56Z
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:50:23Z
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:50:23Z
Problem opening package libopenrazer-0.1.0-1.1.x86_64.rpm
Problem opening package razergenie-1.0.0-1.1.x86_64.rpm
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: GPG check FAILED
```

kierun, 2023-05-17:
I am having the same issue here: multiple upgrades of fedora with OpenRazer installed.

```
√ ; sudo rpm -e gpg-pubkey-d6d11ce4
√ ; sudo dnf up
Last metadata expiration check: 0:03:59 ago on Wed 17 May 2023 09:52:13 BST.
Dependencies resolved.
====================================================================================================================================================================================================
 Package                                                      Architecture                          Version                                     Repository                                     Size
====================================================================================================================================================================================================
Upgrading:
 openrazer-daemon                                             noarch                                3.6.1-1.1                                   hardware_razer                                189 k
 openrazer-kernel-modules-dkms                                noarch                                3.6.1-1.1                                   hardware_razer                                 62 k
 openrazer-meta                                               noarch                                3.6.1-1.1                                   hardware_razer                                7.0 k
 python3-openrazer                                            noarch                                3.6.1-1.1                                   hardware_razer                                 60 k

Transaction Summary
====================================================================================================================================================================================================
Upgrade  4 Packages

Total size: 317 k
Is this ok [y/N]: y
Downloading Packages:
[SKIPPED] openrazer-daemon-3.6.1-1.1.noarch.rpm: Already downloaded
[SKIPPED] openrazer-kernel-modules-dkms-3.6.1-1.1.noarch.rpm: Already downloaded
[SKIPPED] openrazer-meta-3.6.1-1.1.noarch.rpm: Already downloaded
[SKIPPED] python3-openrazer-3.6.1-1.1.noarch.rpm: Already downloaded
hardware:razer (Fedora_38)                                                                                                                                          6.9 kB/s | 1.1 kB     00:00
Importing GPG key 0xD6D11CE4:
 Userid     : "hardware OBS Project <hardware@build.opensuse.org>"
 Fingerprint: E96C 496E 4E77 C159 C2FC A053 062F 9FD4 D6D1 1CE4
 From       : https://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key
Is this ok [y/N]: y
Key imported successfully
Running transaction check
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-30T13:06:10Z
error: rpmdbNextIterator: skipping h#   23265
Header V3 RSA/SHA256 Signature, key ID d6d11ce4: BAD
Header SHA256 digest: OK
Header SHA1 digest: OK
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-30T13:06:10Z
error: rpmdbNextIterator: skipping h#   23265
Header V3 RSA/SHA256 Signature, key ID d6d11ce4: BAD
Header SHA256 digest: OK
Header SHA1 digest: OK
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: An rpm exception occurred: package not installed
```

I am happy to test a few things if that helps…

kierun, 2023-05-19:
@term1n4l Thank you very much indeed! Your help is much appreciated.

For posterity, I had to remove some old keys via the following commands:

```
rpm -q gpg-pubkey --qf '%{NAME}-%{VERSION}-%{RELEASE}\t%{SUMMARY}\n' | rg -i open
rpm -e  gpg-pubkey-XXX
```

Thereafter, everything worked. Although, I have not restarted my machine so…

z3ntu, 2023-05-25:
Fwiw the key that was present in the repo a couple of weeks ago, is now valid until 2025
```
$ curl -s http://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key | gpg
[...]
pub   rsa2048 2014-09-16 [SC] [expires: 2025-07-17]
[...]
```
Still cannot really help more with the rpm/Fedora issues, unfortunately.

closed 2023-10-24 by github-actions[bot]

smartbit, 2023-11-03:
[This gist](https://gist.github.com/e7d/3b786c7410ca14a5ded61eec36de9874) explains how to remove the expired pubkeys.
```bash
sudo rpm -q gpg-pubkey --qf '%{NAME}-%{VERSION}-%{RELEASE}\t%{SUMMARY}\n' | grep -iE 'obs|suse'
```