openrazer/openrazer #2062: KeyError: 'backlight_active' (Razer Naga 2014)

winf-bash, 2023-05-05:
### Distribution

Open Suse Tumbleweed

### Support

- [yes ] Have you tried rebooting?
- [yes] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

After updating OS Tumbleweed and Openrazer to 3.6 I got several issues. I'm using Openrazse with Polychromatic and was able to solve all the issues except this one here: 

Daemon is running -> Failed 

If I run this command "openrazer-daemon -Fv" 

I get this response: 

(long list of "loaded device specification") ..... and then: 

2023-05-05 09:50:09 | razer                          | INFO     | Found device.0: 0003:1532:0040.0006
2023-05-05 09:50:09 | razer.device0                  | INFO     | Initialising device.0 RazerNaga2014
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSerial method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDriverVersion method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceImage method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.restoreLastEffect method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.backlight.getBacklightEffect method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.backlight.getBacklightEffectSpeed method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.backlight.getBacklightWaveDir method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.backlight.getBacklightEffectColors method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoEffect method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoEffectSpeed method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoEffectColors method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoWaveDir method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollEffectColors method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollEffect method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollEffectSpeed method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollWaveDir method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.dpi.maxDPI method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.dpi.getDPI method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.dpi.setDPI method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.getPollRate method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.misc.setPollRate method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoActive method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.setLogoActive method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollActive method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.setScrollActive method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.backlight.setBacklightActive method to DBus
2023-05-05 09:50:09 | razer.device0                  | DEBUG    | Adding razer.device.lighting.backlight.getBacklightActive method to DBus
Traceback (most recent call last):
  File "/usr/lib64/python3.10/configparser.py", line 791, in get
    value = d[option]
  File "/usr/lib64/python3.10/collections/__init__.py", line 986, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib64/python3.10/collections/__init__.py", line 978, in __missing__
    raise KeyError(key)
KeyError: 'backlight_active'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/daemonize.py", line 248, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 141, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3.10/site-packages/openrazer_daemon/daemon.py", line 117, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3.10/site-packages/openrazer_daemon/daemon.py", line 491, in _load_devices
    razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
  File "/usr/lib/python3.10/site-packages/openrazer_daemon/hardware/device_base.py", line 271, in __init__
    self.zone[i]["active"] = self.persistence.getboolean(self.storage_name, i + '_active')
  File "/usr/lib64/python3.10/configparser.py", line 830, in getboolean
    return self._get_conv(section, option, self._convert_to_boolean,
  File "/usr/lib64/python3.10/configparser.py", line 810, in _get_conv
    return self._get(section, conv, option, raw=raw, vars=vars,
  File "/usr/lib64/python3.10/configparser.py", line 805, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib64/python3.10/configparser.py", line 794, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'backlight_active' in section: 'PM1450104436789'

Stopping daemon.
Exception ignored in atexit callback: <bound method Daemonize.exit of <daemonize.Daemonize object at 0x7f736afcbb50>>
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/daemonize.py", line 72, in exit
    sys.exit(0)
SystemExit: 0

---
Does anyone know how I can fix this error?

z3ntu, 2023-05-05:
Could you please share the file `~/.config/openrazer/persistence.conf` ? You can then delete that file and the daemon should start again. Not sure what went wrong there.

winf-bash, 2023-05-05:
It worked! Thanks a lot! 

This was the content of the old persistence.conf file: 

[PM1450104436789]
dpi_x = 800
dpi_y = 800
poll_rate = 1000
logo_active = True
logo_brightness = 75.0
logo_effect = spectrum
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1
scroll_active = True
scroll_brightness = 75.0
scroll_effect = spectrum
scroll_colors = 0 255 0 0 255 255 0 0 255
scroll_speed = 1
scroll_wave_dir = 1

commit b99f6027 (2023-05-13): daemon: Handle missing values in persistence.conf

This especially bites when a new config key is added, as such happened
recently with backlight_active.

Also let's add some logging when this happens so at least there's some
indication in the log that something didn't go quite right.

Fixes #2062
Fixes #2064
closed 2023-05-05 by winf-bash