openrazer/openrazer #2063: Daemon will not start

iguanatamer, 2023-05-08:
### Distribution

Arch Linux 6.3.1-arch1-1

### Support

- [ ] Have you tried rebooting? Yes
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page? Yes

I have added my user to plugdev and the daemon will still not start. I have also tried the git package too on the AUR with the same error.

``` 
user@hostname ~ $ systemctl --user status openrazer-daemon.service
× openrazer-daemon.service - Daemon to manage razer devices in userspace
     Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Sun 2023-05-07 22:24:51 CDT; 1min 30s ago
       Docs: man:openrazer-daemon(8)
    Process: 178094 ExecStart=/usr/bin/openrazer-daemon -F (code=exited, status=1/FAILURE)
   Main PID: 178094 (code=exited, status=1/FAILURE)
        CPU: 17ms

May 07 22:24:51 hostname systemd[919]: Starting Daemon to manage razer devices in userspace...
May 07 22:24:51 hostname openrazer-daemon[178094]: Traceback (most recent call last):
May 07 22:24:51 hostname openrazer-daemon[178094]:   File "/usr/bin/openrazer-daemon", line 12, in <module>
May 07 22:24:51 hostname openrazer-daemon[178094]:     from openrazer_daemon.daemon import RazerDaemon, __version__
May 07 22:24:51 hostname openrazer-daemon[178094]: ModuleNotFoundError: No module named 'openrazer_daemon'
May 07 22:24:51 hostname systemd[919]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
May 07 22:24:51 hostname systemd[919]: openrazer-daemon.service: Failed with result 'exit-code'.
May 07 22:24:51 hostname systemd[919]: Failed to start Daemon to manage razer devices in userspace.

z3ntu, 2023-05-08:
https://wiki.archlinux.org/title/Arch_User_Repository#Installing_and_upgrading_packages

> Note: The AUR is unsupported, so any packages you install are your responsibility to update, not pacman's. If packages in the official repositories are updated, you will need to rebuild any AUR packages that depend on those libraries.

In this case, recently Python was upgraded to 3.11 so you need to rebuild any AUR packages that depend on Python. Check `pacman -Qo /usr/lib/python3.10`

iguanatamer, 2023-05-08:
Thank you,

I have never ran into that ever and I've been running arch linux for like 5 years now.

closed 2023-05-08 by iguanatamer