openrazer/openrazer #2064: Backlight not working on latest update.

Izisery, 2023-05-10:
### Distribution

    OS: Pop!_OS 22 
    Kernel: 6.2.6- 
    OpenRazer: v3.6.0

 lsusb | grep 1532
Bus 003 Device 003: ID 1532:021e Razer USA, Ltd Ornata Chroma
Bus 002 Device 002: ID 1532:003f Razer USA, Ltd Naga Epic Chroma (Wireless)

### Support

- [Y ] Have you tried rebooting?
- Only Cold boots have any effect
- [ Y] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Razer daemon fails to start, keyboard backlight doesn't turn on unless I do a cold boot, where I completely shutdown the PC, remove power for 30 seconds, then reboot, but Journalctrl still reports that the daemon failed to start. Polychromatic reports devices as unknown, fails to restart Daemon. 

I'm new to linux so I'm not sure how to troubleshoot further on my own, issue only started after the recent openrazer update. 

Here's the error log when I try to start the Daemon myself:
```
izi@abhorsen:~$ openrazer-daemon -F
Starting daemon.
2023-05-09 19:52:21 | razer                          | INFO     | Initialising Daemon (v3.6.0). Pid: 11059
2023-05-09 19:52:21 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2023-05-09 19:52:21 | razer                          | INFO     | Found device.0: 0003:1532:003F.0001
2023-05-09 19:52:21 | razer.device0                  | INFO     | Initialising device.0 RazerNagaEpicChromaWireless
Traceback (most recent call last):
  File "/usr/lib/python3.10/configparser.py", line 790, in get
    value = d[option]
  File "/usr/lib/python3.10/collections/__init__.py", line 986, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.10/collections/__init__.py", line 978, in __missing__
    raise KeyError(key)
KeyError: 'backlight_active'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 248, in start
    self.action(*privileged_action_result)
  File "/usr/bin/openrazer-daemon", line 141, in run_daemon
    daemon = RazerDaemon(verbose=args.verbose,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 117, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/daemon.py", line 491, in _load_devices
    razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/mouse.py", line 1580, in __init__
    super(RazerNagaEpicChromaWired, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/openrazer_daemon/hardware/device_base.py", line 271, in __init__
    self.zone[i]["active"] = self.persistence.getboolean(self.storage_name, i + '_active')
  File "/usr/lib/python3.10/configparser.py", line 829, in getboolean
    return self._get_conv(section, option, self._convert_to_boolean,
  File "/usr/lib/python3.10/configparser.py", line 809, in _get_conv
    return self._get(section, conv, option, raw=raw, vars=vars,
  File "/usr/lib/python3.10/configparser.py", line 804, in _get
    return conv(self.get(section, option, **kwargs))
  File "/usr/lib/python3.10/configparser.py", line 793, in get
    raise NoOptionError(option, section)
configparser.NoOptionError: No option 'backlight_active' in section: 'PM1541123700079'

Stopping daemon.
Exception ignored in atexit callback: <bound method Daemonize.exit of <daemonize.Daemonize object at 0x7ff6c0717f70>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/daemonize.py", line 72, in exit
    sys.exit(0)
SystemExit: 0

```

z3ntu, 2023-05-10:
See https://github.com/openrazer/openrazer/issues/2062#issuecomment-1536154232 (no need to share that file)

Appears to be an actual bug though, I'll look into fixing this but you can use that workaround for now.

ThornedRose, 2023-05-11:
> See [#2062 (comment)](https://github.com/openrazer/openrazer/issues/2062#issuecomment-1536154232) (no need to share that file)
> 
> Appears to be an actual bug though, I'll look into fixing this but you can use that workaround for now.

I'm having the same  issue on Arch Linux. I've gone through the entire Troubleshooting guide and everything all checks out but for the daemon:
```
$ openrazer-daemon -Fv
Traceback (most recent call last):
  File "/usr/bin/openrazer-daemon", line 12, in <module>
    from openrazer_daemon.daemon import RazerDaemon, __version__
ModuleNotFoundError: No module named 'openrazer_daemon'
```

I've tried reinstalling multiple times, including installing the git version instead. Removing the `persistence.conf` file made no difference.

z3ntu, 2023-05-11:
@ThornedRose See https://github.com/openrazer/openrazer/issues/2063#issuecomment-1537938508 And this has nothing to do with the issue here.

commit b99f6027 (2023-05-13): daemon: Handle missing values in persistence.conf

This especially bites when a new config key is added, as such happened
recently with backlight_active.

Also let's add some logging when this happens so at least there's some
indication in the log that something didn't go quite right.

Fixes #2062
Fixes #2064
closed 2023-05-13 by z3ntu