openrazer/openrazer #2066: Settings Not Applied Persistent On Dual Boot

0pLuS0, 2023-05-11:
### Distribution

Slackware 15.0 & Ubuntu 23.04

openrazer 3.6.0
Razer Baslisk Essential
Huntsman v2 TKL

### Support

- [ X] Have you tried rebooting?
- [X ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

This is an issue I've run across with two distros, Slack and Ubuntu at present, so this is not an issue specific to any Linux distro.

When running a Dual Boot box, I've noticed everytime, when I have booted into Windows, then restart the box, and at a later point in time, then booted into Linux that the settings for my keyboard and mouse are not applied, evenn though I have; ```restore_persistence = true```.

I use polychromatic on both distros, and with each one, after I've booted into Windows, then boot into Linux, I have to open polychromatic and play with the settings until I get them to stick.

So as an example, both in Windows and Linux I have the mouse rgb lighting off, when I boot into Linux the rgb is only and changing colors. So I'll click on Spectrum, or any other option, it doesn't matter, then click on None, but I might have to do this a few times, before the rgb light also goes off. This is the same with my keyboard a Huntsman v2 TKL. I use Static for the default Razer green, it will cycle through colors when I boot into Linux and I have to click the other settings then Static for it to apply, again, I might have to do this several times.

I've already had several talks in the past, on another issue I reported, and nothing works.

I hope there is something can be finally done, either on my end to make this work, or in the near future you will figure out a way to make this work on Dual Boot.

THANKS

z3ntu, 2023-05-11:
Any errors from razerkbd/razermouse in dmesg after bootup? Also check `~/.config/openrazer/persistence.conf` (I hope I got the path correct). There you basically should have one entry per device with the settings for that device.

0pLuS0, 2023-05-12:
I've never seen any errors before...

Path is; ```~/.config/openrazer/persistence.conf```

z3ntu, 2023-05-12:
You can see on the mouse it's saved none effect and zero brightness, and on the keyboard 35% brightness (or 35/255, not sure right now how it's saved) and static effect. In theory when you restart openrazer-daemon these effects should be applied to the devices, given you have the restore effect option enabled in the config.

After changing an effect normally it takes a few seconds until the effect is written to disk (batching to avoid lots of writes) but then it should be updated.

0pLuS0, 2023-05-12:
Sorry that I mentioned the mouse earlier, the reason I brought that up, is that in polychromatic the Y axis always shows 0. But then I noticed, I click on the dpi I want, it will show that number for both the x and y axis, but when I close and reopen polychromatic, the number I picked for the dpi will only show for the x axis, y is always showing as 0.

As for the mouse lighting staying off, for the Effect, this is working.

The keyboard will stay at the Brightness Level, but the colors are cycling through different colors. It never applies the Static I picked, I have to manually pick it over.

0pLuS0, 2023-05-12:
I just rebooted the computer several times, I was working on something else. I only rebooted to the grub screen, then booted back into Ubuntu.
 
Doing these several reboots, one of the times I logged back into X, the mouse dpi didn't stick and it was on 1800 when I opened up polychromatic.

hmm

z3ntu, 2023-05-13:
When you see the 400 x 0 DPI in polychromatic can you please run `qdbus --literal org.razer /org/razer/device/XX0000000065 razer.device.dpi.getDPI` and send the output? Not sure yet whether this is an openrazer bug or a UI bug in polychromatic.

0pLuS0, 2023-05-14:
For starters, qdbus wasn't installed on my system, not sure if it's suppose to be in Ubuntu 23.04 by default, or at least installed when I install polychromatic? hmm

I ran the cmd, here's the output;

qdbus --literal org.razer /org/razer/device/XX0000000065 razer.device.dpi.getDPI
qdbus: could not find a Qt installation of ''

z3ntu, 2023-05-15:
Damn Ubuntu, I think you need to replace qdbus with qdbus-qt5 (or qdbus-qt6 should also work..), so you probably have to install whatever package provides that.

This error is from qtchooser which is some weird wrapper around some tools but doesn't even give any useful errors for people that don't know about it.

0pLuS0, 2023-06-25:
qdbus isn't even installed on my box. Wasn't it suppose to be when installing polychromatic, or openrazer?

0pLuS0, 2023-06-25:
Ubuntu 23.04 uses qt5 so I ran sudo apt install qdbus, which installed qdbus-qt5.

When I rebooted into Windows, then booted back into Ubuntu the settings were applied to the keyboard this time, so maybe now it's ok, that qdbus is installed?

I get this now from that cmd;

qdbus --literal org.razer /org/razer/device/XX0000000065 razer.device.dpi.getDPI
Error: org.freedesktop.DBus.Error.UnknownMethod
Method "getDPI" with signature "" on interface "razer.device.dpi" doesn't exist

z3ntu, 2023-06-25:
Ah sorry, guess you need to replace `XX0000000065` with `632031H11805501` in the command, the serial number of the device.

0pLuS0, 2023-06-27:
Installing qdbus didn't do anything, after I was booted into Windows, then later booted into Ubuntu, the keyboard settings weren't applied when I logged into X, I then had to log out and back in for them to take effect. hmm ;/

Here's the output;

qdbus --literal org.razer /org/razer/device/632031H11805501 razer.device.dpi.getDPI
[Argument: ai {500, 0}]

0pLuS0, 2023-06-28:
Also, if I open up polychromatic and click on the various effects, and click the Static one, which is what I originally applied it will eventually work.

Also, to make any of the keyboard Effects work, I have to click on them several times. I can't just click them one time and they work, they have to be clicked several times.

0pLuS0, 2023-08-26:
I'm now using 3.6.1 in Slackware, and still seeing the same issue, where I have to open Polychromatic when I log into Slackware, after using Windows and click on the effect I use, to get it working.

When I was playing around in Ubuntu this seemed to start working with the latest 3.6.1, but not in Slackware.

Other than systemd, what would be going on in Ubuntu that this would work now, but not in Slackware?

z3ntu, 2023-08-28:
Not aware of anything that should give you different behavior there...

lah7, 2023-08-28:
Do both installations have the same `~/.config/openrazer/razer.conf` settings?
In particular,  `restore_persistence = True`?

Some clues could be to compare the daemon logs to know if it is actually sending a command to set effect after login.

1. Delete `~/.local/share/openrazer/logs/razer.log` 
2. Set `verbose_logging = True` in `razer.conf` ready for next daemon launch.
3. Replicate the problem and compare the log to the other distro.

I'd **suspect** that booting Linux to Linux will be OK, but booting Windows into Linux is when the hardware state doesn't play nice.

0pLuS0, 2023-08-28:
I'm only using Slackware now, so I can't tell you about Ubuntu.

This is the contents of razer.conf in Slackware.

[General]
# Verbose logging (logs debug messages - lotsa spam)
verbose_logging = False

[Startup]
# Set the sync effects flag to true so any assignment of effects will work across devices
sync_effects_enabled = True

# Turn off the devices when the systems screensaver kicks in
devices_off_on_screensaver = True

# Battery notifier
battery_notifier = True

# Battery notification frequency [s] (0 to disable)
battery_notifier_freq = 600

# Battery notifications appear when device reaches this percentage
battery_notifier_percent = 33

# Apply effects saved to disk when daemon starts
restore_persistence = True

lah7, 2023-08-29:
As a **guess**, this keyboard might not be "cooperating" when the daemon starts. After all, the device is asked a bunch of stuff like serial, DPI, brightness, poll rate, etc and then told to set an effect (with `restore_persistence`). Some devices don't like being asked too much at once, so it might be ignoring the `static` request. If that's the case, there _should_ be errors in the driver log:

    sudo dmesg | grep razer

If you're looking for a workaround / something to try, you could set up a script to run at login to always _enforce_ the static effect is being set. Like so:

```python
#!/usr/bin/python3
import openrazer.client

devman = openrazer.client.DeviceManager()
devman.sync_effects = True
devices = devman.devices

devices[0].fx.static(0,255,0)
#    devices[0].fx.advanced.backlight.static(0,255,0)    # <-- or it might be this
```

If that doesn't work, then it might need a second or two delay before setting the effect.

0pLuS0, 2023-08-30:
I never seen any errors in the logs for the keyboard or mouse.

>sudo dmesg | grep razer
[    5.324500] razermouse 0003:1532:0065.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Basilisk Essential] on usb-0000:08:00.3-1/input0

[    5.380361] razermouse 0003:1532:0065.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Basilisk Essential] on usb-0000:08:00.3-1/input1

[    5.437392] razermouse 0003:1532:0065.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Basilisk Essential] on usb-0000:08:00.3-1/input2

[    5.496407] razerkbd 0003:1532:026B.0005: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Huntsman V2 Tenkeyless] on usb-0000:08:00.3-2/input0

[    5.548444] razerkbd 0003:1532:026B.0006: input,hidraw4: USB HID v1.00 Keyboard [Razer Razer Huntsman V2 Tenkeyless] on usb-0000:08:00.3-2/input1

[    5.550464] razerkbd 0003:1532:026B.0007: input,hidraw5: USB HID v1.00 Mouse [Razer Razer Huntsman V2 Tenkeyless] on usb-0000:08:00.3-2/input2

[    5.550963] razerkbd 0003:1532:026B.0008: No inputs registered, leaving

[    5.550998] razerkbd 0003:1532:026B.0008: hidraw6: USB HID v1.00 Device [Razer Razer Huntsman V2 Tenkeyless] on usb-0000:08:00.3-2/input3

[    5.603451] razerkbd 0003:1532:026B.0009: input,hiddev96,hidraw7: USB HID v1.00 Device [Razer Razer Huntsman V2 Tenkeyless] on usb-0000:08:00.3-2/input4

0pLuS0, 2023-09-12:
@lah7 I tried using that python script you posted it doesn't work.

0pLuS0, 2023-09-15:
@lah7 

The python script works like this;

```#!/usr/bin/python3
import openrazer.client

devman = openrazer.client.DeviceManager()
devman.sync_effects = True
devices = devman.devices

devices[0].fx.advanced.backlight.static(0,255,0)
```

But then I get this output;
```
File "/home/foo/.config/autostart/razer_startup.py", line 9, in <module>
    devices[0].fx.advanced.backlight.static(0,255,0)    # <-- or it might be this
AttributeError: 'RazerAdvancedFX' object has no attribute 'backlight'
```

lah7, 2023-09-16:
Oops, should've been `fx.misc.backlight` instead of `fx.advanced.backlight`, my bad.

Just realised since you have 2 devices, `devices[0]` could be either one of them. This code would be more robust if you're looking to set static for both:

```python
for device in devman.devices:
    if device.has("lighting_static"):
        device.fx.static(0, 255, 0)
    if device.has("lighting_backlight"):
        device.fx.misc.backlight.static(0, 255, 0)
```

Reference: [examples/cpu_temperature.py](https://github.com/openrazer/openrazer/blob/master/examples/cpu_temperature.py)

0pLuS0, 2023-09-16:
So do I just make it like this?

```
#!/usr/bin/python3
import openrazer.client

for device in devman.devices:
    if device.has("lighting_static"):
        device.fx.static(0, 255, 0)
    if device.has("lighting_backlight"):
        device.fx.misc.backlight.static(0, 255, 0)
```

THANKS

lah7, 2023-09-17:
Sorry, I wasn't clear. The new code replaced the last line of the original script, so you'll need this line after `import`:

    devman = openrazer.client.DeviceManager()

I amended the [earlier comment](https://github.com/openrazer/openrazer/issues/2066#issuecomment-1722241578).

0pLuS0, 2023-09-19:
Hi,

Ok so this is what I have now and it works good for the keyboard, but the mouse is lit up the same color green and I don't want the light on the mouse, I want it off.

What would I change to have the mouse light off?

THANKS

```
#!/usr/bin/python3
import openrazer.client
devman = openrazer.client.DeviceManager()

for device in devman.devices:
    if device.has("lighting_static"):
        device.fx.static(0, 255, 0)
    if device.has("lighting_backlight"):
        device.fx.misc.backlight.static(0, 255, 0)
```

lah7, 2023-09-19:
Sure, you use an `if` statement, like so:

```python3
#!/usr/bin/python3
import openrazer.client
devman = openrazer.client.DeviceManager()

for device in devman.devices:
    if "Basilisk" in device.name:
        device.fx.misc.logo.none()
        device.fx.misc.scroll_wheel.none()

    if "Huntsman" in device.name:
        device.fx.static(0, 255, 0)
```

"None" is a hardware effect. Other alternates could be to turn down the brightness or set static colour to black (0,0,0), but it should work.

In any case, the start-up script seems like a good workaround for now. Still a bit strange that software persistence doesn't work as expected with your setup (or just these particular devices).

0pLuS0, 2023-09-22:
Hi,

This is not working for the keyboard, the colors keep rotating.
````
#!/usr/bin/python3
import openrazer.client
devman = openrazer.client.DeviceManager()

for device in devman.devices:
    if "Basilisk" in device.name:
        device.fx.misc.logo.none()
        device.fx.misc.scroll_wheel.none()

    if "Huntsman" in device.name:
        device.fx.static(0, 255, 0)
```

Saw this also in the logs;

 File "/home/foo/.config/autostart/razer_startup.py", line 8, in <module>
    device.fx.misc.scroll_wheel.none()
AttributeError: 'NoneType' object has no attribute 'none'

THANKS

lah7, 2023-09-22:
Remove the `device.fx.misc.scroll_wheel.none()` line. When I pieced that code together, I could've swore the mouse had a `scroll_wheel` zone. It should work next time.

0pLuS0, 2023-09-22:
The log shows an AttributeError for the mouse, but the keyboard isn't working.

When I log into Openbox, the keyboard is on None not Static and it's not lit up.

THANKS

0pLuS0, 2023-09-24:
This is what I now have for the script, but the keyboard isn't working, it's cycling through the colors.

```
#!/usr/bin/python3
import openrazer.client
devman = openrazer.client.DeviceManager()

for device in devman.devices:
    if "Basilisk" in device.name:
		device.dpi = (400,400)
        device.fx.misc.logo.none()

    if "Huntsman" in device.name:
        device.fx.static(0, 255, 0)
```

THANKS

lah7, 2023-09-24:
It's likely not executing because Python is picky about indention. They need to be consistent (i.e. use 4 spaces, not mixing with tab indents), like line 7:

```diff
     if "Basilisk" in device.name:
-		device.dpi = (400,400)
+        device.dpi = (400,400)
         device.fx.misc.logo.none()
```

0pLuS0, 2023-09-25:
I don't have an extra indentation in the script, I guess that just happened when I pasted it here.

This is a screen shot of the script, and it's not working for the keyboard. hmm 

[image]

lah7, 2023-09-26:
If you run the script manually, does it run or are there errors?

Feels like we've come full circle to the original problem. I'm thinking either:

* By the time this script/daemon executes, one of the devices hasn't finished initialising, so it's not in the list.
* The hardware has a brief period of ignoring our commands after it's initialised. 

We can add something that must wait for 2 devices:

```python
#!/usr/bin/python3
import time
import openrazer.client
devman = openrazer.client.DeviceManager()

while len(devman.devices) < 2:
    time.sleep(0.5)
    devman = openrazer.client.DeviceManager()

for device in devman.devices:
    if "Basilisk" in device.name:
        device.dpi = (400,400)
        device.fx.misc.logo.none()

    if "Huntsman" in device.name:
        device.fx.static(0, 255, 0)
```

Let's see if that helps, if not, we'll add some more lines to repeat the command a few times. We could also do some logging to try and understand what's going on.

0pLuS0, 2023-09-28:
It looks like this last script is now working.

If it acts up again, I'll post here.

Thanks for everything, greatly appreciated! Keep up the great work!

0pLuS0, 2024-05-30:
Hi @lah7 

By the way, I'm a bit nervous you'll be frustrated again, I hope not. I don't believe I'm creating any end-user errors compiling openrazer from source, it's just that, with everything it has, it just seems like my keyboard the Huntsman V2 TKL 
doesn't want to persist with my Effect settings.

I just want you to know, I can certainly just use the startup script you created for me, and be perfectly happy. I'm only replying if there is an issue with openrazer to simply help you, by showing my support by making this report.

```I'm now using openrazer 3.8.0```

I wanted to come back to this, because hopefully you remember telling me a few days ago, when we were messing around with the DPI settings for my mouse, that openrazer should not need startup scripts.

I just noticed today, if I use Windows and shut down the computer from Windows, and then when I restart the computer, log into Slackware first, the ```Effect``` setting doesn't persist.

I put the Effect for the keyboard on Static with a green color, but it cycles instead through all the colors.

If I shut down the computer from Slackware, then restart the computer, and boot only back into Slackware the Effect then persists.

I have in the startup run command script etc/rc.d/rc.modules.local ```/sbin/modprobe razerkbd```

The ```~/.config/openrazer/persistence.conf``` has the keyboard listed.

```
[632031H11805501]
dpi_x = 800
dpi_y = 800
poll_rate = 1000
logo_active = True
logo_brightness = 0
logo_effect = none
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1

[PM2138F58600896]
poll_rate = 1000
backlight_active = True
backlight_brightness = 35
backlight_effect = static
backlight_colors = 0 255 0 0 255 255 0 0 255
backlight_speed = 1
backlight_wave_dir = 2
```

The ```~/.config/openrazer/razer.conf``` has ```restore_persistence = True```

```
[General]
# Verbose logging (logs debug messages - lotsa spam)
verbose_logging = False

[Startup]
# Set the sync effects flag to true so any assignment of effects will work across devices
sync_effects_enabled = True

# Turn off the devices when the systems screensaver kicks in
devices_off_on_screensaver = True

# Battery notifier
battery_notifier = True

# Battery notification frequency [s] (0 to disable)
battery_notifier_freq = 600

# Battery notifications appear when device reaches this percentage
battery_notifier_percent = 33

# Apply effects saved to disk when daemon starts
restore_persistence = True

THANKS

lah7, 2024-05-30:
It's quite normal for effects to be "forgotten" after booting into Windows with Razer Synapse installed (aka. dual boot).

Mostly because Razer Synapse sets the hardware to a **"custom"** effect _(like creating an effect in Polychromatic)_. Razer 🐍  emulates the effects in software (sent over USB, even if it's all just green) so that causes the device to default to spectrum on next power cycle since it doesn't save what was set in this "custom" effect state. On the other hand, OpenRazer told the hardware "hey, set static effect, green"

I suspect the first time your system loads `openrazer-daemon`, the keyboard isn't seen. Persistence (when enabled) _should_ execute as soon as it loads the device. It would be worth looking at the logs to check what is going on. 

    ~/.local/share/openrazer/razer.log 

Might be worth deleting it first before rebooting to test.

For all we know, the instructions are sent, but the hardware ignored it (see `dmesg | grep razer` for any timeout errors)

0pLuS0, 2024-05-31:
Hi @lah7 I hope you're doing better, keep healthy man! I really appreciate all your time too!

This is what's in the log for today.
```
2024-05-31 10:36:54 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 1475
2024-05-31 10:36:54 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-05-31 10:36:54 | razer                          | INFO     | Found device.0: 0003:1532:0065.0001
2024-05-31 10:36:54 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskEssential
2024-05-31 10:36:54 | razer                          | INFO     | Found device.1: 0003:1532:026B.0006
2024-05-31 10:36:54 | razer.device1                  | INFO     | Initialising device.1 RazerHuntsmanV2Tenkeyless
2024-05-31 10:36:54 | razer.device1                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-05-31 10:36:54 | razer                          | INFO     | Serving DBus
2024-05-31 10:37:11 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 1637
2024-05-31 10:37:11 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-05-31 10:37:11 | razer                          | INFO     | Found device.0: 0003:1532:0065.0001
2024-05-31 10:37:11 | razer.device0                  | INFO     | Initialising device.0 RazerBasiliskEssential
2024-05-31 10:37:11 | razer                          | INFO     | Found device.1: 0003:1532:026B.0006
2024-05-31 10:37:11 | razer.device1                  | INFO     | Initialising device.1 RazerHuntsmanV2Tenkeyless
2024-05-31 10:37:11 | razer.device1                  | INFO     | Putting device into driver mode. Daemon will handle special functionality
2024-05-31 10:37:11 | razer                          | INFO     | Serving DBus
```

So because of this Dual Boot issue, is something being looked into, on how to deal with this, that will later be implemented in a newer release/update, or will we always then need to use a startup script to work around this?

THANKS

lah7, 2024-05-31:
Thanks - I did forget one thing: Please could you enable verbose logging? This will print more information - specifically it should say what it's doing when it restores persistence.

In `~/.config/openrazer/razer.conf`:
```ini
[General]
verbose_logging = True
```

(Takes effect at next `openrazer-daemon` start)

The issue at hand isn't actively being at, since we don't know what the exact problem is. I suspect it's a firmware quirk.

0pLuS0, 2024-06-01:
Ok I'll set ```verbose_logging = True``` and when I shut down the box and restart it tomorrow I'll report back.

Today, when I logged into X, I thought I could just run ```openrazer-daemon -r``` it worked for the keyboard, but my mouse DPI changed from 800 to 18000. I just thought respawing the dameon, the settings in the config would still persist, not understanding why the mouse DPI changed running the cmd.

THANKS

0pLuS0, 2024-06-02:
With ```verbose_logging = True``` here's the log file.

THANKS

[razer.log.txt](https://github.com/user-attachments/files/15526853/razer.log.txt)

0pLuS0, 2024-06-06:
Hi @lah7 

The other issue, yes I had similar problems with a Baslisk mouse, but I'm using an Essential now, and this mouse's settings persist for some reason.

I have not changed keyboards since I've started these reports, and it's always been the keyboard not persisting between systems.

THANKS

0pLuS0, 2024-06-07:
Hi,

When I started the computer today and logged into Slack it didn't work. So I logged out of X and back and this time it didn't set itself either. So these changes also are effecting it from setting, when I log out and back into X, because before, I could log out and back and then it would finally change.

I'm attaching the patches for these changes.

THANKS

[Dont_want_to_restore_poll_rate_persistence.patch.txt](https://github.com/user-attachments/files/15745044/Dont_want_to_restore_poll_rate_persistence.patch.txt)
[Set_driver_mode_first.patch.txt](https://github.com/user-attachments/files/15745045/Set_driver_mode_first.patch.txt)

lah7, 2024-06-07:
I'm guessing the daemon _crashed_ because Python is particular about indentation.

❌ 
```
+    def __init__(self, *args, **kwargs):
+    self.set_device_mode(0x03, 0x00) # HACK: Set driver mode first
+    super().__init__(*args, **kwargs)
```

✅  (4 spaces)
```
+    def __init__(self, *args, **kwargs):
+        self.set_device_mode(0x03, 0x00) # HACK: Set driver mode first
+        super().__init__(*args, **kwargs)
```

0pLuS0, 2024-06-08:
I use Geany, and I just realized, when I copied the lines from here and pasted them, they made the spaces on their own, which I wasn't paying attention.

I'm not a coder, but I'm assuming being a Python script, Geany knew to apply the correct spacing?

Ok, I'll let ya know tomorrow how this works.

Here's the new patch...
[Set_driver_mode_first.patch.txt](https://github.com/user-attachments/files/15749811/Set_driver_mode_first.patch.txt)

THANKS

0pLuS0, 2024-06-09:
The changes with the spacing don't help, this doesn't work.

THANKS

lah7, 2024-06-10:
The last thing you could try is to undo those patches and activate persistence twice -- maybe setting the "static" hardware effect twice is what makes your keyboard's firmware behave after it's had an encounter with Razer Synapse on Windows?

    daemon/openrazer_daemon/hardware/device_base.py

```diff
         if self.config.getboolean('Startup', "restore_persistence") is True:
             self.restore_effect()
+            self.restore_effect()
```

Otherwise, the best solution we have for now is to stick with your custom start-up script, until we understand the quirks of Razer firmware a bit better.

0pLuS0, 2024-06-10:
Ok, I'll let you know tomorrow...

Here's the new patch
[activate_persistence_twice.patch.txt](https://github.com/user-attachments/files/15779990/activate_persistence_twice.patch.txt)

0pLuS0, 2024-06-11:
This works!

I hope this helps you to understand something now about this situation.

THANKS

lah7, 2024-07-02:
@0pLuS0 The next version of OpenRazer (or master branch) adds a new setting in `razer.conf` (#2262)

    [Startup]
    persistence_dual_boot_quirk = True

You'll need to add this to your `~/.config/openrazer/razer.conf` as it won't do so automatically. This applies the effect twice, as we saw fixed your issue after dual booting from Windows.

0pLuS0, 2024-07-11:
Hi @lah7 

Well the patch works, you sure I need this option?

THANKS

lah7, 2024-07-12:
The option and the patch are doing the same thing and touched the same code. 

It's just a heads up that local changes may be necessary for you in the next release (not released yet) since the patch won't apply.

0pLuS0, 2024-11-05:
Hi @lah7 

I just updated to 3.9.0 and the keyboard is not persisting. I guess I was assuming it was going to be fixed in this update.

But the mouse seems to be working ok without using the patch. 

I guess I'll have to use the activate_persistence_twice.patch for the keyboard.

THANKS

lah7, 2024-11-05:
The patch can be dropped for your local installation. [v3.9.0](https://github.com/openrazer/openrazer/releases/tag/v3.9.0) added the config option in `~/.config/openrazer/razer.conf`, so you just need to add there:

```diff
 [Startup]
 restore_persistence = True
+persistence_dual_boot_quirk = True
```

0pLuS0, 2024-11-07:
> The patch can be dropped for your local installation. [v3.9.0](https://github.com/openrazer/openrazer/releases/tag/v3.9.0) added the config option in `~/.config/openrazer/razer.conf`, so you just [...]

Ahhh this is great, really appreciated!

Keep up the great work!

THANKS

0pLuS0, 2025-03-30:
@lah7 

I am still having problems with this. It seemed like it was working for a while, then it stopped.

I am also now using 3.10.1, and the only way I can make this work, once I log into my DE, I need to log out and log back in.

I also have in my razer.conf these settings, but they're not helping.

restore_persistence = True
persistence_dual_boot_quirk = True

THANKS

lah7, 2025-03-31:
We recently discovered that the BlackWidow V4 Pro didn't fully apply settings at login, because "driver mode" was activated afterwards, but it worked when driver mode is set before applying any persistence. It would be worth testing that theory.

* https://github.com/openrazer/openrazer/pull/2416

For your install:

1. [Apply the patch from that PR.](https://patch-diff.githubusercontent.com/raw/openrazer/openrazer/pull/2416.patch)
2. Find your affected device in `openrazer_daemon/hardware/keyboards.py` which I believe is `RazerHuntsmanV2Tenkeyless`
3. Add the variable `EARLY_DEVICE_MODE = True`

Then, see if that works at next boot. If you have `verbose_logging = True`, you'll know the code is working if you see `Setting device to "driver" mode` before any persistence is applied.

0pLuS0, 2025-03-31:
Hello @lah7 

Before I forget...

I don't recall if I mentioned this before, but this seems to be an issue if I shutdown the computer from Windows, and then later start the computer and boot straight into Slack, and then I have to log out of X and log back into for it to work, then once I get it working like this, if I boot into Windows, and then boot back to Slack it's ok.

It only seems to be an issue shutting down the computer in Windows and then restarting, and then booting straight into Slack.

0pLuS0, 2025-04-02:
Reboot City, here we come, get the fire extinguisher ready! LOL 🤪 

1. Device lights off - Yes
2. Restarting Windows to Slack same problem  - No
3. Yes, if I power off from Slack
4. a. Grub Halt - No 
4. b. Shut down Bios - No
5. Before X starts replug - No

I tried the patch, I'm assuming this last one you mentioned;

https://patch-diff.githubusercontent.com/raw/openrazer/openrazer/pull/2416.patch

EARLY_DEVICE_MODE = True is in the config...

The patch didn't work, and I see in the razer log it says;

Setting device to "driver" mode. Daemon will handle special functionality.

THANKS

lah7, 2025-04-02:
To echo the findings:

* 👍🏼 **No problem:**
   * From complete power off.
   * Restarting from Windows to Slack.
   * Shutting down from Slack.
   * Replugging device after X11 starts.
* 👎🏼 **Persistence fails:**
   * Shutting down from Windows.
   * Shutting down from GRUB/BIOS.
   * Replugging device while booting before X11 starts.

And by "shut down" we mean the system is powered off (S5 state), but still plugged in to the wall.
And "power off" meaning the PSU switch (assuming desktop) is turned off or off/unplugged at wall.

I can't see any obvious pattern, only that it behaves as expected if Linux was the last one to initiate the shutdown. If the above is wrong, or an observation changes, please correct me.

----

Thanks for testing the patch. Assuming you did follow the [3 steps _to a T_](https://github.com/openrazer/openrazer/issues/2066#issuecomment-2766267919), then the order of when device mode is triggered seems irrelevant. There doesn't appear to be anything else we can do with code, besides increasing time after `persistence_dual_boot_quirk`.

I'd try enabling power saving features in the BIOS, to be sure that the mouse is completely "off" when your system is shut down.... unless you have 'press USB device to power up' enabled. It might be under "ErP Ready" or ACPI/USB. Your motherboard's manual might have some tips. Assuming this is a desktop computer?

And is this keyboard connected directly to the motherboard? Nothing in the middle like a KVM switch or USB hub?

0pLuS0, 2025-04-03:
And by "shut down" we mean the system is powered off (S5 state), but still plugged in to the wall.
And "power off" meaning the PSU switch (assuming desktop) is turned off or off/unplugged at wall.

```Yes PSU button is off and still plugged into the wall. When I shutdown from the OS, I then always press the PSU button off.```

I can't see any obvious pattern, only that it behaves as expected if Linux was the last one to initiate the shutdown. If the above is wrong, or an observation changes, please correct me.

```Yes it behaves as expected if Linux was the last one to  initiate the shutdown```.

Following the 3 Steps to a T, you mean the Reboots I just did?

Yes this is a desktop, I built it, so if you have the time you might want to look at the specs on it, and see what you find out as well. 

```Asus B550-F Gaming``` (I have the Non-WiFi version)
https://rog.asus.com/motherboards/rog-strix/rog-strix-b550-f-gaming-model/

Oh, I see a new BIOS just came out yesterday  Version 3621, but on April 1st LOL, hopefully doesn't blow up the PC. I just updated the other day to  Version 3611

https://rog.asus.com/motherboards/rog-strix/rog-strix-b550-f-gaming-model/helpdesk_bios/

Keyboard is connected directly to the motherboard.

THANKS

[29 more replies]

commit 7f830cb2 (2024-07-02): Add "persistence_dual_boot_quirk" config

It seems a handful of newer devices' have quirky firmware behaviour
after having an encounter with Razer Synapse (for users dual booting
with Windows) when restoring the persisted effect.

Even though we send the request and got no error, the device physically
didn't change its lighting, but it'll work by sending a second time.

Up to now, the user needs to go into a frontend to manually re-apply
their effect. Should the user run into this scenario, they can turn this
option on for a quality of life "now it just works".

Fixes #1807 (Basilisk V3)
Fixes [... cut]
PR #2262 (2024-07-02): Add `persistence_dual_boot_quirk` to razer.conf
It seems a handful of newer devices' have quirky firmware behaviour after having an encounter with Razer Synapse (for users dual booting with Windows) when restoring the persisted effect. Even though we send the request and got no error, the device physically didn't change its lighting, but it'll work by sending a second time.

Up to now, the user needs to go into a frontend to manually re-apply their effect. Should they run into this scenario, they can now turn this option on for a quality of life _"now it just works!"_.

Fixes #1807 (Basilisk V [... cut]
closed 2025-04-11 by lah7