openrazer/openrazer #2067: spoofing id to the X Version?

Noahbx, 2023-05-11:
Hey I Have A Razer Ornata V3 (not the "Razer Ornata V3 X") and im wondering if i could spoof the id to the "X" version and see if it would work, and how would i go about spoofing it?
Also is there any plans to add support for the  Standard Ornata V3?

z3ntu, 2023-05-11:
Please share `lsusb | grep 1532` otherwise I don't know which device you're talking about.

But generally no you cannot, but you can add support for it which maybe shouldn't be more difficult than doing what you want to do. See https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

CrisisModel, 2023-05-14:
> Please share `lsusb | grep 1532` otherwise I don't know which device you're talking about.

Not OP, but I have the same keyboard. The result of `lsusb | grep 1532` is:
`Bus 003 Device 006: ID 1532:02a1 Razer USA, Ltd Razer Ornata V3`

z3ntu, 2023-05-15:
@CrisisModel For your device see https://github.com/openrazer/openrazer/issues/2053

closed 2023-10-13 by github-actions[bot]