openrazer/openrazer #2069: Support for [device name here]

kreezxil, 2023-05-12:
### Device Info
[image]

* Name: Razer Hyperion V3 X Hyperspeed
* Product Number: RZ01-04870100-0000
* Model Number: RZ01-0487

### Device Information
```
Bus 001 Device 007: ID 1532:00b9 Razer USA, Ltd Razer Basilisk V3 X HyperSpeed
```

```
total 0
drwxr-xr-x 2 root root 320 May 12 04:10 .
drwxr-xr-x 4 root root 900 May 12 04:10 ..
lrwxrwxrwx 1 root root   9 May 12 03:54 usb-Acrox_GK500_Gaming_keyboard-event-if00 -> ../event6
lrwxrwxrwx 1 root root   9 May 12 03:54 usb-Acrox_GK500_Gaming_keyboard-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 May 12 03:54 usb-Acrox_GK500_Gaming_keyboard-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 May 12 03:54 usb-Acrox_GK500_Gaming_keyboard-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 May 12 03:54 usb-Generic_USB2.0_PC_CAMERA-event-if00 -> ../event31
lrwxrwxrwx 1 root root  10 May 12 03:54 usb-Logitech_G533_Gaming_Headset-event-if03 -> ../event21
lrwxrwxrwx 1 root root  10 May 12 03:54 usb-MSI_MYSTIC_LIGHT_A02021090806-event-if00 -> ../event30
lrwxrwxrwx 1 root root  11 May 12 04:10 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-event-if01 -> ../event264
lrwxrwxrwx 1 root root  11 May 12 04:10 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-event-mouse -> ../event261
lrwxrwxrwx 1 root root  11 May 12 04:10 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if01-event-kbd -> ../event262
lrwxrwxrwx 1 root root  11 May 12 04:10 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if01-event-mouse -> ../event263
lrwxrwxrwx 1 root root   9 May 12 04:10 usb-Razer_Razer_Basilisk_V3_X_HyperSpeed_000000000000-if01-mouse -> ../mouse3

```

<details>

```
001:007:002:DESCRIPTOR         1683883779.897586
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:007:001:DESCRIPTOR         1683883779.904902
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:007:000:DESCRIPTOR         1683883779.908746
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 06 00 FF 05 0C 0A 38 02 15 81 25 7F 75 08 95 01
 81 06 05 01 09 38 15 81 25 7F 75 08 95 01 81 06
 09 30 09 31 16 01 80 26 FF 7F 75 10 95 02 81 06
 C0 06 00 FF 09 02 15 00 25 01 75 08 95 5A B1 01
 C0

```

</details>

Switching to windows now for this data.
### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

kreezxil, 2023-08-20:
I responded to you guys, said the new mouse was working great, no unwanted
double clicks.

bakicelebi, 2023-08-20:
> I responded to you guys, said the new mouse was working great, no unwanted double clicks.
> […](#)

My hyprshift button and mouse button 5 are not registering correctly. How did you manage to get it working correctly?

kreezxil, 2023-08-20:
nevermind, i thought you were the razer company, i didn't pay attention to
the email. i need to delete what i wrote. note to self: pay attention more.

NeroReflex, 2023-09-04:
Hello, I have this mouse too. I can get a wireshark USB capture on windows if that is still needed.

z3ntu, 2023-09-04:
@NeroReflex Yes, please: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

NeroReflex, 2023-09-04:
> @NeroReflex Yes, please: https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol

Fine for me. The device is actually my brother's and I was told this is a bluetooth model. Shall I do that with a USB bluetooth dongle? I am not award of any bundled USB. So I assume it's bluetooth only....

NeroReflex, 2023-09-05:
[capture.zip](https://github.com/openrazer/openrazer/files/12527100/capture.zip)
At the begin there are a lot of brightness changes, from 100 to 0. then the last command should be another brightness change from 0. Those in the middle are all lighting effects that hopefully you understand. I did one with multiple colors, others were with the default green color. There are a few commands light modes that never stopped sending data (unlike static) so I left those run for a few seconds.
I hope there is everything you need, in case something is missing please let me know.

P.S. I had to use wireshark as root. I'm sorry.

NeroReflex, 2023-09-05:
There was in fact a USB coming in the bundle and that's what I have been using. This mouse however supporta bluetooth too

NeroReflex, 2023-09-30:
Did someone looked to my pcap? Is another one needed or is this good?

serious-bobo, 2023-10-04:
I also have the similar mouse. Can I help with something?

z3ntu, 2023-10-14:
Yeah, packet captures look good! Transaction ID used is 0x1f, brightness is using the extended matrix brightness calls with ZERO_LED, dpi stages, battery level, etc. look normal

See also https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md on how to add support for it :)

LN-Dark, 2023-11-03:
hi, i have the same mouse 1532:00B9, Basilisk v3 X HyperSpeed, is there a chance to have support to it? Thanks for your work :)

Ann0ying, 2023-11-07:
I have the same mouse as well, let me know if any additional data needed.

taylor6655, 2023-11-09:
I'm waiting for support on this device as well. let me know if you require any device information and ill see what I can do.

bakicelebi, 2023-12-14:
Why is this issue closed?

z3ntu, 2023-12-14:
@bakicelebi There's a big green badge on the top that says Open so this issue is open, not closed.

taylor6655, 2023-12-14:
@bakicelebi you may be confused with another open issue involving this mouse which was closed for being a duplicate issue.

bakicelebi, 2023-12-14:
Yes @taylor6655 😄 .

mandofrog, 2024-01-02:
Same mouse here, Basilisk V3 X Hyperspeed.  Device 1532:00B9.  USB or Bluetooth.  It can't be that much different the Basilisk V3 Hyperspeed or the Basilisk X Hyperspeed, can it?  Waiting on device support.
-Randy

PiersonCourtois, 2024-01-06:
same mouse, would love support for it!

edgarogh, 2024-01-06:
Sorry for being the annoying guy but please refrain from sending messages such as "I'm interested too". You can add a thumbs up on the original issue if you like it, or subscribe to notifications using the button on the right like I'm doing, if that's why you're commenting.

Sending these messages creates spam e-mail to all subscribers while adding nothing substantial to the discussion.

(no hard feelings 😉)

strycore, 2024-02-25:
Speaking of sending email to every subscriber of this ticket, there is a working PR open to add support for the Basilisk V3 X Hyperspeed.

z3ntu, 2024-03-03:
Fixed by #2191 / 479a8f330bf126abee6c54ca9ccfd3af7f4e2ed3

closed 2024-03-03 by z3ntu