openrazer/openrazer #2070: Huntsman v2 TKL - Effects Don't Change In Polychromatic

0pLuS0, 2023-05-12:
### Distribution

Ubuntu 23.04
openrazer 3.6.0
polychromatic 0.8.1

### Support

- [ x] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I assumed when clicking on the different Effects in polychromatic the changes would be immediate, because with my Baslisk Essential mouse, all the effects work with I click on them, the change is immediate. So I'm assuming a lack of support issue here.

None of the effects work, except Static for the Huntsman v2 TKL, rebooting doesn't do anything. Static is the only one that has ever worked.

The only thing I can find in the logs, when I look for any errors is this;

```razerkbd: module verification failed: signature and/or required key missing - tainting kernel```

What is this sig or key missing, is this something I can fix on my end?

Here's the log file attached.

[razer.log](https://github.com/openrazer/openrazer/files/11468173/razer.log)

z3ntu, 2023-05-13:
@saibotk @rbelando @lwndow I'm guessing on your Huntsman setting the effects work correctly? If so it's possible that the devices are running a different firmware version.

@0pLuS0 Could you share the firmware version of your device please?

0pLuS0, 2023-05-14:
I just logged onto Ubuntu today, after a few days away, strange thing, there was an openrazer update, now everything works... LOL

closed 2023-05-15 by z3ntu

0pLuS0, 2023-06-25:
Firmware version 1.0 for the keyboard....

0pLuS0, 2023-06-28:
Might as well just follow up on everything in this Post.

https://github.com/openrazer/openrazer/issues/2066#issuecomment-1610591621

THANKS