openrazer/openrazer #2074: Deathstalker V2 TKL razerkbd driver breaks basic functionality

stlhrt, 2023-05-17:
### Distribution
Distributor ID:	Pop
Description:	Pop!_OS 22.04 LTS
Release:	22.04
Codename:	jammy

openrazer 3.6.1

### Support

After installing updated openrazer supporting Deathstalker V2 TKL  fn+ keys stop working correctly.

- Brightness down toggles microphone
- Brighness up does nothing visible
- Printscreen does nothing visible

Haven't tried others and investigated further yet.

z3ntu, 2023-05-29:
> Brightness down toggles microphone

Known bug unfortunately. The keycode we use for driver-daemon communication conflicts with XF86MicMute

> Brighness up does nothing visible

This should tell the daemon to increase backlight brightness. Is there anything in the daemon logs when you press this? Also consider starting openrazer-daemon with the `-v` flag for verbose.

> Printscreen does nothing visible

`libinput debug-events` also says nothing happens? On my (non-Razer) laptop keyboard PrintScr is ` event3   KEYBOARD_KEY            +2.009s       KEY_SYSRQ (99) pressed`

ArDark2, 2023-06-01:
@stlhrt  Hi, ive created PR for this one, but closed because cant find how to resolve bindings multimedia key ( mute, playpause, prev and next track).  But all other is working on my DS v2 pro tkl on ubuntu 22.04 (prnscn, scrlk, pause, sleep)
Here is PR https://github.com/openrazer/openrazer/pull/2078
there is working FN + F10-12 but you need bind keys to use it. 
I suppose you can check it with:
`sudo libinput debug-events --show-keycodes`2222

closed 2023-10-30 by github-actions[bot]

loafylemon, 2024-06-04:
Any chance the PR can be merged? It's still very much an unresolved issue.

z3ntu, 2024-06-05:
@loafylemon That PR is closed so it's not considered for merging. If you're interested in it, can you open a new one and revive it?

loafylemon, 2024-06-05:
Done. Please let me know if there's anything I can do. Thanks!