openrazer/openrazer #2079: can not rebuild after update from python 3.10. to 3.11

hanny00, 2023-06-04:
~ ❯❯❯ pacman -Qoq /usr/lib/python3.10/ openrazer-daemon python-openrazer

~ ❯❯❯ pamac build $(pacman -Qoq /usr/lib/python3.10)
Vorbereitung...
Überprüfe python-openrazer Abhängigkeiten...
Überprüfe openrazer-daemon Abhängigkeiten...
Warnung: python-openrazer-3.6.1-1 ist aktuell -- Reinstalliere
Warnung: openrazer-daemon-3.6.1-1 ist aktuell -- Reinstalliere
Abhängigkeiten werden aufgelöst...
Interne Konflikte werden überprüft...

Zu erstellen (2):
  openrazer-daemon  3.6.1-1             AUR
  python-openrazer  3.6.1-1             AUR

Build-Dateien bearbeiten : [e] 
Transaktion anwenden ? [e/j/N] j

** (pamac:14060): WARNING **: 15:13:00.971: transaction.vala:1026: Fehler beim Öffnen der Datei »/var/tmp/pamac-build-marika/openrazer/.SRCINFO«: Datei oder Verzeichnis nicht gefunden

lah7, 2023-06-04:
Not familiar with `pamac` myself, but a clean build is essential.

Possibly this: <sup>[[1]](https://linuxcommandlibrary.com/man/pamac)</sup>

    pamac clean --build-files
    pamac build openrazer-daemon
    pamac build python-openrazer

hanny00, 2023-06-05:
Thank you for replay and help. I did the steps, but that does not help.
Then i delete the /var/tmp/openrazer folder by myself and that startet a clean install. All seems to be working now.
Thank you for your super Tools for Razer hardware

closed 2023-06-05 by hanny00