openrazer/openrazer #2080: Support for Razer Kraken X USB

ornelasnarciso, 2023-06-04:
### Device Info

* Name: Razer USA, Ltd Razer Kraken X USB
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
    
Bus 002 Device 004: ID 1532:0526 Razer USA, Ltd Razer Kraken X USB
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0526 
  bcdDevice            0.34
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken X USB
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x00fa
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x004f
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               0 
[... 356 lines cut ...]
lrwxrwxrwx 1 root root   9 jun  4 15:52 usb-Wacom_Co._Ltd._CTL-460-mouse -> ../mouse1

```

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed

libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/002/004, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
Failed to find and open the devices: Access denied (insufficient permissions)

```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2023-06-05:
Duplicate of #1490

closed 2023-06-05 by z3ntu

ornelasnarciso, 2023-06-05:
So, no support?  And thanks for your effort, works well with my keyboard. :)

ornelasnarciso, 2023-06-05:
It's out of my field of knowledge but I would definitely help if I could.  Thanks.