openrazer/openrazer #2082: Signing and installing Openrazer kernel modules on Ubuntu 23.04 for Huntsman V2 TKL

ppaulis, 2023-06-10:
### Distribution

Ubuntu 23.04

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm trying to use Polychromatic on Ubuntu 23.04 with my Huntsman V2 TKL to enable lighting. Polychromatic tells me :
[image]

I have Secure Boot enabled and really would like to keep it that way. So I'm trying to sign the openrazer modules. This is what I have done so far :

Generating and importing new key :
```
openssl req -new -x509 -newkey rsa:2048 -keyout /var/lib/shim-signed/mok/RAZER.priv -outform DER -out /var/lib/shim-signed/mok/RAZER.der -nodes -days 36500 -subj "/CN=PPAULIS/"
mokutil --import /var/lib/shim-signed/mok/RAZER.der
```

Reboot and then I sign the modules :
```
sudo kmodsign sha512 /var/lib/shim-signed/mok/RAZER.priv /var/lib/shim-signed/mok/RAZER.der /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkbd.ko
sudo kmodsign sha512 /var/lib/shim-signed/mok/RAZER.priv /var/lib/shim-signed/mok/RAZER.der /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razeraccessory.ko
sudo kmodsign sha512 /var/lib/shim-signed/mok/RAZER.priv /var/lib/shim-signed/mok/RAZER.der /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkraken.ko
sudo kmodsign sha512 /var/lib/shim-signed/mok/RAZER.priv /var/lib/shim-signed/mok/RAZER.der /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razermouse.ko
```

Installing the module :
```
sudo insmod /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkbd.ko
sudo insmod /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razeraccessory.ko
sudo insmod /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkraken.ko
sudo insmod /var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razermouse.ko
```

`lsmod` then shows that the 4 modules are loaded.

Then I try to refresh Polychromatic. But the Keyboard remains unrecognized by the app.

I tried to load the modules at boot time by adding :

```
razerkbd
razeraccessory
razerkraken
razermouse
```

to the file `/etc/modules-load.d/modules.conf` but that doesn't work.

(btw, all above commands run without errors)

If somebody could tell me what I'm doing wrong in my procedure here, that would be great!

Thank you,
best regards,
Pascal

z3ntu, 2023-06-10:
What version of openrazer do you have installed? Check e.g. `openrazer-daemon --version`

ppaulis, 2023-06-12:
@z3ntu 3.6.1
Sorry for that, should have mentionned it in the original post!

Also forgot to mention, that I added my user to the `plugdev` group, as the setup suggests:
```
sudo gpasswd -a $USER plugdev
```

z3ntu, 2023-06-12:
Anything interesting in `sudo journalctl -b | grep -i razer_mount` ?

The device doesn't show up in `ls -al /sys/bus/hid/drivers/razer*/` ?

ppaulis, 2023-06-12:
@z3ntu 
```
ppaulis@ppaulis-ubuntu-desktop:~$ sudo journalctl -b | grep -i razer_mount
ppaulis@ppaulis-ubuntu-desktop:~$
```
and the second command, before I run the `insmod` commands :
```
ppaulis@ppaulis-ubuntu-desktop:~$ ls -al /sys/bus/hid/drivers/
insgesamt 0
drwxr-xr-x 3 root root 0 Jun 12  2023 .
drwxr-xr-x 4 root root 0 Jun 12  2023 ..
drwxr-xr-x 2 root root 0 Jun 12  2023 hid-generic
```
and after I run the `insmod` commands :
```
ppaulis@ppaulis-ubuntu-desktop:~$ ls -al /sys/bus/hid/drivers/razer*/
/sys/bus/hid/drivers/razeraccessory/:
insgesamt 0
drwxr-xr-x 2 root root    0 Jun 12 21:04 .
drwxr-xr-x 7 root root    0 Jun 12  2023 ..
--w------- 1 root root 4096 Jun 12 21:07 bind
lrwxrwxrwx 1 root root    0 Jun 12 21:07 module -> ../../../../module/razeraccessory
--w------- 1 root root 4096 Jun 12 21:07 new_id
--w------- 1 root root 4096 Jun 12 21:04 uevent
--w------- 1 root root 4096 Jun 12 21:07 unbind

/sys/bus/hid/drivers/razerkbd/:
insgesamt 0
drwxr-xr-x 2 root root    0 Jun 12 21:04 .
drwxr-xr-x 7 root root    0 Jun 12  2023 ..
--w------- 1 root root 4096 Jun 12 21:07 bind
lrwxrwxrwx 1 root root    0 Jun 12 21:07 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Jun 12 21:07 new_id
--w------- 1 root root 4096 Jun 12 21:04 uevent
--w------- 1 root root 4096 Jun 12 21:07 unbind

/sys/bus/hid/drivers/razerkraken/:
insgesamt 0
drwxr-xr-x 2 root root    0 Jun 12 21:04 .
drwxr-xr-x 7 root root    0 Jun 12  2023 ..
--w------- 1 root root 4096 Jun 12 21:07 bind
lrwxrwxrwx 1 root root    0 Jun 12 21:07 module -> ../../../../module/razerkraken
--w------- 1 root root 4096 Jun 12 21:07 new_id
--w------- 1 root root 4096 Jun 12 21:04 uevent
--w------- 1 root root 4096 Jun 12 21:07 unbind

/sys/bus/hid/drivers/razermouse/:
insgesamt 0
drwxr-xr-x 2 root root    0 Jun 12 21:04 .
drwxr-xr-x 7 root root    0 Jun 12  2023 ..
--w------- 1 root root 4096 Jun 12 21:07 bind
lrwxrwxrwx 1 root root    0 Jun 12 21:07 module -> ../../../../module/razermouse
--w------- 1 root root 4096 Jun 12 21:07 new_id
--w------- 1 root root 4096 Jun 12 21:04 uevent
--w------- 1 root root 4096 Jun 12 21:07 unbind
```

After hitting the "Refresh" Option in Polychromatic, the device still remains unrecognized. I'll try now to figure out how to load those modules at boot time. Perhaps they need to be loaded before the daemon starts.

ppaulis, 2023-06-12:
@z3ntu I reinstalled & re-signed everything.
Installing the modules using `insmod` and then de-/re-connecting the keyboard does the trick.

So I guess there is a good chance that loading the modules at boot time will fix the problem.

I tried adding them to `/etc/modules` in different ways :
```
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkbd
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razeraccessory
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkraken
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razermouse
```
or :
```
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkbd.ko
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razeraccessory.ko
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razerkraken.ko
/var/lib/dkms/openrazer-driver/kernel-6.2.0-20-generic-x86_64/module/razermouse.ko
```
or simply :
```
razerkbd
razeraccessory
razerkraken
razermouse
```

Nothing seems to work. I'm probably doing it wrong :thinking:

z3ntu, 2023-06-13:
When plugging in the device (or booting up for that matter) the udev rules should trigger https://github.com/openrazer/openrazer/blob/master/install_files/udev/99-razer.rules and run the razer_mount script https://github.com/openrazer/openrazer/blob/master/install_files/udev/razer_mount which loads the driver (modprobe) and rebinds them.

Seems this is not running for you for whatever reason, since there's no razer_mount output in journalctl.

TriMoon, 2023-08-01:
Just to add my 2 cents to this issue:

I'm running `openrazer-daemon 3.6.1` on Kubuntu 23.04 using the packages from the original repo's of here and polychromatic, see: https://github.com/openrazer/openrazer/issues/2101

And i'm using SecureBoot also in combination with DKMS modules..

closed 2023-12-31 by github-actions[bot]