openrazer/openrazer #2084: Hunstman ELite : (Un)Mute key not working

audetto, 2023-06-18:
### Distribution

Ubuntu 23.04

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Mute-unmute key does not work.
And by that I mean it does not arrive to `showkey`, let alone any GNOME shortcut.

Haven't seen anything in `dmesg`.

The rest seems to work fine.

System info

`Bus 003 Device 008: ID 1532:0226 Razer USA, Ltd Huntsman Elite`

```
andrea@ubuntu:~$ uname -r
6.2.0-23-generic
andrea@ubuntu:~$ sudo dkms status
openrazer-driver/3.6.1, 6.2.0-20-generic, x86_64: installed
openrazer-driver/3.6.1, 6.2.0-23-generic, x86_64: installed
andrea@ubuntu:~$ sudo modprobe razerkbd
andrea@ubuntu:~$ 
```

audetto, 2023-06-22:
No Fn needed, these are the multimedia keys at the top right of the keyboard.

Very similar to the one in this picture

https://www.amazon.co.uk/Razer-Huntsman-Keyboard-Opto-Mechanical-UK-Layout/dp/B07D21BX4C

There are 6 keys:

3 keys work, `showkey` shows 163 164 165
a wheel: 114 115
a button at the centre of the wheel (supposedly mute): no show.

> Did it work without openrazer installed? 

what do I remove exactly?

z3ntu, 2023-06-25:
> what do I remove exactly?

Depends on the distro, but generally anything in the package list containing "openrazer".

audetto, 2023-06-26:
So, I have booted on Finnix 124 using kernel 5.16.0 and it made no difference.
Only 5 out of 6 key codes can be generated.

z3ntu, 2023-07-16:
You could look into the razer_event / razer_raw_event functions and see if you get any custom data event from the keyboard when you do this. If that's the case you could add some code doing proper key translation to the openrazer code. That's kind of all the suggestions I have right now.

audetto, 2023-07-16:
Are we looking for a **kernel** issue, or an **openrazer** issue?
In the end, it is just a keyboard and standard keypresses should just work regardless? Or am I wrong?

Since it did not work with Finnix (which definitely does not have any openrazer code running), it is either a non-standard keyboard event, or an hardware issue?

Would a USB logger help?

z3ntu, 2023-07-17:
If no key event shows up without openrazer installed, then it's the firmware not sending a key event when the device is in the "device mode" (which is active then, or in BIOS, or wherever). Normally when we have openrazer support we switch the device to "driver mode" where we get different events (razer_event / razer_raw_event as mentioned above) which we need to translate ourselves into proper key events or whatever we want to do. In this mode we should be able to handle all of the keys and weird key combinations (in these two event methods). In device mode we get some stuff from the firmware but not more.

closed 2023-12-15 by github-actions[bot]

notmayo, 2025-12-06:
I am still having this issue. I'm not sure if this is a device limitation or software? I have tried troubleshooting this with several key identifying softwares? (sorry, not sure what to call them) and none identify a key press like any other HID key presses.

Sorry for the necro post, I would really like to have my mute button. If not, this keyboard is going to have to be put away, sold, or given away.

z3ntu, 2026-01-15:
@notmayo You're also talking about the Huntsman Elite device or a different one?