openrazer/openrazer #2087: Razer blade 14 2023

mahdi65, 2023-06-22:
### Distribution

Ubuntu 20.04 5.15.0-75-generic
### Support

Speakers are not working/ detected. I have tried pulseaudio solution with no success. 
OpenRazer doesn't support the model.
Multiple ACPI BIOS error on boot screen.

pramodhrachuri, 2023-06-26:
For adding support to your laptop, I think you can try adding your laptop's VID and PID and check if everything works. Here is an example of what to add.
https://github.com/openrazer/openrazer/commit/988668ecf55613e0818c9d539d4aea30bc30737e

As far as I know, openrazer doesn't do anything for sound or mic. I recommend installing a newer version of Ubuntu.

z3ntu, 2023-07-16:
> Speakers are not working/ detected. I have tried pulseaudio solution with no success.
> Multiple ACPI BIOS error on boot screen.

Won't be resolved through openrazer support.

> OpenRazer doesn't support the model.

Please provide `lsusb | grep 1532` output.

cameronbosch, 2023-07-16:
Hi, I am using EndeavourOS (Arch-based with the stock Arch repos) on the same 2023 Razer Blade 14, and while I have the same audio issues on the latest kernels (with no success getting the speakers to work), I am aware that that is not what this project is for.

Do you guys need any help with getting this laptop to work for OpenRazer?

BTW, the output of `lsusb | grep 1532` is the following:

`Bus 001 Device 002: ID 1532:029d Razer USA, Ltd Razer Blade`

z3ntu, 2023-07-17:
Okay that's a new device.

Can you please fill out the template found at https://github.com/openrazer/openrazer/issues/new?assignees=&labels=Device+Support&projects=&template=new-device-support.md&title=Support+for+%5Bdevice+name+here%5D ? You can copy-paste the content here, that's fine.

Other than that, normally adding Blade laptops should be quite simple, see https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

m31317015, 2023-10-13:
Hello, may I ask if Blade 14 2023 is being worked on? I see the other 2023 blades being added to master branch two months ago. Just asking, thanks for your work!

quentingigon, 2023-10-30:
Hello, I also would like to see support for the 2023 model. I filled the template you previously linked and will past it here:

### Device Info

* Name: Razer Blade 14 2023
* Product Number: Blade 14 - RZ09-0482
* Model Number: 9.04

### Device Information

1532:029d

lrwxrwxrwx 1 root root  10 Oct 30 10:55 usb-Razer_Razer_Blade-event-if01 -> ../event12
lrwxrwxrwx 1 root root  10 Oct 30 10:55 usb-Razer_Razer_Blade-event-kbd -> ../event10
lrwxrwxrwx 1 root root  10 Oct 30 10:55 usb-Razer_Razer_Blade-if01-event-kbd -> ../event11
lrwxrwxrwx 1 root root  10 Oct 30 10:55 usb-Razer_Razer_Blade-if02-event-mouse -> ../event13
lrwxrwxrwx 1 root root   9 Oct 30 10:55 usb-Razer_Razer_Blade-if02-mouse -> ../mouse3

001:003:002:DESCRIPTOR         1698664046.547943
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:003:001:DESCRIPTOR         1698664046.552540
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:003:000:DESCRIPTOR         1698664046.553993
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

quentingigon, 2023-12-18:
Hello, just want to know if this is being worked on, or if I can provide something more to make the process faster ?

z3ntu, 2023-12-28:
@quentingigon You can check https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md and copy-paste a bit and with a bit of luck and trial&error you can get device support working.

matchbook-o, 2024-02-18:
Hi all, any word on this? My Razer Blade 14 2022 received support around this time last year. Let me know if you need any info on the device and I can contribute!

pramodhrachuri, 2024-02-20:
I added the support by just adding the Blade 14 2022's USB IDs to some files. I compiled it, and it worked. Maybe that's all you'll need for the 2023 version as well. Look at the following commit, try to make similar changes, compile it and test it. If it works, then make a pull request.
https://github.com/openrazer/openrazer/commit/988668ecf55613e0818c9d539d4aea30bc30737e

Support for the 2021's version was added in the exact same way in this commit. I do not see why it won't work this time too.
https://github.com/openrazer/openrazer/commit/b91dbee1f66a29bc5bd8be3d21396eef25d1cf67

This page has instructions on how to compile.
https://github.com/openrazer/openrazer/wiki/Building-a-package

PR #2237 (2024-04-25): Add support for Razer Blade 14 (2023)
Changes based on https://github.com/openrazer/openrazer/pull/2092. I mainly replaced instances of `029E` with `029D` and grabbed an image from the support website.

```bash
$ lsusb | grep '1532:'
Bus 001 Device 002: ID 1532:029d Razer USA, Ltd Razer Blade
```

I've signed and installed the driver on my own Razer Blade 14 2023 running Fedora. I've been able to change my keyboard and logo backlight behavior with RazerGenie, and the changes have persisted through multiple reboots.

Fixes #2087
closed 2024-04-25 by z3ntu