openrazer/openrazer #2089: Support for Razer Viper Mini SE

fireflame90051, 2023-06-28:
### Device Info

* Name: Razer Viper Mini SE
* Product Number: RZ01-04410100
* Model Number: RZ01-0441

Something to note, this device did not come with the standard razer mini dongle, just a HyperPolling wireless dongle that seems to be a one off for this specific device.

I'm not much of a programmer, but if there anything I can do to help add this mouse to your project, please let me know.

### Device Pictures
[image]
[image]

### Device Information
```
Bus 005 Device 010: ID 1532:009f Razer USA, Ltd Razer Viper Mini Signature Edition
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x009f 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Viper Mini Signature Edition
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              2 Razer Viper Mini Signature Edition
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.00
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
[... 128 lines cut ...]
</details>

### Packet Captures

GitHub wouldn't allow me to attach the .pcapng files directly so I zipped them. They are separated by function, and stripped of everything that isn't mouse <-> synapse communication. I also have some notes about what I think each thing in the protocol is doing if needed. Although I imagine these packets are fairly standard as they match your packet template under the reverse engineering guide.

[viperSE-pcap.zip](https://github.com/openrazer/openrazer/files/11899582/viperSE-pcap.zip)

Not sure if it makes a difference, but I triggered all the packets by clicking the settings from left to right, or top to bottom in the case of the Indicator LED on the HyperPolling dongle.

For the keybind configuration I set all keys to their default value under standard and hypershift In the order of top -> bottom on the left side, followed by top -> bottom on the right. Hope these help.

### Synapse Screenshots
<details>

<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

[image]
[image]
[image]
[image]
[image]
[image]

</details>

fireflame90051, 2023-07-02:
So, after looking through some of the commits for previous device additions, it appears adding a device to this driver is a bit less daunting that I originally presumed. I've created a fork with support for the Viper Mini SE in both wireless and wired configurations.

It compiles and works on my Arch desktop and an Ubuntu VM I was using for testing. Almost all of the features seem to be working great, except for configuring the asymmetric cutoff and smart tracking. Configuring the hyperpolling indicator works when set by 
```
echo $STATE > /sys/bus/hid/drivers/razermouse/$DEVICE/hyperpolling_wireless_dongle_indicator_led_mode 
```
but this configurable option doesn't seem to be picked up by any of the GUI's available. I imagine it's because it hasn't been implemented in the frontends, but I thought I would mention it anyway.

Lastly, not sure if this is driver related or not, but there is a bit of a weird thing with the reported default DPI stages in Polychromatic. They don't match the defaults in synapse at all, and are quite high. Not entirely sure why that is.

Anyway. Changes can be found on my github page, or by following the link here.
https://github.com/openrazer/openrazer/compare/master...fireflame90051:openrazer:master

If anyone has any insights on the issues listed above, or if I've missed something glaring, please let me know. Thanks!

I've also attached a couple screenshots showing the mouse working in Polychromatic.
<details>

[image]
[image]

</details>

z3ntu, 2023-07-16:
If you haven't already opened a pull request with your changes, please do so!

fireflame90051, 2023-08-03:
Sorry, got caught up with another project and forgot to submit.

Pull request submitted. Thanks!

I'll go ahead and close this issue now.

z3ntu, 2023-08-13:
Let's keep this open until support is merged, so people can find it.

PR #2107 (2023-09-02): Add Support for Razer Viper Mini SE
Added Viper Mini SE definitions to relevant functions in razermouse_driver
Added Viper Mini SE definitions to daemon
Added Udev Definitions
Added Fake Driver definitions
Added Viper Mini SE to supported devices

Modified razer_attr_write_hyperpolling_wireless_dongle_indicator_led_mode to allow for the Viper Mini SE's transaction ID.
Modified relevant dbus method to prevent daemon crash caused by switching from wireless -> wired when polling rate is set above 1000hz.

Fixes #2089
closed 2023-09-02 by z3ntu