openrazer/openrazer #2090: Orochi v2 bluetooth not detected; dongle works

ostrich, 2023-07-02:
### Distribution

Arch

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Orochi v2 mouse isn't detected when connected via Bluetooth. When I plug in the USB dongle it shows up in RazerGenie , so I seem to be most of the way there. 

#1774 had the same issue but it was closed without a resolution.

Packages:

    $ pacman -Q openrazer-meta-git openrazer-driver-dkms-git openrazer-daemon-git python-openrazer-git
    openrazer-meta-git 3.6.1.r71.gf50681d2-1
    openrazer-driver-dkms-git 3.6.1.r71.gf50681d2-1
    openrazer-daemon-git 3.6.1.r71.gf50681d2-1
    python-openrazer-git 3.6.1.r71.gf50681d2-1

DKMS:

    $ sudo dkms install openrazer-driver/3.6.1
    Module openrazer-driver/3.6.1 already installed on kernel 6.3.9-arch1-1.1 (x86_64), skip. You may override by specifying --force.

Groups:

    $ groups ostrich
    wheel lp libvirt plugdev ostrich

journalctl after plugging in and removing the USB dongle:

    $ journalctl -b | grep -i razer_mount
    Jul 02 16:58:20 archlaptop razer_mount[156354]: Driver razermouse
    Jul 02 16:58:20 archlaptop razer_mount[156355]: Device_ID 3-4
    Jul 02 16:58:20 archlaptop razer_mount[156356]: Modprobing razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156360]: Modprobed razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156368]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156370]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156373]: Device_ID 3-4:1.1
    Jul 02 16:58:21 archlaptop razer_mount[156374]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156375]: Device_ID 3-4:1.0
    Jul 02 16:58:21 archlaptop razer_mount[156376]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156377]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156378]: Device_ID 3-4:1.2
    Jul 02 16:58:21 archlaptop razer_mount[156379]: Device_ID 3-4:1.3
    Jul 02 16:58:21 archlaptop razer_mount[156380]: Device_ID 3-4:1.4
    Jul 02 16:58:21 archlaptop razer_mount[156383]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156385]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156388]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156392]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156391]: Device_ID 0003:1532:0094.0017
    Jul 02 16:58:21 archlaptop razer_mount[156390]: Device_ID 0003:1532:0094.0018
    Jul 02 16:58:21 archlaptop razer_mount[156389]: Driver razermouse
    Jul 02 16:58:21 archlaptop razer_mount[156393]: Device_ID 0003:1532:0094.001B
    Jul 02 16:58:21 archlaptop razer_mount[156395]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0017/
    Jul 02 16:58:21 archlaptop razer_mount[156396]: Device_ID 0003:1532:0094.0019
    Jul 02 16:58:21 archlaptop razer_mount[156394]: Device_ID 0003:1532:0094.001A
    Jul 02 16:58:21 archlaptop razer_mount[156397]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0018/
    Jul 02 16:58:21 archlaptop razer_mount[156399]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.001B/
    Jul 02 16:58:21 archlaptop razer_mount[156400]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0019/
    Jul 02 16:58:21 archlaptop razer_mount[156401]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.001A/
    Jul 02 16:58:21 archlaptop razer_mount[156406]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.001B/
    Jul 02 16:58:21 archlaptop razer_mount[156407]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.001A/
    Jul 02 16:58:21 archlaptop razer_mount[156408]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0019/
    Jul 02 16:58:21 archlaptop razer_mount[156409]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0017/
    Jul 02 16:58:21 archlaptop razer_mount[156410]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0018/

systemctl for openrazer-daemon.service after plugging in and removing the USB dongle:
	 
    $ systemctl --user status openrazer-daemon.service
    ● openrazer-daemon.service - Daemon to manage razer devices in userspace
         Loaded: loaded (/usr/lib/systemd/user/openrazer-daemon.service; enabled; preset: enabled)
         Active: active (running) since Fri 2023-06-30 19:05:02 EDT; 1 day 22h ago
           Docs: man:openrazer-daemon(8)
       Main PID: 679 (openrazer-daemo)
          Tasks: 4 (limit: 19046)
         Memory: 35.1M
            CPU: 460ms
         CGroup: /user.slice/user-1000.slice/user@1000.service/app.slice/openrazer-daemon.service
                 └─679 openrazer-daemon
    
    Jun 30 19:05:01 archlaptop systemd[671]: Starting Daemon to manage razer devices in userspace...
    Jun 30 19:05:02 archlaptop openrazer-daemon[679]: Starting daemon.
    Jun 30 19:05:02 archlaptop systemd[671]: Started Daemon to manage razer devices in userspace.
    Jun 30 19:05:02 archlaptop openrazer-daemon[679]: 2023-06-30 19:05:02 | razer                          | INFO     | Initialising Daemon (v3.6.1). Pid: 679
    Jun 30 19:05:02 archlaptop openrazer-daemon[679]: 2023-06-30 19:05:02 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
    Jun 30 19:05:02 archlaptop openrazer-daemon[679]: 2023-06-30 19:05:02 | razer                          | INFO     | Serving DBus
    Jul 02 16:58:23 archlaptop openrazer-daemon[679]: 2023-07-02 16:58:23 | razer                          | INFO     | Found valid device.0: 0003:1532:0094.0017
    Jul 02 16:58:23 archlaptop openrazer-daemon[679]: 2023-07-02 16:58:23 | razer.device0                  | INFO     | Initialising device.0 RazerOrochiV2Receiver
    Jul 02 16:59:04 archlaptop openrazer-daemon[679]: 2023-07-02 16:59:04 | razer                          | WARNING  | Removing 0003:1532:0094.0017

z3ntu, 2023-07-16:
Does the mouse show up in `lsusb | grep 1532` when connected via Bluetooth?

ostrich, 2023-07-16:
It doesn't.

z3ntu, 2023-07-17:
Then I believe you're out of luck. If the device uses some other way of doing HID then it's not supported at all by openrazer. So far I believe all the supported Bluetooth devices just show up as USB device and that's why we can easily interface with them, just like they were connected via cable.

ostrich, 2023-07-18:
Ok, thanks. I note that Device Support on the main page indicates "Razer Orochi V2 (Bluetooth)	1532:0095" as supported. Are other Razer Bluetooth mice known to work? I don't see any currently connected Bluetooth devices in `lsusb`, only the Bluetooth controller.

z3ntu, 2023-08-06:
I cannot give much advise on this topic since I don't own any Razer Bluetooth mouse myself, however as written above normally they should show up in lsusb I believe. You could find the github user who added support for that device and ping them if you want more info.

viik420, 2023-10-06:
I have the same problem. It works with dongle. When trying to connect via Bluetooth, it doesn't show up in Bluetooth scan.

VladStulikov, 2023-10-13:
Same problem here

VladStulikov, 2024-02-15:
The problem still exists

z3ntu, 2024-03-03:
I've marked it as bug now but as written in my last comment, I can't do anything on this topic - it seems to work for some people somehow and not for others.

moondownr, 2024-03-05:
My Orochi V2 mouse does show up in lsusb | grep 1532 twice when connected to Bluetooth:

Bus 003 Device 009: ID 1532:0f20 Razer USA, Ltd Razer Base Station V2 Chroma
Bus 003 Device 005: ID 1532:48f0 Razer USA, Ltd Razer Base Station V2 Chroma

Polychromatic reports following:

The following PIDs are not supported in the latest stable release of OpenRazer:
1532:48F0

z3ntu, 2024-04-04:
See also https://github.com/openrazer/openrazer/issues/2205#issuecomment-2037936440 , maybe compare this with the log on your computer?

jokerofalltrades, 2025-01-11:
I'd like to add that I'm running rpiOS, and I also have this issue. I'm using polychromatic as a frontend.
Output from journalctl w/ usb reciever in
```
journalctl -b | grep -i razer_mount
Jan 11 11:13:45 raspberrypi razer_mount[34679]: Driver razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34680]: Device_ID 1-2
Jan 11 11:13:45 raspberrypi razer_mount[34681]: Modprobing razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34685]: Modprobed razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34689]: Driver razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34690]: Driver razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34693]: Device_ID 1-2:1.0
Jan 11 11:13:45 raspberrypi razer_mount[34692]: Device_ID 1-2:1.1
Jan 11 11:13:45 raspberrypi razer_mount[34694]: Driver razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34695]: Device_ID 1-2:1.2
Jan 11 11:13:45 raspberrypi razer_mount[34698]: Driver razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34699]: Driver razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34701]: Device_ID 0003:1532:0094.0004
Jan 11 11:13:45 raspberrypi razer_mount[34702]: Device_ID 0003:1532:0094.0003
Jan 11 11:13:45 raspberrypi razer_mount[34703]: Driver razermouse
Jan 11 11:13:45 raspberrypi razer_mount[34704]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0003/
Jan 11 11:13:45 raspberrypi razer_mount[34705]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0004/
Jan 11 11:13:45 raspberrypi razer_mount[34706]: Device_ID 0003:1532:0094.0005
Jan 11 11:13:45 raspberrypi razer_mount[34709]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0005/
Jan 11 11:13:45 raspberrypi razer_mount[34710]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0003/
Jan 11 11:13:45 raspberrypi razer_mount[34711]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0004/
Jan 11 11:13:45 raspberrypi razer_mount[34718]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0094.0005/
```
the systemctl response is the same as ostrich's.