openrazer/openrazer #2094: Support for Razer Blade 14 2023

cameronbosch, 2023-07-17:
Note: I am still working on some parts of this.

### Device Info

* Name: Razer Blade 14 2023
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Bus 001 Device 002: ID 1532:029d Razer USA, Ltd Razer Blade
```

```
lrwxrwxrwx 1 root root  10 Jul 17 13:36 usb-Razer_Razer_Blade-event-if01 -> ../event14
lrwxrwxrwx 1 root root  10 Jul 17 13:36 usb-Razer_Razer_Blade-event-kbd -> ../event12
lrwxrwxrwx 1 root root  10 Jul 17 13:36 usb-Razer_Razer_Blade-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root  10 Jul 17 13:36 usb-Razer_Razer_Blade-if02-event-mouse -> ../event15
lrwxrwxrwx 1 root root   9 Jul 17 13:36 usb-Razer_Razer_Blade-if02-mouse -> ../mouse4
```

<details>

```
001:003:002:DESCRIPTOR         1689616344.871689
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:003:001:DESCRIPTOR         1689616344.874729
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 C0 05 0C 09 01 A1 01 85
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:003:000:DESCRIPTOR         1689616344.878404
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:002:003:DESCRIPTOR         1689616344.880800
 05 59 09 01 A1 01 85 01 09 02 A1 02 09 03 15 00
 27 FF FF 00 00 75 10 95 01 B1 03 09 04 09 05 09
 06 09 07 09 08 15 00 27 FF FF FF 7F 75 20 95 05
 B1 03 C0 85 02 09 20 A1 02 09 21 15 00 27 FF FF
 00 00 75 10 95 01 B1 02 C0 85 03 09 22 A1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 01 B1 02 09 23
 09 24 09 25 09 27 09 26 15 00 27 FF FF FF 7F 75
 20 95 05 B1 02 09 28 09 29 09 2A 09 2C 09 2D 15
 00 26 FF 00 75 08 95 05 B1 02 C0 85 04 09 50 A1
 02 09 03 09 55 15 00 25 08 75 08 95 02 B1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 08 B1 02 09 51
 09 52 09 53 09 51 09 52 09 53 09 51 09 52 09 53
 09 51 09 52 09 53 09 51 09 52 09 53 09 51 09 52
 09 53 09 51 09 52 09 53 09 51 09 52 09 53 15 00
 26 FF 00 75 08 95 18 B1 02 C0 85 05 09 60 A1 02
 09 55 15 00 25 08 75 08 95 01 B1 02 09 61 09 62
 15 00 27 FF FF 00 00 75 10 95 02 B1 02 09 51 09
 52 09 53 15 00 26 FF 00 75 08 95 03 B1 02 C0 85
 06 09 70 A1 02 09 71 15 00 25 01 75 08 95 01 B1
 02 C0 85 07 06 00 FF 15 00 26 FF 00 09 02 75 08
 95 3F B1 02 C0 05 0C 09 01 A1 01 85 08 19 D0 29
 D7 15 00 25 01 75 01 95 08 81 02 C0

001:002:004:DESCRIPTOR         1689616344.883700
 05 59 09 01 A1 01 85 01 09 02 A1 02 09 03 15 00
 27 FF FF 00 00 75 10 95 01 B1 03 09 04 09 05 09
 06 09 07 09 08 15 00 27 FF FF FF 7F 75 20 95 05
 B1 03 C0 85 02 09 20 A1 02 09 21 15 00 27 FF FF
 00 00 75 10 95 01 B1 02 C0 85 03 09 22 A1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 01 B1 02 09 23
 09 24 09 25 09 27 09 26 15 00 27 FF FF FF 7F 75
 20 95 05 B1 02 09 28 09 29 09 2A 09 2C 09 2D 15
 00 26 FF 00 75 08 95 05 B1 02 C0 85 04 09 50 A1
 02 09 03 09 55 15 00 25 08 75 08 95 02 B1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 08 B1 02 09 51
 09 52 09 53 09 51 09 52 09 53 09 51 09 52 09 53
 09 51 09 52 09 53 09 51 09 52 09 53 09 51 09 52
 09 53 09 51 09 52 09 53 09 51 09 52 09 53 15 00
 26 FF 00 75 08 95 18 B1 02 C0 85 05 09 60 A1 02
 09 55 15 00 25 08 75 08 95 01 B1 02 09 61 09 62
 15 00 27 FF FF 00 00 75 10 95 02 B1 02 09 51 09
 52 09 53 15 00 26 FF 00 75 08 95 03 B1 02 C0 85
 06 09 70 A1 02 09 71 15 00 25 01 75 08 95 01 B1
 02 C0 85 07 06 00 FF 15 00 26 FF 00 09 02 75 08
 95 3F B1 02 C0

001:002:002:DESCRIPTOR         1689616344.886698
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:002:001:DESCRIPTOR         1689616344.889695
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:002:000:DESCRIPTOR         1689616344.892694
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

cameronbosch, 2023-07-22:
Unfortunately, I no longer have this laptop. I returned it because of the speaker issues and because I'm getting the Framework Laptop 16. Sorry. 🫤

closed 2023-07-22 by cameronbosch

z3ntu, 2023-08-06:
Anyways duplicate of #2087