openrazer/openrazer #2098: gpg key issue: Certificate has no valid binding signature

SingingBush, 2023-07-27:
After adding the repo to my Fedora 38 system and running `sudo dnf install razergenie -y` I found that there is an issue with the gpg key:

1. Certificate 062F9FD4D6D11CE4 invalid: policy violation because: No binding signature at time 2023-04-21T06:48:56Z
2. Certificate has no valid binding signature as of the signature's creation time, but is valid now.  The certificate has probably been stripped or minimized.

full output here:

```
$ sudo dnf install razergenie -y
Last metadata expiration check: 0:33:06 ago on Thu 27 Jul 2023 09:25:29 BST.
Dependencies resolved.
=========================================================================================================================================================================================================
 Package                                           Architecture                                Version                                         Repository                                           Size
=========================================================================================================================================================================================================
Installing:
 razergenie                                        x86_64                                      1.0.0-1.1                                       hardware_razer                                      125 k
Installing dependencies:
 libopenrazer                                      x86_64                                      0.1.0-1.1                                       hardware_razer                                      110 k

Transaction Summary
=========================================================================================================================================================================================================
Install  2 Packages

Total size: 235 k
Installed size: 1.1 M
Downloading Packages:
[SKIPPED] libopenrazer-0.1.0-1.1.x86_64.rpm: Already downloaded                                                                                                                                         
[SKIPPED] razergenie-1.0.0-1.1.x86_64.rpm: Already downloaded                                                                                                                                           
hardware:razer (Fedora_38)                                                                                                                                               9.5 kB/s | 1.1 kB     00:00    
Importing GPG key 0xD6D11CE4:
 Userid     : "hardware OBS Project <hardware@build.opensuse.org>"
 Fingerprint: E96C 496E 4E77 C159 C2FC A053 062F 9FD4 D6D1 1CE4
 From       : https://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key
Key imported successfully
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  1. Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:48:56Z
  2. Certificate has no valid binding signature as of the signature's creation time, but is valid now.  The certificate has probably been stripped or minimized.
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  1. Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:48:56Z
  2. Certificate has no valid binding signature as of the signature's creation time, but is valid now.  The certificate has probably been stripped or minimized.
Import of key(s) didn't help, wrong key(s)?
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  1. Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:50:23Z
  2. Certificate has no valid binding signature as of the signature's creation time, but is valid now.  The certificate has probably been stripped or minimized.
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  1. Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:50:23Z
  2. Certificate has no valid binding signature as of the signature's creation time, but is valid now.  The certificate has probably been stripped or minimized.
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  1. Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:50:23Z
  2. Certificate has no valid binding signature as of the signature's creation time, but is valid now.  The certificate has probably been stripped or minimized.
error: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
  1. Certificate 062F9FD4D6D11CE4 invalid: policy violation
      because: No binding signature at time 2023-04-21T06:50:23Z
  2. Certificate has no valid binding signature as of the signature's creation time, but is valid now.  The certificate has probably been stripped or minimized.
Public key for libopenrazer-0.1.0-1.1.x86_64.rpm is not trusted. Failing package is: libopenrazer-0.1.0-1.1.x86_64
 GPG Keys are configured as: https://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key
Public key for razergenie-1.0.0-1.1.x86_64.rpm is not trusted. Failing package is: razergenie-1.0.0-1.1.x86_64
 GPG Keys are configured as: https://download.opensuse.org/repositories/hardware:/razer/Fedora_38/repodata/repomd.xml.key
The downloaded packages were saved in cache until the next successful transaction.
You can remove cached packages by executing 'dnf clean packages'.
Error: GPG check FAILED
```

pabaucom, 2023-08-03:
I have the same issue with a fresh Fedora 38 install, fully updated, and openrazer being the first thing installed/tested.  I get this gpg error with polychromatic and razergenie, but razergenie can be installed via flatpak.  I need polychromatic features, though.  All the same logs and issues described here and in issue #2061.  There is no fix I am aware of.

I don't know who needs to address the issue since I don't fully understand the output of those logs.  If someone can advise, I will open an issue with the appropriate party.

pabaucom, 2023-08-03:
The polychromatic issue [here](https://github.com/polychromatic/polychromatic/issues/479) has a workaround that allowed for the install of polychromatic on my system.  I don't how well this will hold up after updates, etc.  But this may work for someone else that finds this:

https://github.com/polychromatic/polychromatic/issues/479#issuecomment-1645932063

lah7, 2023-10-10:
The "No binding signature" issue was fixed last month for Fedora 38:

* https://github.com/rpm-software-management/rpm-sequoia/issues/46

I can confirm OpenRazer now installs normally in a fresh Fedora 38 container 👍🏼

closed 2023-10-10 by lah7