openrazer/openrazer #2099: Not Sure where to Ask, any suggestions on getting the Naga Hyperspeed V2 working with Bluetooth and not just 2.4

Madcliff, 2023-07-28:
Pretty sure this is not strictly a OpenRazer issue, but there seems to be an issue with the bluetooth mode for the Naga V2 Hyperspeed not sustaining a connection. I am almost confident that is related to some measure of power savings setting, and is likely a udev thing, but I am not sure where to begin in working out what needs to be changed so was hoping I might get some pointers here. 

### Distribution

### Support

- [ ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

z3ntu, 2023-08-06:
Maybe you can find some github users in the issue for adding support for this device and get advise from them? I can't offer anything here since I don't own any Bluetooth-capable Razer device.

stephanlamoureux, 2023-08-12:
I was able to get my deathadder connected to Bluetooth on a ThinkPad and Fedora 38. The laptop got fried, I replaced it with the same exact one, and now the mouse won't show up in Bluetooth devices 😢

closed 2024-05-11 by github-actions[bot]