openrazer/openrazer #2100: Doesn't install on Debian

tomcass240, 2023-07-28:
Trying to install the .deb package on Debian 12 and it's not working and saying it has unmet dependencies. 
```

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'openrazer-meta' instead of './openrazer-meta_3.6.1-0_all.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 openrazer-meta : Depends: openrazer-driver-dkms (= 3.6.1-0) but it is not going to be installed
                  Depends: openrazer-daemon (= 3.6.1-0) but it is not going to be installed
                  Depends: python3-openrazer (= 3.6.1-0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

z3ntu, 2023-07-29:
Why are you trying to install the .deb file directly? Just add the repo and use the apt commands to install it.

tomcass240, 2023-07-29:
The website hasn't been updated and still says Debian 11 is the current release but we're on 12 now. I found the package on Debian's own repo so just installed it from that. The problem was the package I downloaded was trying to install 3.6 but only 3.5 is available in the repo.

closed 2023-07-29 by tomcass240

z3ntu, 2023-07-29:
Correct, Debian 12 doesn't seem to have been enabled on OBS yet. I've just enabled it and the packages should be available now in the repo linked on our website.