openrazer/openrazer #2101: [Kubuntu 23.04] Errors in logs

TriMoon, 2023-08-01:
See: https://github.com/polychromatic/polychromatic/issues/490

---
I'm not sure if i have to file this info here or on [the razer repo](https://github.com/openrazer/openrazer).

Anyhow here are the log entries i see:
- `journalctl -b0 --grep "(polychromatic|razer)"`
```
Aug 01 15:17:50 kubuntu kernel: usb 1-10.4: Product: Razer Naga Pro
Aug 01 15:17:50 kubuntu kernel: usb 1-10.4: Manufacturer: Razer
Aug 01 15:17:50 kubuntu kernel: input: Razer Razer Naga Pro as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.0/0003:1532:0090.0002/input/input3
Aug 01 15:17:50 kubuntu kernel: hid-generic 0003:1532:0090.0002: input,hidraw1: USB HID v1.11 Mouse [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input0
Aug 01 15:17:50 kubuntu kernel: input: Razer Razer Naga Pro Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.1/0003:1532:0090.0003/input/input4
Aug 01 15:17:50 kubuntu kernel: input: Razer Razer Naga Pro Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.1/0003:1532:0090.0003/input/input5
Aug 01 15:17:50 kubuntu kernel: input: Razer Razer Naga Pro as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.1/0003:1532:0090.0003/input/input6
Aug 01 15:17:50 kubuntu kernel: hid-generic 0003:1532:0090.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input1
Aug 01 15:17:50 kubuntu kernel: input: Razer Razer Naga Pro as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.2/0003:1532:0090.0004/input/input7
Aug 01 15:17:50 kubuntu kernel: hid-generic 0003:1532:0090.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input2
Aug 01 15:17:50 kubuntu kernel: hid-generic 0003:1532:0090.0005: hiddev0,hidraw4: USB HID v1.11 Device [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input4
Aug 01 15:17:50 kubuntu kernel: input: Razer Razer Naga Pro Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.3/0003:1532:0090.0006/input/input8
Aug 01 15:17:50 kubuntu kernel: hid-generic 0003:1532:0090.0006: input,hiddev1,hidraw5: USB HID v1.11 Device [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input3
Aug 01 15:17:51 kubuntu kernel: input: Razer Razer Naga Pro as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.0/0003:1532:0090.0002/input/input11
Aug 01 15:17:51 kubuntu kernel: razermouse 0003:1532:0090.0002: input,hidraw1: USB HID v1.11 Mouse [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input0
Aug 01 15:17:51 kubuntu razer_mount[681]: Driver razermouse
Aug 01 15:17:51 kubuntu razer_mount[692]: Modprobing razermouse
Aug 01 15:17:51 kubuntu kernel: input: Razer Razer Naga Pro as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.1/0003:1532:0090.0003/input/input24
Aug 01 15:17:51 kubuntu kernel: razermouse 0003:1532:0090.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input1
Aug 01 15:17:52 kubuntu kernel: input: Razer Razer Naga Pro as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.2/0003:1532:0090.0004/input/input25
Aug 01 15:17:52 kubuntu kernel: razermouse 0003:1532:0090.0004: input,hidraw3: USB HID v1.11 Keyboard [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input2
Aug 01 15:17:52 kubuntu kernel: razermouse 0003:1532:0090.0005: hiddev0,hidraw4: USB HID v1.11 Device [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input4
Aug 01 15:17:52 kubuntu kernel: input: Razer Razer Naga Pro as /devices/pci0000:00/0000:00:14.0/usb1/1-10/1-10.4/1-10.4:1.3/0003:1532:0090.0006/input/input26
Aug 01 15:17:52 kubuntu kernel: razermouse 0003:1532:0090.0006: input,hiddev1,hidraw5: USB HID v1.11 Device [Razer Razer Naga Pro] on usb-0000:00:14.0-10.4/input3
Aug 01 15:17:52 kubuntu razer_mount[1105]: Modprobed razermouse
Aug 01 15:17:52 kubuntu razer_mount[1137]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1146]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1147]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1151]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0006/
Aug 01 15:17:52 kubuntu razer_mount[1155]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0006/
Aug 01 15:17:52 kubuntu razer_mount[1156]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1162]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1170]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1174]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1175]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0005/
Aug 01 15:17:52 kubuntu razer_mount[1177]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1180]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0005/
Aug 01 15:17:52 kubuntu razer_mount[1182]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1186]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0004/
Aug 01 15:17:52 kubuntu razer_mount[1185]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0003/
Aug 01 15:17:52 kubuntu razer_mount[1188]: Driver razermouse
Aug 01 15:17:52 kubuntu razer_mount[1192]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0004/
Aug 01 15:17:52 kubuntu razer_mount[1197]: Changing group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0002/
Aug 01 15:17:52 kubuntu razer_mount[1227]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0003/
Aug 01 15:17:52 kubuntu razer_mount[1235]: Changed group /sys/bus/hid/drivers/razermouse/0003:1532:0090.0002/
Aug 01 15:17:52 kubuntu systemd-logind[1061]: Watching system buttons on /dev/input/event7 (Razer Razer Naga Pro)
Aug 01 15:17:52 kubuntu systemd-logind[1061]: Watching system buttons on /dev/input/event20 (Razer Razer Naga Pro)
Aug 01 15:17:54 kubuntu systemd[1447]: Starting openrazer-daemon.service - Daemon to manage razer devices in userspace...
Aug 01 15:17:55 kubuntu openrazer-daemon[1468]: 2023-08-01 15:17:55 | razer                          | CRITICAL | User is not a member of the plugdev group
Aug 01 15:17:55 kubuntu openrazer-daemon[1468]: 2023-08-01 15:17:55 | razer                          | CRITICAL | Please run the command 'sudo gpasswd -a $USER plugdev' and then reboot!
Aug 01 15:17:55 kubuntu systemd[1447]: openrazer-daemon.service: Main process exited, code=exited, status=1/FAILURE
Aug 01 15:17:55 kubuntu systemd[1447]: openrazer-daemon.service: Failed with result 'exit-code'.
Aug 01 15:17:55 kubuntu systemd[1447]: Failed to start openrazer-daemon.service - Daemon to manage razer devices in userspace.
Aug 01 15:18:01 kubuntu systemd[1558]: Starting openrazer-daemon.service - Daemon to manage razer devices in userspace...
Aug 01 15:18:01 kubuntu systemd[1558]: Started openrazer-daemon.service - Daemon to manage razer devices in userspace.
Aug 01 15:18:01 kubuntu openrazer-daemon[1565]: 2023-08-01 15:18:01 | razer                          | INFO     | Initialising Daemon (v3.6.1). Pid: 1565
Aug 01 15:18:01 kubuntu openrazer-daemon[1565]: 2023-08-01 15:18:01 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
Aug 01 15:18:01 kubuntu openrazer-daemon[1565]: 2023-08-01 15:18:01 | razer                          | INFO     | Found device.0: 0003:1532:0090.0002
Aug 01 15:18:01 kubuntu openrazer-daemon[1565]: 2023-08-01 15:18:01 | razer.device0                  | INFO     | Initialising device.0 RazerNagaProWireless
Aug 01 15:18:01 kubuntu kernel: razermouse: Command not supported. status: 05 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 03, command_class: 03, command_id.id: 0d Params: 01050400000000000000000000000000 .
Aug 01 15:18:01 kubuntu dbus-daemon[1582]: [session uid=1000 pid=1582] Activating service name='org.freedesktop.Notifications' requested by ':1.8' (uid=1000 pid=1565 comm="openrazer-daemon                             " label="unconfined")
Aug 01 15:18:04 kubuntu systemd[1558]: Starting app-polychromatic\x2dautostart@autostart.service - Polychromatic Autostart...
Aug 01 15:18:04 kubuntu systemd[1558]: Started app-polychromatic\x2dautostart@autostart.service - Polychromatic Autostart.
Aug 01 15:18:05 kubuntu openrazer-daemon[1565]: 2023-08-01 15:18:05 | razer                          | INFO     | Serving DBus
Aug 01 15:18:06 kubuntu polychromatic-helper[2230]: /usr/bin/polychromatic-tray-applet:203: DeprecationWarning: Gtk.ImageMenuItem.set_image is deprecated
[... cut ...] /python3 /usr/bin/polychromatic-tray-apple" label="unconfined")
Aug 01 15:18:07 kubuntu kded5[1765]: Registering ":1.49/org/ayatana/NotificationItem/polychromatic_tray_applet" to system tray
Aug 01 15:24:12 kubuntu kernel: razermouse: Command timed out. status: 04 transaction_id.id: 1f remaining_packets: 00 protocol_type: 00 data_size: 02, command_class: 07, command_id.id: 80 Params: 00000000000000000000000000000000 .
```

PS: My user **_IS_** a member of the `plugdev` group... :woman_shrugging:

lah7, 2023-08-01:
Is OpenRazer (and apps) working though?  From that output, it does look like the daemon initially fails to start due to lack of `plugdev` permissions for some reason, but it appears to work on its second attempt (`Serving DBus`). Since `polychromatic-tray-applet` started before OpenRazer was ready, if this app isn't listing the devices at start-up, you might need to add a delayed start - the app has a setting for this.

> razermouse: Command timed out. 

That can happen. Some devices don't like receiving too many requests at once.

TriMoon, 2023-08-03:
> @lah7 
Is OpenRazer (and apps) working though?

I forgot to mention that the popup notifications of polychromatic always say my battery  is 0% which is of-course not correct...

lah7, 2023-08-03:
> You guys might check your code to account for that, it aint good to fill logs with problems that can be eliminated by checks.

One of OpenRazer's to-dos to perform some error checking in the driver. For now, it is how it is so the driver doesn't crash userspace applications. It can be useful output when implementing support for a device.

---

> popup notifications of polychromatic always say my battery is 0% which is of-course not correct...

Battery notifications are a feature of `openrazer-daemon`, not Polychromatic. That 0% could be because the command timed out or maybe it is a daemon bug if notifications always spit out 0% but not in apps. Might be related to #2025.  It's not clear why - plus, we don't have wireless hardware to test.

If it's annoying, those notifications can be turned off via `~/.config/openrazer/razer.conf`. A long term solution like #87 would be much better.

----

What is the end goal for this issue? If it's about the `openrazer-daemon.service` startup failures,  I'm not sure why that happens -- unless there are two user services (running under two different user IDs). If that's not the case, I'd try disabling the service and rely on applications (like `polychromatic-tray-applet`) to summon `openrazer-daemon` automatically when it utilises the OpenRazer API.

    systemctl --user disable openrazer-daemon.service

TriMoon, 2023-08-04:
> What is the end goal for this issue?

The end goal was to inform you of these log messages so they can be fixed and not appear anymore...

TriMoon, 2023-09-03:
@openrazer @lah7 
I came across #2033 while searching on the internet for `razermouse: Command timed out.` error messages.

I tried the script mentioned in https://github.com/openrazer/openrazer/issues/2033#issuecomment-1469843711 and found it showed the below while the mouse was connected using both wired and wireless connection at same time:
- `0%` when checking the **Wireless** connection of my mouse.
- Actual charge info when using the **Wired** connection of the same mouse.

Then found this reply by @lah7:
> https://github.com/openrazer/openrazer/issues/1796#issuecomment-1326745382
> UNKWN******* I believe is because the device isn't returning a valid serial number.

I also got these entries in my `persistence.conf` file which i was confused about also.
Now the question is _"Why doesn't the same device not return same info when using different connections?"_
IMHO, it is related to the error messages i started with and not the mouse it self, eg the openrazer driver and not the connection technology used hardware-wise.

FYI: I'm nowadays using https://github.com/z3ntu/RazerGenie instead of https://github.com/polychromatic/polychromatic and the issue is same due to the @openrazer driver i guess.

---
```console
alias razer-list-devices='python3 /usr/share/doc/openrazer-doc/examples/list_devices.py'
alias razer-battery-tray='~/git/RazerBatteryTray/razer_battery_tray.py'
```
- `razer-list-devices`:
   ```console
   Found 2 Razer devices

   Razer Naga Pro (Wired):
      type: mouse
      serial: <actual SN>
      firmware version: v2.2
      driver version: 3.6.1

   Razer Naga Pro (Wireless):
      type: mouse
      serial: UNKWN000000003654
      firmware version: v0.0
      driver version: 3.6.1
   ```
- `razer-battery-tray`:
   ```console
   Starting Razer Battery Tray
   Pillow version:  9.4.0
   Openrazer version:  3.6.1
   Python version:  3.11.4 (main, Jun  9 2023, 07:59:55) [GCC 12.3.0]
   Path of the icons: <My $HOME>/git/RazerBatteryTray/icons
   Pystray version:  0.19.4

   !! OOPS: Missing device name as argument !!

   Usage: razer_bat_tray.py <part_of_the_device_name>
   Example 1: razer_bat_tray.py 'Razer Viper Ultimate'
   Example 2: razer_bat_tray.py 'DeathAdder'

   Listing Available Devices:
   Razer Naga Pro (Wired) 
       Has Battery: True
   Razer Naga Pro (Wireless) 
       Has Battery: True
   ```

TriMoon, 2023-09-03:
PS:
I just check with `cat /sys/bus/hid/drivers/razermouse/*/charge_level` and it shows `0` for the wireless mouse device...

TriMoon, 2023-10-16:
Extra info, while using openrazer v3.6.1 wrt #2122 

- After booting while the mouse is in "Wireless" mode, connected only using WiFi:
   ```console
   $  cat /sys/bus/hid/drivers/razer*/*1532*/device_serial
   <empty line output>
   $  ll /sys/bus/hid/drivers/razer*/*1532*/device_serial
   -r--r----- 1 root plugdev 4,0K Oct 16 18:23 /sys/bus/hid/drivers/razermouse/0003:1532:0090.0003/device_serial
   ```
- After switching the USB-cable from the Dongle to the mouse, and thus using it in "Wired" mode:
   ```console
   $  cat /sys/bus/hid/drivers/razer*/*1532*/device_serial
   <actual SN>
   <actual SN>
   $  ll /sys/bus/hid/drivers/razer*/*1532*/device_serial
   -r--r----- 1 root plugdev 4,0K Oct 16 21:10 /sys/bus/hid/drivers/razermouse/0003:1532:008F.000A/device_serial
   -r--r----- 1 root plugdev 4,0K Oct 16 21:10 /sys/bus/hid/drivers/razermouse/0003:1532:008F.000D/device_serial
   ```
- After switching the USB-cable back to the Dongle and thus returning to the "Wireless" mode:
   ```console
   $  cat /sys/bus/hid/drivers/razer*/*1532*/device_serial
   <actual SN>
   $  ll /sys/bus/hid/drivers/razer*/*1532*/device_serial
   -r--r----- 1 root plugdev 4,0K Oct 16 21:22 /sys/bus/hid/drivers/razermouse/0003:1532:0090.000E/device_serial
   ```

The above shows there is something really weird going on, because every time i change operation modes between "Wired" and "Wireless" the numbers after the "0003:1532" part change _(Especially the last number)_...
And why the double device when only 1 is active? :thinking:

z3ntu, 2023-10-16:
If two USB devices are present on the computer, then two will appear in OpenRazer. There's nothing there to prevent this from happening, and apart from the daemon "hiding" one, I don't see how that would be possible.

TriMoon, 2023-10-17:
@z3ntu i have only 1 mouse, it's either connected directly via usb cable or via a Wifi dongle that's attached to the same usb port.
I need to disconnect the dongle to connect the mouse directly, so it's never a case of 2 devices present at same time...

z3ntu, 2023-10-17:
Okay that sounds unusual then. Normally we guard with this https://github.com/openrazer/openrazer/blob/master/driver/razermouse_driver.c#L4603-L4614 against matching with multiple interfaces of the same device, maybe yours is a bit different there?

Can you share `lsusb -d 1532:008F -v` ?

TriMoon, 2023-10-22:
@z3ntu
Because the `idProduct` changes on each reconnect, here is the output while connected via wireless:
```console
$ lsusb -d 1532:0090 -v | xc
Bus 003 Device 007: ID 1532:0090 Razer USA, Ltd Razer Naga Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0090 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 Default configuration
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
[... 109 lines cut ...]
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```

```console
$ razer-list-devices | xc
Found 1 Razer devices

Razer Naga Pro (Wireless):
   type: mouse
   serial: <actual SN>
   firmware version: v0.0
   driver version: 3.6.1

```

---
```console
$ alias xc
alias xc='deprivate | xclip -selection clipboard'
$ alias razer-list-devices
alias razer-list-devices='python3 /usr/share/doc/openrazer-doc/examples/list_devices.py'
```

TriMoon, 2023-10-23:
Fresh reboot and mouse connected directly via it's own USB cable:
```console
$ lsusb -d 1532:008f -v | xc

Bus 003 Device 003: ID 1532:008f Razer USA, Ltd Razer Naga Pro
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x008f 
  bcdDevice            1.00
  iManufacturer           0 
  iProduct                2 Razer Naga Pro
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     325
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
```console
$ razer-list-devices | xc
Found 1 Razer devices

Razer Naga Pro (Wired):
   type: mouse
   serial: <actual SN>
   firmware version: v2.2
   driver version: 3.6.1

```

z3ntu, 2023-10-23:
Okay, I think it makes sense at least from the output that it shows up twice in Wired.

Can you also provide `usbhid-dump -m 1532 -ed` when connected wired?

z3ntu, 2023-10-23:
Hm so if we follow https://github.com/openrazer/openrazer/pull/1920 / https://github.com/openrazer/openrazer/pull/2029 then we see this for the first descriptor:
```
0x06, 0x00, 0xFF,  //   Usage Page (Vendor Defined 0xFF00)
0x95, 0x3F,        //   Report Count (63)
```
And for the last:
```
0x06, 0x00, 0xFF,  //   Usage Page (Vendor Defined 0xFF00)
0x95, 0x5A,        //   Report Count (90)
```
This kind of implies if we want to send the 90 byte reports to the device, we should choose the last one. Maybe I should look at that PR again to at least get the infrastructure in place for new devices...