openrazer/openrazer #2103: The linux kernel does not see hardware with a VID of 1532

hitchhooker, 2023-08-02:
### Distribution
EndeavourOS
3537 🕸 uname -r                                         ⏎ [~]
6.4.7-arch1-1

3488 🕸 sudo dkms install openrazer-driver/3.6.1                                                                                   [~]
Module openrazer-driver/3.6.1 already installed on kernel 6.4.7-arch1-1 (x86_64), skip. You may override by specifying --force.
user@pc01  𓅨 ~
3489 🕸                                                                                                                            [~]
user@pc01  𓅨 ~
3489 🕸 lsusb                                                                                                                    ⏎ [~]
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 1a2c:9df5 China Resource Semico Co., Ltd USB Gaming Keyboard
Bus 003 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 007: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 001 Device 008: ID 8087:1041 Intel Corp. Realtek USB MIC
Bus 001 Device 005: ID 0bda:0511 Realtek Semiconductor Corp. USB Camera
Bus 001 Device 003: ID 214b:7250 Huasheng Electronics USB2.0 HUB
Bus 001 Device 002: ID 30fa:1140 INSTANT RAZERmini Gaming Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

my hardware is  AMD Ryzen 9 7940HS w/ Radeon 780M Graphics

instead of 1532, our VID is 30fa so any application wont recognize anything with id 1532 and wont work...

z3ntu, 2023-08-02:
Are you sure this device is a genuine Razer device? The device name of `INSTANT RAZERmini Gaming Mouse` doesn't invoke particularly much trust.

Or in other words: Does Razer Synapse on Windows detect the device and is it possible to interact with it there?

hitchhooker, 2023-08-02:
https://www.lazada.co.th/products/i4700304819-s19373951645.html
it could also be incredibly well made fake now that you say about it. is there any way to identify?
S/N: PM2146H15054752
model no: rz01-0325
product no: RZ01-03250100

i dont have windows available

z3ntu, 2023-08-02:
You can launch a Windows VM (e.g. with quickemu or something) and do USB passthrough to check there.

But so far doesn't look like genuine internals at least, e.g. https://linux-hardware.org/index.php?id=usb:30fa-1140 has the same USB VID/PID and also from "INSTANT" brand

hitchhooker, 2023-08-03:
thank you very much spotting it out. its incredibly well made fake. regardless of drivers im actually quite happy of its performance +  build quality. I guess these are just things that asian markets are flooded with  and made good enough to cheat majority of people. will buy another one to see if I can notice the difference.

closed 2023-08-03 by hitchhooker