openrazer/openrazer #2108: What does the phrase mean "You may override by specifying --force"

VlaK0r, 2023-08-09:
Hello!
Thank you for your work.

When downloading Debian 12.1, the mouse does not connect Polychromatic automatically.
When the daemon is forcibly rebooted, the mouse connects and works until the next Debian reboot.

When checking , it gives the following  (sudo dkms install openrazer-driver/3.6.1)

Module openrazer-driver/3.6.1 already installed on kernel 6.1.0-10-amd64 (x86_64), skip. You may override by specifying --force.

Please help me to sort out the problem.

VlaK0r, 2023-08-10:
> Here you mean something like `systemctl --user restart openrazer-daemon.service` ?

I mean through the program interface (Polychromatic )
And until I reboot it, the backlight settings don't apply.

z3ntu, 2023-08-13:
Can you check the openrazer log `~/.local/share/openrazer/logs/razer.log` on bootup before you restart the daemon? Maybe there's some info in there that could give a hint why this happens for you.

VlaK0r, 2023-08-13:
> 

`2023-08-13 14:55:09 | razer                          | INFO     | Found valid device.0: 0003:1532:0090.0001
2023-08-13 14:55:09 | razer.device0                  | INFO     | Initialising device.0 RazerNagaProWireless
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:1
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:2
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:3
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:4
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:5
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.restoreLastEffect method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDriverVersion method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getRazerUrls method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getSerial method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceImage method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceMode method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.setDeviceMode method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getEffect method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getWaveDir method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getEffectColors method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.getEffectSpeed method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoEffect method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoEffectColors method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoEffectSpeed method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.logo.getLogoWaveDir method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollEffectColors method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollEffectSpeed method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollWaveDir method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.scroll.getScrollEffect method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.dpi.maxDPI method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.dpi.getDPI method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.dpi.setDPI method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.getPollRate method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.misc.setPollRate method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.dpi.getDPIStages method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.dpi.setDPIStages method to DBus
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | Adding razer.device.macro.getMacros method to DBus
[... cut ...]
2023-08-13 14:56:18 | razer.device0                  | DEBUG    | DBus call get_current_logo_wave_dir
2023-08-13 14:56:18 | razer.device0                  | DEBUG    | DBus call get_current_logo_effect_speed
2023-08-13 14:56:18 | razer.device0                  | DEBUG    | DBus call get_current_logo_effect_colors
2023-08-13 14:56:18 | razer.device0                  | DEBUG    | DBus call get_logo_brightness
2023-08-13 14:56:18 | razer.device0                  | DEBUG    | DBus call get_dpi_xy
2023-08-13 14:56:18 | razer.device0                  | DEBUG    | DBus call get_dpi_xy`

VlaK0r, 2023-08-13:
> Can you check the openrazer log `~/.local/share/openrazer/logs/razer.log` on bootup before you restart the daemon? Maybe there's some info in there that could give a hint why this happens for you.

[razer.log](https://github.com/openrazer/openrazer/files/12329462/razer.log)

I watched the mouse work.
1. If you start working with the mouse immediately after restarting (without waiting for the program to load), then the internal profile is probably loaded. At the same time, opening the program, we see the default settings. After restarting the daemon, the program settings change to mouse settings.
2. If you wait for the program to load after restarting the PC. The program settings (by default) are loaded into the mouse . And if you restart the daemon after that, then the settings are loaded from the mouse.

z3ntu, 2023-08-13:
Okay so based on the first few lines
```
2023-08-13 14:55:09 | razer                          | INFO     | Found valid device.0: 0003:1532:0090.0001
2023-08-13 14:55:09 | razer.device0                  | INFO     | Initialising device.0 RazerNagaProWireless
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:1
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:2
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:3
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:4
2023-08-13 14:55:10 | razer.device0                  | DEBUG    | getting serial:  count:5
```
It's failing to get the serial number for some reason... Not sure (without trying) if the daemon can work correctly if the serial cannot be get. This for sure isn't well handled. Can you also post dmesg after reboot when the daemon doesn't work? I imagine to find a bunch of lines from `razermouse` about some failures.

VlaK0r, 2023-08-13:
> Okay so based on the first few lines
>
> ```
> 2023-08-13 14:55:09 | razer                          | INFO     | Found valid device.0: 0003:1532:0090.0001
> 2023-08-13 14:55:09 | [...]

dmesg  What is this?

z3ntu, 2023-08-13:
Kernel log, you run the command `dmesg` to print it. Use `dmesg | grep razer` so it's less to copy-paste.

VlaK0r, 2023-08-13:
> Kernel log, you run the command `dmesg` to print it. Use `dmesg | grep razer` so it's less to copy-paste.

dmesg | grep razer
[    5.593243] razermouse: loading out-of-tree module taints kernel.
[    5.593287] razermouse: module verification failed: signature and/or required key missing - tainting kernel
[    5.944733] razermouse 0003:1532:0090.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Naga Pro] on usb-0000:02:00.0-11/input0
[    6.972392] razermouse 0003:1532:0090.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Naga Pro] on usb-0000:02:00.0-11/input1
[    7.216297] razermouse 0003:1532:0090.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Naga Pro] on usb-0000:02:00.0-11/input2
[    7.216519] razermouse 0003:1532:0090.0004: hiddev0,hidraw3: USB HID v1.11 Device [Razer Razer Naga Pro] on usb-0000:02:00.0-11/input4
[    7.324314] razermouse 0003:1532:0090.0005: input,hiddev1,hidraw4: USB HID v1.11 Device [Razer Razer Naga Pro] on usb-0000:02:00.0-11/input3

z3ntu, 2023-08-14:
Can you manually run `cat /sys/bus/hid/drivers/razermouse/0*/device_serial` ? Should output some number like `XX123456789`

z3ntu, 2023-08-14:
Very weird then, not sure what's going on...

VlaK0r, 2023-08-14:
> Very weird then, not sure what's going on...

Maybe I didn't install it correctly?

binarykitchen, 2024-01-03:
Same problem here. When I do 

```
sudo systemctl --user restart openrazer-daemon.service
```

I see:

```
Failed to connect to bus: No medium found
```

z3ntu, 2024-01-03:
Don't use sudo for `systemctl --user`

binarykitchen, 2024-01-03:
Thanks, @z3ntu, lesson learnt.

z3ntu, 2024-01-05:
But I think the original problem/question has been solved, so I'm closing this.

closed 2024-01-05 by z3ntu