openrazer/openrazer #2109: Device battery reported as 0% (Naga Pro, DSV2PTKL, Pro Click)

Hamster42069, 2023-08-10:
### Distribution
Manjaro Linux, installed via yay. Openrazer doesn't seem to be correctly reporting my device battery, and it is giving constant notifications that my peripherals are at 0%.

z3ntu, 2023-08-10:
Is the battery percent always reported at 0% or sometimes also at the correct percentage? Any error from razermouse in dmesg?

VlaK0r, 2023-08-12:
> ### Distribution
> Manjaro Linux, installed via yay. Openrazer doesn't seem to be correctly reporting my device battery, and it is giving constant notifications that my peripherals are at 0%.

The same problem.

Hamster42069, 2023-08-13:
hello, i am new to linux but this is my dmesg report https://pastebin.com/3Ch7VnPF

z3ntu, 2023-08-13:
Okay, timeout unfortunately. I have some plans for the future how to improve the situation but unfortunately it's still a bit out.

closed 2024-01-11 by github-actions[bot]