openrazer/openrazer #2110: How to sign the modules myself in OpenSuse?

AceOfCrust, 2023-08-10:
Before i start I'm not a expert, and I'm not a developer so please keep the technical jargon to a minimum. And turning off secure boot is not a option, as again I'm not a expect and I've tried many times to turn it off....it just won't on my computer I don't know why.

Sorry if this comes off as a bit shirty, it really isn't my intention. On to the issue.

What files do I need to look for? 
What directory are they usually stored in?
What command do I use to sign the module?
Also, does anyone know any other commands for modprobe, or any other way, as the sudo modprobe razerkbd command listed doesn't seem to work.

z3ntu, 2023-08-10:
I'm not aware of any documentation that goes into how to sign DKMS-built kernel modules on openSUSE but I've never searched for it.

For Debian there's this page which goes into it: https://wiki.debian.org/SecureBoot

Perhaps you'll be able to find some help online using keywords like "dkms secure boot opensuse" or similar, it should be exactly the same process for any other DKMS module as you'll need for openrazer.

TriMoon, 2023-08-29:
@AceOfCrust maybe this page will help you to install/enable DKMS:
https://tutorialforlinux.com/2021/12/10/dkms-opensuse-installation-step-by-step-guide/2/

closed 2024-01-28 by github-actions[bot]