openrazer/openrazer #2111: Support for Razer Goliathus Extended

PeachesMLG, 2023-08-11:
Just wanted to prefix this by saying, according to the readme it says that this device is already supported, so not sure if there is something wrong on my end or a bug on your end, but Polychromatic says its not supported.

[image]

### Device Info

* Name: Razer Goliathus Extended Chroma
* Product Number: <See under device>
* Model Number: RZ02-025003

### Device Information
```
Bus 001 Device 006: ID 1532:0c02 Razer USA, Ltd Goliathus Extended
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0c02 Goliathus Extended
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Goliathus Extended Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
[... 196 lines cut ...]
 91 02 05 01 09 3B 75 10 27 FF FF 00 00 91 02 06
 00 FF 09 0A A1 02 05 0A 19 01 29 0A 95 0A 75 08
 26 FF 00 91 02 C0 C0 06 00 FF 09 22 A1 02 85 C4
 09 33 95 01 81 02 09 34 81 02 05 01 09 3B 75 10
 27 FF FF 00 00 81 02 06 00 FF 09 0A A1 02 05 0A
 19 01 29 3B 95 3B 75 08 26 FF 00 81 02 C0 C0 06
 00 FF 09 22 A1 02 85 C4 09 33 95 01 91 02 09 34
 91 02 05 01 09 3B 75 10 27 FF FF 00 00 91 02 06
 00 FF 09 0A A1 02 05 0A 19 01 29 3B 95 3B 75 08
 26 FF 00 91 02 C0 C0 06 00 FF 09 0C A1 02 85 FF
 05 01 09 3B 95 01 B1 02 09 3A A1 02 05 0A 19 01
 29 3E 95 3E B1 02 C0 C0 06 00 FF 09 0C A1 02 85
 E2 05 01 09 3B 95 01 81 02 09 3A A1 02 05 0A 19
 01 29 3E 95 3E 81 02 C0 C0 C0
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

lah7, 2023-08-11:
Most likely, the installation is not fully complete. What does it say when you press **Troubleshoot**?

PeachesMLG, 2023-08-19:
Yeah, looks like this is an error on my side, appoligies

closed 2023-08-19 by PeachesMLG