openrazer/openrazer #2112: Support for Basilisk X Hyperspeed V3

Lorkrei, 2023-08-12:
### Device Info

* Name: Basilisk X Hyperspeed V3
* Product Number: RZ01-04870100-R3G1
* Model Number: RC30-04870100-000

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```

```
Please post the output of
    ls -lah /dev/input/by-id/
```

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

Lorkrei, 2023-08-12:
sorry these are all the information i can get for this device

z3ntu, 2023-08-13:
Appears to be a duplicate of #2069 , not 100% sure because you didn't provide e.g. `lsusb | grep 1532` output.

closed 2023-08-13 by z3ntu

Lorkrei, 2023-08-13:
Hello.
I don't know how provide the data you need.
Can you tell me how?

Thanks

z3ntu, 2023-08-13:
Run the command `lsusb | grep 1532` in your terminal and copy paste the output here. But as I said probably it's a duplicate.