openrazer/openrazer #2114: 1532:020f Razer USA, Ltd Blade QHD Not - This Supported Device Was Not Detected By OpenRazer Daemon

Ndashka-Dev, 2023-08-15:
### Distribution
RisiOS 37

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I was trying to configure OpenRazer on my laptop, for the Razer Blade QHD keyboard. I have installed the `kernel-devel` package on RisiOS (a fork of Fedora), added the repository and installed `openrazer-meta` which installed `openrazer-daemon` and `openrazer-kernel-modules-dkms`. I have added myself to the `plugdev` group and restarted. I am now in the following groups: `wheel plugdev`. `lsusb | grep '1532:'` displays `Bus 001 Device 004: ID 1532:020f Razer USA, Ltd Blade QHD`. However, the contents of `ls /sys/bus/hid/drivers/` are only `hid-generic  hid-rmi  magicmouse  ntrig`. `razer-cli -l` from PyPI shows me `Found 0 Razer device(s)`. I am lost.

lah7, 2023-08-15:
Sounds like the driver didn't install correctly or couldn't be loaded. What's the output of:

    sudo dkms status
    sudo modprobe razerkbd

Might be worth checking the kernel log too:

    dmesg | grep razer

Ndashka-Dev, 2023-08-15:
> Sounds like the driver didn't install correctly or couldn't be loaded. What's the output of:
> 
> ```
> sudo dkms status
> sudo modprobe razerkbd
> ```
> 
> Might be worth checking the kernel log too:
> 
> ```
> dmesg | grep razer
> ```

For the above I got `openrazer-driver/3.6.1, 6.4.10-100.fc37.x86_64, x86_64: installed` and `modprobe: ERROR: could not insert 'razerkbd': Key was rejected by service`. `dmesg` gave me nothing...

lah7, 2023-08-15:
> modprobe: ERROR: could not insert 'razerkbd': Key was rejected by service

Sounds like [Secure Boot](https://github.com/openrazer/openrazer/wiki/Troubleshooting#3-kernel-modules-do-they-load-properly) is the culprit. Easiest option would be to turn it off in your BIOS/UEFI settings, and then it should start working. The more complicated alternate is to sign the modules yourself.

closed 2023-08-15 by Ndashka-Dev