openrazer/openrazer #2116: opnerazer library not installed in Arch Linux

AetherPrior, 2023-08-21:
### Distribution
Arch Linux version: 6.4.2-arch1-1

### Support

- [X] Have you tried rebooting?
- [X] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

- Polychromatic says that I do not have any backend installed when I run the troubleshooter, however, I have installed the backend using the `yay -S openrazer-meta` command. 
- I notice that the python version that polychromatic uses is python3.11, which also happens to be the default python version used by my system. 
The output of `python3 -c "import sys; print('\n'.join(sys.path))"` is as follows:  

```
/usr/lib/python311.zip
/usr/lib/python3.11
/usr/lib/python3.11/lib-dynload
/usr/lib/python3.11/site-packages
```
I'm not sure if yay is installing the library for a different version. I've tried reinstalling `python-openrazer` separately and then installing `openrazer-meta` but that still doesn't work. I've also tried rebooting. Can someone help? Thanks!!

z3ntu, 2023-08-21:
What's the output of `pacman -Ql python-openrazer | head` ?

AetherPrior, 2023-08-21:
```
python-openrazer /usr/
python-openrazer /usr/lib/
python-openrazer /usr/lib/python3.9/
python-openrazer /usr/lib/python3.9/site-packages/
python-openrazer /usr/lib/python3.9/site-packages/openrazer-3.6.1-py3.9.egg-info/
python-openrazer /usr/lib/python3.9/site-packages/openrazer-3.6.1-py3.9.egg-info/PKG-INFO
python-openrazer /usr/lib/python3.9/site-packages/openrazer-3.6.1-py3.9.egg-info/SOURCES.txt
python-openrazer /usr/lib/python3.9/site-packages/openrazer-3.6.1-py3.9.egg-info/dependency_links.txt
python-openrazer /usr/lib/python3.9/site-packages/openrazer-3.6.1-py3.9.egg-info/requires.txt
python-openrazer /usr/lib/python3.9/site-packages/openrazer-3.6.1-py3.9.egg-info/top_level.txt
```
good point, it's installed for python3.9, I'm not sure how to get it installed for 3.11

z3ntu, 2023-08-21:
What does the command `which python3` output where you build the AUR package?

AetherPrior, 2023-08-21:
It outputs `/usr/bin/python3` which is python3.11

AetherPrior, 2023-08-21:
Reinstalling everything, the library, driver, the dkms module and polychromatic after deactivating my environment worked. Thanks for the help!

closed 2023-08-21 by AetherPrior

z3ntu, 2023-08-21:
Seems it only took 3 months until someone hit this issue (and told me) after removing a custom check in the PKBUILD: https://aur.archlinux.org/cgit/aur.git/commit/?h=openrazer&id=2e14f5e57804c3bef8a2934816e0b2a9f8c76bf4

Let's see if I'll revert this commit afterall.

AetherPrior, 2023-08-23:
Should I reopen this issue just in case?

z3ntu, 2023-08-23:
No, I'll just monitor issues and if necessary take action.