openrazer/openrazer #2118: Support for [device name here]

SunkenHero, 2023-08-26:
### Device Info

* Name: Razer Naga Trinity
* Product Number: RZ01-02410100
* Model Number: RZ01-0241

### Device Information
```
Bus 001 Device 003: ID 1532:0067 Razer USA, Ltd Naga Trinity
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0067 Naga Trinity
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Trinity
  iSerial                 3 00000000001A
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
[... 147 lines cut ...]
 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02 95
 01 75 10 81 00 75 08 95 13 81 01 C0 05 01 09 80
 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95 03
 81 02 95 05 81 01 75 08 95 14 81 01 C0 05 01 09
 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF
 00 75 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05
 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15
 81 00 C0

001:002:000:DESCRIPTOR         1693058113.999219
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0
```

</details>

### Synapse Screenshots
[image]

### Extra

* The mouse has 3 different zones that can be controlled: Mousewheel, Logo and Numpad (addon that can be attached to the side).
* OpenRGB got Static, Breathing, Spectrum Cycle and Direct (for directly controlling the 3 zones) to work

z3ntu, 2023-08-27:
Support for this device was added about 4 years ago. Is it not working for you?

closed 2023-08-27 by z3ntu

SunkenHero, 2023-08-28:
It's working for me but I can only set a static color. I don't know if its because I am using Polychromatic but there should be a lot more options available.

z3ntu, 2023-08-28:
Then see #1433