openrazer/openrazer #2120: gimp_device_info_set_device - Basilisk Essential & Huntsman V2 TKL

0pLuS0, 2023-08-29:
### Distribution
Slackware 15.0

openrazer 3.6.1

Baslisk Essential - 1532:0065 - 632031H11805501
Huntsman V2 TKL - 1532:026B - PM2138F58600896]

### Support

- [x ] Have you tried rebooting?
- [ x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm not reporting this for support, as I don't see anything causing me problems. 

```I'm reporting this as bug, you don't have a Bug Report section```

When I run Gimp I noticed this output;

gimp_device_info_set_device: trying to set GdkDevice 'Razer Razer Basilisk Essential' on GimpDeviceInfo which already has a device
gimp_device_info_set_device: stored 'num-keys' for device 'Razer Razer Basilisk Essential' doesn't match number of keys present in device
gimp_device_info_set_device: trying to set GdkDevice 'Razer Razer Huntsman V2 Tenkeyless' on GimpDeviceInfo which already has a device
gimp_device_info_set_device: stored 'num-keys' for device 'Razer Razer Huntsman V2 Tenkeyless' doesn't match number of keys present in device

z3ntu, 2023-08-31:
So... what's the problem? Just GIMP printing a few lines which mentions the device name?

0pLuS0, 2023-09-04:
I really have no idea what this is about, that is why I posted it, especially if this is some sort of issue;

'''doesn't match number of keys present in device'''

THANKS

lah7, 2023-09-04:
It's _probably_ because our Razer hardware exposes a couple of interfaces, like 2 keyboards and 1 mouse, when the device is in "driver" mode:

```bash
$ evtest
/dev/input/event4:      Razer Razer BlackWidow Ultimate 2016
/dev/input/event5:      Razer Razer BlackWidow Ultimate 2016
/dev/input/event6:      Razer Razer BlackWidow Ultimate 2016

$ ls /dev/input/by-id/*Razer*-event*
/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate_2016-event-kbd
/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate_2016-if01-event-kbd
/dev/input/by-id/usb-Razer_Razer_BlackWidow_Ultimate_2016-if02-event-mouse
```

GIMP initializes a `GdkDevice` (using GTK 2, via `GimpDeviceInfo`), which can be seen under **Edit → Input Devices**. I'd guess that only one of these interfaces is initialized per physical hardware (and that only one of them receives input from keypresses).

 [`GdkDevice` is just describing the device](https://github.com/GNOME/gtk/blob/cecab7801e917c6085eef48624202c1be12a3291/docs/reference/gtk/input-handling.md?plain=1#L11-L25), if you'd like to read more on how GTK handles input devices. 

Seems harmless. I think [the functionality it concerns](https://docs.gimp.org/2.10/en/gimp-edit-input-devices.html) is more for advanced input like graphic tablets. _Is this "Razer" keyboard appearing as a mouse actually a tablet? Who knows?_

closed 2024-06-02 by github-actions[bot]