openrazer/openrazer #2121: Arch - Basilisk V3 Pro - 'razerkbd': Exec format error

BAlmeidaS, 2023-09-01:
### Distribution

Arch Linux
```
$ uname -srm
Linux 6.1.50-1-lts x86_64
```

### Support

- [ x ] Have you tried rebooting?
- [ ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

Polychromatic is not recognizing my Basilisk V3 Pro Wired
```
$ lsusb | grep '1532:'
Bus 009 Device 003: ID 1532:00aa Razer USA, Ltd Razer Basilisk V3 Pro
```

On troubleshooting the only red flag is related to `modprob razerkbd`:
```
$ sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Exec format error
```

My `dkms install openrazor` is the following:
```
$ sudo dkms install openrazer-driver/3.6.1
Module openrazer-driver/3.6.1 already installed on kernel 6.1.50-1-lts (x86_64), skip. You may override by specifying --force.
```

I have already tried to force reinstall with `--force` and also uninstall and reinstall:
```
$ sudo dkms remove openrazer-driver/3.6.1 --all
$ sudo dkms install openrazer-driver/3.6.1
```

z3ntu, 2023-09-01:
Check `dmesg` please, there should be a message about this.

BAlmeidaS, 2023-09-01:
First, thanks for the fastest answer I ever received in an open-source project!

Then, you are absolutely right, I got the problem on `dmesg` and I was able to fixed, so I will close this issue.

Just to keep to other people that may find the same problem. I installed `linux-headers` in Arch, but I should have installed `linux-lts-headers`, in the end a simple:

```
sudo pacman -S linux-lts-headers
```

fixed the problem