openrazer/openrazer #2122: Metaissue: Devices responding with an unknown serial number (UNKWN*) or 0% battery

lah7, 2023-09-09:
There are few identical issues popping up with the same root problem. This issue will link them.

## Cause

The daemon responds with a serial number like `UNKWN00000000XXXX` or battery status as `0%`. This might be random or consistent. 

 `Command timed out` appears in the kernel log:

    dmesg | grep razer

## Symptoms

* After reboot, persistent settings are not applied - like poll rate, brightness or effect.
* Battery notification incorrectly says the device reached 0% battery.
* Battery % is incorrect in applications.
* Rarely, some items (e.g. DPI) reports the wrong value.

## Workaround

1. [Write a script](https://github.com/openrazer/openrazer/tree/master/examples) that runs at logon to apply your desired settings.
1. Turn off persistence in `razer.conf` if the device is capable of retaining settings across power cycles **and** don't dual boot.
1. Restart `openrazer-daemon` and any open applications connected to the daemon.

## Solution

This happens because the device didn't send a valid response, so the driver generates a dummy serial to prevent applications and scripts from crashing. It's theortically possible that a handful of devices are incapable of reading their serial number, or are unreadable under certain connections (e.g. wired vs wireless)

In future, the driver will introduce a retry mechanism to send the request again.

## To check

    cat /sys/bus/hid/drivers/razer*/*1532*/device_serial

Serials should not start with `UNKWN`. If you're affected, please let us know which devices are intermittent or always wrong - just in case there is a problem with that device's implementation.

There is one Razer device with a serial that literally said `As printed in the D cover`.
* https://github.com/openrazer/openrazer/issues/192

## Affected Devices & Issues

* https://github.com/openrazer/openrazer/issues/2048 (DeathAdder V2 X Hyperspeed)
* https://github.com/openrazer/openrazer/issues/2076 (Viper Ultimate)
* https://github.com/openrazer/openrazer/issues/2101 (Naga Pro Wireless)
* https://github.com/openrazer/openrazer/issues/2109 (Naga Pro)
* https://github.com/polychromatic/polychromatic/issues/493 (Viper V2 Pro)
* https://github.com/openrazer/openrazer/issues/1367

h0x32, 2023-12-24:
I can confirm same behavior on Orochi V2:
1532:0094 Razer USA, Ltd Orochi V2

Adam-Englebright, 2024-01-31:
I had this issue using the DeathAdder V3 Pro (Wireless). However, for me the issue only occurs if I don't touch my mouse during boot up (so it is sleeping), then move the mouse once I am at the desktop. If I continually move my mouse during boot up, so that it tires to connect to the USB dongle as soon as possible, I seem to avoid this issue.

pdf, 2024-02-15:
Device: 0003:1532 RazerViperUltimateWireless / RazerViperUltimateWired

When both the wireless dongle and the wired cable are connected (e.g. when charging), only one of the devices reports a serial number via sysfs, the other is blank.

When plugging in the second device, the daemon throws an exception:

```
openrazer-daemon[1423]: Exception in thread Thread-14 (_collecting_udev_method):
openrazer-daemon[1423]: Traceback (most recent call last):
openrazer-daemon[1423]:   File "/usr/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
openrazer-daemon[1423]:     self.run()
openrazer-daemon[1423]:   File "/usr/lib/python3.11/threading.py", line 982, in run
openrazer-daemon[1423]:     self._target(*self._args, **self._kwargs)
openrazer-daemon[1423]:   File "/usr/lib/python3.11/site-packages/openrazer_daemon/daemon.py", line 605, in _collecting_udev_method
openrazer-daemon[1423]:     self._add_device(d)
openrazer-daemon[1423]:   File "/usr/lib/python3.11/site-packages/openrazer_daemon/daemon.py", line 530, in _add_device
openrazer-daemon[1423]:     razer_device = device_class(device_path=sys_path, device_number=device_number, config=self._config,
openrazer-daemon[1423]:                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
openrazer-daemon[1423]:   File "/usr/lib/python3.11/site-packages/openrazer_daemon/hardware/device_base.py", line 136, in __init__
openrazer-daemon[1423]:     super().__init__(object_path)
openrazer-daemon[1423]:   File "/usr/lib/python3.11/site-packages/openrazer_daemon/dbus_services/service.py", line 57, in __init__
openrazer-daemon[1423]:     self.add_to_connection(bus, object_path)
openrazer-daemon[1423]:   File "/usr/lib/python3.11/site-packages/dbus/service.py", line 579, in add_to_connection
openrazer-daemon[1423]:     connection._register_object_path(path, self._message_cb,
openrazer-daemon[1423]: KeyError: "Can't register the object-path handler for '/org/razer/device/PM1234H12345678': there is already a handler"
```

So it seems the daemon does see the serial.

Subsequent (frequent) battery notifications display `0%` , with timeouts in dmesg.

Plugging in only wired or only wireless dongle results in the single device successfully reporting a serial correctly.

z3ntu, 2024-03-03:
> openrazer-daemon[1423]: KeyError: "Can't register the object-path handler for '/org/razer/device/PM1234H12345678': there is already a handler"

We don't support connecting a device two times, as you can see it breaks some assumptions OpenRazer has. We probably need to fix this in the future but it's not a simple fix as everything relies on the serial number being unique.

NeilGirdhar, 2024-05-13:
Linking related PR: https://github.com/openrazer/openrazer/pull/2248

chuckie987, 2024-07-10:
Hi Team,
Having this issue, not with the battery % but the polling rate, DPI and colour settings never being persistent and creating a new "UNKWN" serial number every boot

Running Mint 21 and/or Mint 22

[UNKWN000000003042]
dpi_x = 1400
dpi_x = 1400
poll_rate = 1000
logo_active = True
logo_brightness = 75
logo_effect = spectrum
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1

[UNKWN000000002007]
dpi_x = 1400
dpi_x = 1400
poll_rate = 1000
logo_active = True
logo_brightness = 75
logo_effect = spectrum
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1

[UNKWN000000000178]
dpi_x = 1400
dpi_x = 1400
poll_rate = 1000
logo_active = True
logo_brightness = 75
logo_effect = spectrum
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1

[UNKWN000000003444]
dpi_x = 1400
dpi_y = 1400
poll_rate = 1000
logo_active = True
logo_brightness = 75
logo_effect = spectrum
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1

CKr0nzz, 2024-07-10:
I'm having the same issue with Razer Basilisk Ultimate, settings are not persistent. I have to manually reset the poll rate and brightness level at every reboot.

Adam-Englebright, 2024-07-10:
@chuckie987 @CKr0nzz 
Have you tried moving the mouse around when booting? This still "fixes" the issue for me using the DeathAdder V3 Pro (Wireless). If I first move the mouse after booting up and logging in, then I'll get an UNKWN serial and default settings.

chuckie987, 2024-07-11:
> @chuckie987 @CKr0nzz Have you tried moving the mouse around when booting? This still "fixes" the issue for me using the DeathAdder V3 Pro (Wireless). If I first move the mouse after booting up and logging in, then I'll get an UNKWN serial and default settings.

I tried giving this a go constantly moving the mouse before during and after boot but didnt help with my Viper Ultimate, Currently Running Mint 22
Daemon 3.8.0
DKMS 3.0.11

CKr0nzz, 2024-07-11:
I tried moving the mouse around (It's always on the dock usually when I booth) but nothing. I checked the log file, it gives me that:
2024-07-11 08:36:34 | razer                          | INFO     | Initialising Daemon (v3.8.0). Pid: 5205
2024-07-11 08:36:34 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2024-07-11 08:36:34 | razer                          | INFO     | Found device.0: 0003:1532:007E.000A
2024-07-11 08:36:34 | razer.device0                  | INFO     | Initialising device.0 RazerMouseDock
2024-07-11 08:36:34 | razer                          | INFO     | Found device.1: 0003:1532:0088.0007
2024-07-11 08:36:34 | razer.device1                  | INFO     | Initialising device.1 RazerBasiliskUltimateReceiver
2024-07-11 08:36:34 | razer.device1                  | WARNING  | malformed serial: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
2024-07-11 08:36:35 | razer.device1                  | WARNING  | malformed serial: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
2024-07-11 08:36:35 | razer.device1                  | WARNING  | malformed serial: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
2024-07-11 08:36:35 | razer.device1                  | WARNING  | malformed serial: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
2024-07-11 08:36:35 | razer.device1                  | WARNING  | malformed serial: 'utf-8' codec can't decode byte 0xff in position 0: invalid start byte
2024-07-11 08:36:35 | razer                          | INFO     | Serving DBus

That's regardless if the "restore software settings" is checked or not.

I also noticed that the mouse keeps the settings from the previous shutdown, it only resets when the daemon boots up. Uninstalling might be the solution haha.

m0mosenpai, 2024-07-24:
Can confirm the issue happens on DeathAdder V3 Pro Wireless after a reboot. `(openrazer-driver/3.8.0, 6.10.0-arch1-2, x86_64)`. Disconnecting and reconnecting the dongle seems to fix it for the rest of the session. You can see it get the malformed serial number in the logs till you reconnect and the errors stop.

While it's in the malformed state, the mouse will goto sleep almost instantly, once you stop using it. Moving it again for a couple of seconds gets it connected again and working fine as long as you keep moving it.

Adam-Englebright, 2024-07-27:
I've just discovered that this issue occurs if I switch from wired mode (for charging) to wireless mode during general PC usage, causing my DPI etc. to be reset to defaults. However, the issue does not occur when switching from wireless mode to wired mode.

Now on Arch using `v3.8.0`.

lah7, 2024-10-08:
@m0mosenpai, Of course, you're welcome to give it a try!  I believe the long term fix would be to add a retry mechanism at a driver level. Something that can retry failed requests another X times, maybe with time apart, before giving up.

m0mosenpai, 2024-10-09:
@lah7 yeah makes sense. I'll have a look at it then, thanks! Also, is there a community chat where I could discuss navigating through this issue with the maintainers?

lah7, 2024-10-09:
@m0mosenpai Yes, there is a community chat room via [Telegram](https://t.me/joinchat/AMhHjj9Txhkhn8JM5_LeMg) or [Matrix](https://matrix.to/#/#openrazer:matrix.org).

Sazzels, 2025-01-08:
i just bought the Viper V3 Pro and i'm facing the same issues, settings are not persistent after reboot (when the mouse is powered off in that time). Reboots while the mouse is on are not a problem but i notice the settings are reset after a reboot the next day. 

```
openrazer-daemon 3.9.0-2
OS: EndeavourOS Linux x86_64
Kernel: 6.12.8-arch1-1
```

`persistence.conf` creates some UNKWN entries. (From RazerGenie, the device should be PM2435H32105368)
```
[PM2435H32105368]
dpi_x = 800
dpi_y = 800
poll_rate = 2000

[UNKWN000000001695]
dpi_x = 1800
dpi_y = 1800
poll_rate = 1000

[UNKWN000000003828]
dpi_x = 1000
dpi_y = 1000
poll_rate = 1000

[UNKWN000000001077]
dpi_x = 1000
dpi_y = 1000
poll_rate = 2000

[UNKWN000000002322]
dpi_x = 1000
dpi_y = 1000
poll_rate = 2000
```

`razer.conf`
```
[General]
verbose_logging = False

[Startup]
sync_effects_enabled = False
devices_off_on_screensaver = True
battery_notifier = True
battery_notifier_freq = 900
battery_notifier_percent = 20
restore_persistence = True
persistence_dual_boot_quirk = False
```

z3ntu, 2025-02-15:
@Sazzels With #2220 these unknown serial numbers should be at least somewhat more predictable, so no longer lots of random ones, in theory at least.

Sazzels, 2025-02-17:
> [@Sazzels](https://github.com/Sazzels) With [#2220](https://github.com/openrazer/openrazer/pull/2220) these unknown serial numbers should be at least somewhat more predictable, so no longer lots of random ones, in theory at least.

i´m currently on version `openrazer-daemon 3.9.0-2` and i have no issues anymore. i still left the files `razer.conf & persistence.conf` on read-only. All settings are correctly set on login and are persistent after reboot. 

i have no more issues so far.

quietvoid, 2025-02-27:
I'm having issues with the latest version, the persistence config keeps adding generated serial numbers.
I'm my setup I use a USB hub and switcher, making the mouse reconnect multiple times over a day.

When the generated serial number is new (incremented), the settings are completely reset.
If I do delete `persistence.conf`, then the settings persist until it's been X number of times disconnected.