openrazer/openrazer #2123: Support for Razer Blackwidow V4 75%

vihu, 2023-09-09:
### Product link
https://www.razer.com/gaming-keyboards/Razer-BlackWidow-V4-75/RZ03-05000200-R3U1

### Device Info

* Name: Razer Blackwidow V4 75%
* Product Number: RZ03-05000200-R3U1
* Model Number: RZ03-0500

### Device Information

```
Bus 003 Device 007: ID 1532:02a5 Razer USA, Ltd Razer BlackWidow V4 75%
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x02a5 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V4 75%
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
[... 186 lines cut ...]
 10 81 00 75 08 95 13 81 01 C0 05 01 09 80 A1 01
 85 03 19 81 29 83 15 00 25 01 75 01 95 03 81 02
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

003:007:000:DESCRIPTOR         1694287762.129439
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

```

### Packet Captures

I don't have packet capture setup yet. 

### Synapse Screenshots

[image]

[image]

o-lenczyk, 2024-01-13:
Hi
Can someone guide me how to add pcap?

4sa617, 2024-04-09:
Checking in to see if there are any plans to support this device in the future.

z3ntu, 2024-04-09:
@4sa617 Somebody will need to add support for the device. This doc should outline basically what needs to be done: https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

PR #2287 (2024-09-03): Support for BlackWidow V4 75%
Fixes #2123 

Polychromatic loads it fine, every setting seems to work. Example scripts works fine too. Also, it could probably just inherit the base BlackWidow V4 as they seem to be very similar.
closed 2024-09-03 by z3ntu