openrazer/openrazer #2124: openrazer-meta removed from AUR?

boooooooaq, 2023-09-14:
Today i ran yay -Sua like I  sometimes do to check for updates concerning AUR packages and got: ` -> Packages not in AUR: openrazer-meta`

So I checked the AUR and indeed the package is not to be found, should I from now on use openrazer-meta-git?
Or is it just a mistake that it's removed?

lah7, 2023-09-14:
That's expected - OpenRazer is now in the [official repositories](https://archlinux.org/packages/?sort=&q=openrazer) for Arch Linux! 

* [openrazer-daemon](https://archlinux.org/packages/extra/any/openrazer-daemon/)
* [openrazer-driver-dkms](https://archlinux.org/packages/extra/any/openrazer-driver-dkms/)
* [python-openrazer](https://archlinux.org/packages/extra/any/python-openrazer/)

 `openrazer-meta` can be safely removed, and `openrazer-daemon` should be marked explicitly installed so the other OpenRazer packages are not seen as orphans. `openrazer-meta` was just a _metapackage_ to easily install all the three packages when it was in the AUR.

    sudo pacman -R openrazer-meta
    sudo pacman -D --asexplicit openrazer-daemon

Future updates will work with the rest of the system. That does mean that there _could_ be a delay between a release here and the package being available, since we're not distro packagers.

[`openrazer-meta-git`](https://aur.archlinux.org/packages/openrazer-meta-git) is still around if any one needs to build the latest branch, or any other branch for testing.

boooooooaq, 2023-09-14:
Alright thanks for the info!

closed 2023-09-14 by boooooooaq