openrazer/openrazer #2128: DeathStalker V2 Pro TKL (Wireless) (1532:0296) is shown as unsupported

Adrian-Bielefeldt, 2023-09-15:
### Distribution

Linux Mint 21 Cinnamon

### Support

- [x] Have you tried rebooting?
- [x] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I have installed OpenRazer as well as both Polychromatic and RazerGenie. Neither recognizes my keyboard. It does however show up in lsusb (as Bus 001 Device 004: ID 1532:0296 Razer USA, Ltd  DSV2Pro TKL) and is also shown as an unsupported device in Polychromatic.

As fas as I can see this keyboard should however be supported by OpenRazer. What could I be missing?

lah7, 2023-09-16:
The keyboard has been supported since the v3.6.0 release. Did you install the software [from the PPA](https://openrazer.github.io/#ubuntu)? The one provided in the official Ubuntu/Linux Mint repos tends to be an older version.

Adrian-Bielefeldt, 2023-09-16:
Installing from the PPA solved the issue. Thank you very much.
Maybe it's just me, but if others run into the same issue you might want to add a line to the troubleshooting guide about how the version you can get from the OS repositories might be quite a bit out of date.

closed 2023-09-16 by Adrian-Bielefeldt