openrazer/openrazer #2130: Support for Blackwidow V4 X

0x0015, 2023-09-16:
### Device Info

* Name: Razer Blackwidow V4 X
* Product Number: RZ03-04701900-R3U1
* Model Number: RZ03-0470

### Device Information
```
$ lsusb -d 1532: -v
Bus 001 Device 005: ID 1532:0293 Razer USA, Ltd Razer BlackWidow V4 X
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0293 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V4 X
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x006d
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     157
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0016  1x 22 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              2 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     348
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0002  1x 2 bytes
        bInterval               1
```

```
$ ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 200 Sep 16 09:45 .
drwxr-xr-x 4 root root 600 Sep 16 09:45 ..
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Logitech_USB_Receiver-if01-event-mouse -> ../event2
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Logitech_USB_Receiver-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Razer_Razer_BlackWidow_V4_X-event-if01 -> ../event5
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Razer_Razer_BlackWidow_V4_X-event-if03 -> ../event7
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Razer_Razer_BlackWidow_V4_X-event-kbd -> ../event3
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Razer_Razer_BlackWidow_V4_X-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Razer_Razer_BlackWidow_V4_X-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Sep 16 09:45 usb-Razer_Razer_BlackWidow_V4_X-if02-mouse -> ../mouse1
```

<details>

```
(root)$ usbhid-dump -m 1532 -ed
001:005:003:DESCRIPTOR         1694903253.163770
 05 59 09 01 A1 01 85 01 09 02 A1 02 09 03 15 00
 27 FF FF 00 00 75 10 95 01 B1 03 09 04 09 05 09
 06 09 07 09 08 15 00 27 FF FF FF 7F 75 20 95 05
 B1 03 C0 85 02 09 20 A1 02 09 21 15 00 27 FF FF
 00 00 75 10 95 01 B1 02 C0 85 03 09 22 A1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 01 B1 02 09 23
 09 24 09 25 09 27 09 26 15 00 27 FF FF FF 7F 75
 20 95 05 B1 02 09 28 09 29 09 2A 09 2C 09 2D 15
 00 26 FF 00 75 08 95 05 B1 02 C0 85 04 09 50 A1
 02 09 03 09 55 15 00 25 08 75 08 95 02 B1 02 09
 21 15 00 27 FF FF 00 00 75 10 95 08 B1 02 09 51
 09 52 09 53 09 51 09 52 09 53 09 51 09 52 09 53
 09 51 09 52 09 53 09 51 09 52 09 53 09 51 09 52
 09 53 09 51 09 52 09 53 09 51 09 52 09 53 15 00
 26 FF 00 75 08 95 18 B1 02 C0 85 05 09 60 A1 02
 09 55 15 00 25 08 75 08 95 01 B1 02 09 61 09 62
 15 00 27 FF FF 00 00 75 10 95 02 B1 02 09 51 09
 52 09 53 15 00 26 FF 00 75 08 95 03 B1 02 C0 85
 06 09 70 A1 02 09 71 15 00 25 01 75 08 95 01 B1
 02 C0 85 07 06 00 FF 15 00 26 FF 00 09 02 75 08
 95 3F B1 02 C0 05 0C 09 01 A1 01 85 08 19 D0 29
 D7 15 00 25 01 75 01 95 08 81 02 C0

001:005:002:DESCRIPTOR         1694903253.167759
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:005:001:DESCRIPTOR         1694903253.172763
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 29 A0 15 00 25 01
 75 01 95 A0 81 02 C0 05 0C 09 01 A1 01 85 02 19
 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81 00 75
 08 95 13 81 01 C0 05 01 09 80 A1 01 85 03 19 81
 29 83 15 00 25 01 75 01 95 03 81 02 95 05 81 01
 75 08 95 14 81 01 C0 05 01 09 00 A1 01 85 04 09
 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81
 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00 26 FF
 00 35 00 46 FF 00 75 08 95 15 81 00 C0

001:005:000:DESCRIPTOR         1694903253.175763
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

```

</details>

Presumably it's relatively similar to Blackwidow V4 75% (already has an issue open), and Blackwidow V4 Pro (already has a PR open), but I thought it should be opened nonetheless.

z3ntu, 2023-09-21:
Have a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md , if you like to tinker I'm sure you can add support for the device yourself!

0x0015, 2023-10-30:
Hello,

I'm not sure if I've done something wrong in setting up my system but I've basically bootlegged [this](https://github.com/openrazer/openrazer/commit/1278b22b67e0288c58cda9a1eb1145ef3a3f6c85) commit, as I assume the two keyboards being of the same family, same effects etc... (as I can tell only difference really is the lack of the knob the pro has).

<details>
<summary>Here's the daemon log</summary>

```
Starting daemon.
2023-10-30 01:40:35 | razer                          | INFO     | Initialising Daemon (v3.6.1). Pid: 14062
2023-10-30 01:40:35 | razer.screensaver              | INFO     | Initialising DBus Screensaver Monitor
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus1800 ------------------------ (1532:0020)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus2000 ------------------------ (1532:005E)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEliteDVaEdition ------------- (1532:006A)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusEssential ------------------- (1532:006B)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAbyssusV2 -------------------------- (1532:005B)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerAtherisReceiver -------------------- (1532:0062)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationChroma ------------------ (1532:0F08)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBaseStationV2Chroma ---------------- (1532:0F20)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasilisk --------------------------- (1532:0064)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskEssential ------------------ (1532:0065)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateReceiver ----------- (1532:0088)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskUltimateWired -------------- (1532:0086)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV2 ------------------------- (1532:0085)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3 ------------------------- (1532:0099)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWired ----------------- (1532:00AA)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskV3ProWireless -------------- (1532:00AB)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBasiliskXHyperSpeed ---------------- (1532:0083)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2019 --------------------- (1532:0241)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaOverwatch ---------- (1532:0211)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaV2 ----------------- (1532:0221)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowElite -------------------- (1532:0228)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowEssential ---------------- (1532:0237)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowLite --------------------- (1532:0235)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealth ------------------ (1532:011B)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowStealthEdition ----------- (1532:010E)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowTournamentEdition2014 ---- (1532:011C)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2012 ------------- (1532:010D)
2023-10-30 01:40:35 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowUltimate2013 ------------- (1532:011A)
[... cut ...]
[ 2583.391623] input: Razer Razer BlackWidow V4 X as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.2/0003:1532:0293.0013/input/input32
[ 2583.445841] razerkbd 0003:1532:0293.0013: input,hidraw2: USB HID v1.11 Mouse [Razer Razer BlackWidow V4 X] on usb-0000:00:06.0-1/input2
[ 2583.469089] input: Razer Razer BlackWidow V4 X as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.3/0003:1532:0293.0014/input/input33
[ 2583.533673] razerkbd 0003:1532:0293.0014: input,hiddev96,hidraw3: USB HID v1.11 Device [Razer Razer BlackWidow V4 X] on usb-0000:00:06.0-1/input3
```

</details>

z3ntu, 2023-10-30:
Permission denied might either be that the sysfs file doesn't get created by the driver (but the daemon wants to use it), or it's that our udev script didn't run properly. I'm assuming the first but you can run `ls -al` on that file and see if it actually exists.

But probably best if you submit a PR if you have something working where you can e.g. control the RGB and then already the CI will tell you if some things are wrong.

0x0015, 2023-11-04:
You were absolutely on the mark, I missed installing the udev rules.

now I'm getting Dmesg "Device data transfer failed" messages when I try to make changes to the settings (both brightness and effect type).
<details>
<summary>Dmesg</summary>

```
[ 2588.089612] usb 1-1: new full-speed USB device number 3 using ohci-pci
[ 2588.807536] usb 1-1: New USB device found, idVendor=1532, idProduct=0293, bcdDevice= 2.00
[ 2588.807544] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2588.807546] usb 1-1: Product: Razer BlackWidow V4 X
[ 2588.807548] usb 1-1: Manufacturer: Razer
[ 2588.820439] input: Razer Razer BlackWidow V4 X as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.0/0003:1532:0293.0005/input/input13
[ 2588.876343] razerkbd 0003:1532:0293.0005: input,hidraw0: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4 X] on usb-0000:00:06.0-1/input0
[ 2588.890521] input: Razer Razer BlackWidow V4 X as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.1/0003:1532:0293.0006/input/input14
[ 2588.946043] razerkbd 0003:1532:0293.0006: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer BlackWidow V4 X] on usb-0000:00:06.0-1/input1
[ 2588.975176] input: Razer Razer BlackWidow V4 X as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.2/0003:1532:0293.0007/input/input15
[ 2589.034302] razerkbd 0003:1532:0293.0007: input,hidraw2: USB HID v1.11 Mouse [Razer Razer BlackWidow V4 X] on usb-0000:00:06.0-1/input2
[ 2589.058323] input: Razer Razer BlackWidow V4 X as /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.3/0003:1532:0293.0008/input/input16
[ 2589.116098] razerkbd 0003:1532:0293.0008: input,hiddev96,hidraw3: USB HID v1.11 Device [Razer Razer BlackWidow V4 X] on usb-0000:00:06.0-1/input3
[ 2632.838852] razer driver: Device data transfer failed.
[ 2642.860526] razer driver: Device data transfer failed.
[ 2663.739480] usb 1-1: USB disconnect, device number 3
[ 2664.611424] usbhid 1-1:1.0: Razer Device disconnected
[ 2664.714766] usbhid 1-1:1.1: Razer Device disconnected
[ 2664.811335] usbhid 1-1:1.2: Razer Device disconnected
[ 2664.899161] usbhid 1-1:1.3: Razer Device disconnected
```

</details>

<details>
<summary>Daemon output (it's a long one)</summary>

```
2023-11-04 17:36:01 | razer                          | DEBUG    | Emitted Device Added Signal
2023-11-04 17:36:01 | razer                          | DEBUG    | DBus called get_serial_list
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_device_type
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_device_name
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_device_name
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_current_effect
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_current_effect_colors
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_brightness
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_poll_rate
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_supported_poll_rates
2023-11-04 17:36:01 | razer.device0                  | DEBUG    | DBus call get_firmware
2023-11-04 17:36:06 | razer                          | DEBUG    | State recently changed, writing to disk
2023-11-04 17:36:06 | razer                          | DEBUG    | Writing persistence config
2023-11-04 17:36:12 | razer.device0                  | DEBUG    | DBus call set_none_effect
2023-11-04 17:36:12 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4X:UNKWN000000000180, 'setNone')
2023-11-04 17:36:12 | razer.device0                  | DEBUG    | Set persistence (backlight, effect, none)
2023-11-04 17:36:16 | razer                          | DEBUG    | State recently changed, writing to disk
2023-11-04 17:36:16 | razer                          | DEBUG    | Writing persistence config
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | DBus call set_brightness
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Set persistence (backlight, brightness, 74)
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4X:UNKWN000000000180, 'setBrightness', 190)
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | DBus call set_brightness
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Set persistence (backlight, brightness, 72)
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4X:UNKWN000000000180, 'setBrightness', 184)
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | DBus call set_brightness
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Set persistence (backlight, brightness, 68)
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4X:UNKWN000000000180, 'setBrightness', 175)
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | DBus call set_brightness
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Set persistence (backlight, brightness, 64)
2023-11-04 17:36:17 | razer.device0                  | DEBUG    | Sending observer message: ('effect', RazerBlackWidowV4X:UNKWN000000000180, 'setBrightness', 164)
[... cut ...]
2023-11-04 17:37:14 | razer                          | DEBUG    | Device event [unbind]: /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.3/0003:1532:0293.0008
2023-11-04 17:37:14 | razer                          | DEBUG    | Device event [remove]: /devices/pci0000:00/0000:00:06.0/usb1/1-1/1-1:1.3/0003:1532:0293.0008
```

</details>

It's pretty resonable to assume that this is because I basically copied the device configuration from #2054.  I need to go take some actual packet captures to figure their differences.

At least it shows up in razergenie :)

z3ntu, 2023-11-06:
Sounds like you might need to add your device to a different switch branch inside https://github.com/openrazer/openrazer/blob/master/driver/razerkbd_driver.c#L261 , then at least the serial number call should work correctly.

dvdred, 2024-02-16:
Hello,
i upvote this issue, still not working on ubuntu 22.04, with polychromatic 0.8.3:
[image]

z3ntu, 2024-03-03:
@dvdred Nobody added support for this device yet, so yes it's not supported yet.

If you have some time to tinker around, please read https://github.com/openrazer/openrazer/issues/2130#issuecomment-1730348658

closed 2024-07-02 by z3ntu