openrazer/openrazer #2132: Razer Baslisk Essential - DPI Speeds Are Fast

0pLuS0, 2023-09-19:
### Distribution

Slackware 15.0

openrazer 3.6.1
Razer Baslisk Essential - VID:PID1532:0065 - Serial:632031H11805501

### Support

- [ x] Have you tried rebooting?
- [x ] Have you looked at the [Troubleshooting](https://github.com/openrazer/openrazer/wiki/Troubleshooting) page?

I'm running Dual Boot, Windows 11 Home 22H2 & Slackware 15.0

In Slackware I am using 400 DPI for the mouse, but when I move it around, the cursor moves almost 2x as fast as it should compared to Windows.

400 DPI in Slackware feels like 700-800 DPI in Windows.

Why does a lower DPI in Linux run faster than in Windows?

I also use polychromatic 0.8.1 and the Y Axis always shows as 0 instead of 400. The X Axis only shows as 400.

[image]

But when I look at the `/.config/openrazer/persistence.conf it shows both the Y & X as 400, so I assumed the Y Axis in polychromatic should be on 400 and not 0.

```
[632031H11805501]
dpi_x = 400
dpi_y = 400
poll_rate = 500
logo_active = True
logo_brightness = 0
logo_effect = none
logo_colors = 0 255 0 0 255 255 0 0 255
logo_speed = 1
logo_wave_dir = 1
```

lah7, 2023-09-19:
> Why does a lower DPI in Linux run faster than in Windows?

You may want to check your desktop environment's mouse settings (if any) and/or the Windows settings too. The two that would affect this are:

* Acceleration profile (flat vs adaptive) 
* Pointer speed

Both of those would affect the feel of the cursor movement.

---

> I also use polychromatic 0.8.1 and the Y Axis always shows as 0 instead of 400. The X Axis only shows as 400.

My mouse (Mamba TE) occasionally shows the wrong Y value too - does it work again if you click on "Razer Basilisk Essential" in the sidebar again (or 2, 3 times)?  This might be related to #2122 where the hardware isn't responding correctly.

0pLuS0, 2023-09-21:
Hi,

I only have installed/run Openbox in Slackware, I have nothing setup for the mouse, only these xorg.conf settings, that is all.

Section "InputDevice"
    # generated from default
    Identifier     "Mouse0"
    Driver         "mouse"
    Option         "Protocol" "auto"
    Option         "Device" "/dev/psaux"
    Option         "ZAxisMapping" "4 5"
EndSection

Not sure what ```Sidebar``` you are refering to?  If I click on 400 a few times, both Y & X show as 400, then I close and reopen polychromatic and Y goes back to 0.

THANKS

lah7, 2023-09-21:
> Not sure what Sidebar you are refering to? If I click on 400 a few times, both Y & X show as 400, then I close and reopen polychromatic and Y goes back to 0.

The device list:
[image]
_(Or click between difference devices - some way to refresh the page while the app is running)_

I would guess Razer firmware just acting buggy. When you open the app, it makes a lot of requests _(get name, serial, brightness, DPI, poll rate, etc)_ so that might be the cause (overwhelming the hardware = erratic outputs). We'll have to see after addressing #2122.

Another way to check is to query the driver file directly:

    cat /sys/bus/hid/drivers/razermouse/*1532*/dpi

Be good to know if it is the hardware **always** saying Y = 0, but software is masking it - or if it is randomly happening but too often.

----

Regarding Xorg/Openbox, I'm not familiar with these configs, so I guess your cursor isn't using any _enhancements_ and will be the raw deal. KDE for example would have these - similar to Windows:

[image]

It worth noting that **screen resolution** can make a difference.  For me, **1800 DPI** on 4K would be too fast for 1080p (in which, I'd press down for **800 DPI**). That could be a reason for feeling a difference, but only if you use different screen resolutions between your dual boot.

0pLuS0, 2023-09-22:
Clicking the mouse name in the side bar does nothing. Then I clicked the keyboard, then the mouse and tried this back and forth, also nothing.

cat /sys/bus/hid/drivers/razermouse/*1532*/dpi 
400:0

No, I'm not using any mouse enhancements in Openbox.

My screen resolution in Windows and Slackware is the same - 2560x1440.

lah7, 2023-09-22:
cat /sys/bus/hid/drivers/razermouse/*1532*/dpi
    400:0

Turns out the hardware doesn't support the Y axis. From #1535, it was implemented like that. From the screenshots in #1134, Razer Synapse doesn't show an option to set the Y value. So, Y=0 is expected.

With how OpenRazer and apps are designed, they expect X+Y, so your device is an edge case. There _some_ devices that only support X values, but they are "fixed" in that they only have specific values to choose from (since yours is variable, it's not that kind of device)

What I noted for Polychromatic to do is disable/hide that axis for this particular mouse. Please disregard my previous comments about read errors or needing to refresh.

----

Regarding the issue's original problem - I think it's just software differences between Windows & Linux (under raw Xorg). If you're sure that the DPI is not set correctly - maybe the hardware says it's 400 but it's actually 800 - you could add a line to your startup script workaround https://github.com/openrazer/openrazer/issues/2066#issuecomment-1730594431 to enforce 400 DPI.

```diff
 for device in devman.devices:
     if "Basilisk" in device.name:
+        device.dpi = (400,0)             # or try (400,400) 
         device.fx.misc.logo.none()
```

0pLuS0, 2023-09-22:
I am 100% correct that the DPI speed is faster in Slackware.

400 DPI in Slackware is like 700-800 in Windows.

I tried both ```device.dpi = (400,0)``` & ```device.dpi = (400,400)``` it doesn't change anything and polychromatic still shows the Y Axis on 0, if I use (400,400), wasn't sure if the (400,400) would change the Y Axis to show 400 in polychromatic.

THANKS

0pLuS0, 2024-05-24:
I  hope this isn't go into the dust bin, because my 400 DPI is still like 800 DPI in Windows.

Hopefully this can be fixed?

THANKS

0pLuS0, 2024-05-24:
The last we were working on this, I have these as my Startup script.

#!/usr/bin/python3
import time
import openrazer.client
devman = openrazer.client.DeviceManager()

while len(devman.devices) < 2:
    time.sleep(0.5)
    devman = openrazer.client.DeviceManager()

for device in devman.devices:
    if "Basilisk" in device.name:
        device.dpi = (400,400)
        device.fx.misc.logo.none()

    if "Huntsman" in device.name:
        device.fx.static(0, 255, 0)

0pLuS0, 2024-05-25:
Hi @lah7 I'm using 2560x1440p on both OS. I don't follow what scaling you are referring to?

The feel is fine, it's not a problem, I'm not concerned with looks/feel/cosmetics, I'm only reporting this to try and correct this is all.

THANKS

lah7, 2024-05-25:
> I'm using 2560x1440p on both OS. I don't follow what scaling you are referring to?

For display scaling I was referring to this:

| Windows | e.g. KDE |
| --- | --- |
| [image] | [image]

The global scale of user interface elements. It might not be  directly related, but worth confirming since that might impact the distance for the cursor to travel.

I just remembered you said you're using OpenBox on Xorg, so I'd guess it's the default (100%).

----

> I also don't know what you mean by dual boot correctly apply the right feel at login?

I mean that when `openrazer-daemon` starts -- _or rather, the custom login script_ -- the mouse is set to the intended value that works for you. I'd assume over on Windows, Razer Synapse does the same thing. So, booting Slackware (400 dpi) + booting Windows (1000 dpi) = same result? 

For all we know, Synapse _might_ have "tuned" the DPI slider for this mouse and OpenRazer only knows a "raw" number.  One way to confirm that theory is to set a different DPI in Windows, reboot to Slackware and read the DPI value **without** persistence or a start-up script setting DPI. If it's the same value, there isn't a discrepancy with the value / firmware itself.

I suspect it's all to do with software mouse acceleration -  under OpenBox (not to do with OpenRazer).  Like my experience with moving a mouse cursor in Wayland is a slower, jarring feeling but feels great under Xorg. 

Or it could be Xorg's config file. In my log `/var/log/Xorg.0.log` I can see:

```
[     6.110] (II) XINPUT: Adding extended input device "Logitech G203 LIGHTSYNC Gaming Mouse Keyboard" (type: MOUSE, id 11)
[     6.110] (**) Option "AccelerationScheme" "none"
[     5.917] (**) Logitech G203 LIGHTSYNC Gaming Mouse: (accel) selected scheme none/0
[     5.917] (**) Logitech G203 LIGHTSYNC Gaming Mouse: (accel) acceleration factor: 2.000
[     5.917] (**) Logitech G203 LIGHTSYNC Gaming Mouse: (accel) acceleration threshold: 4
```

 I'd check in that direction. Like, I don't know if this [in your config](https://github.com/openrazer/openrazer/issues/2132#issuecomment-1728686685) is correct:

        Option         "Device" "/dev/psaux"

Isn't `/dev/psaux` for a PS/2 mouse? Should it point to a USB mouse? (I don't know the answer to that)

lah7, 2024-05-25:
By the way, is the poll rate (e.g. 500 Hz) set the same in Slackware and Windows? That's a possible cause of different speeds.

0pLuS0, 2024-05-25:
I don't use Scaling...

If you are referring to X I don't have a custom login script. I log in at the console; user > pass > startx

I've used the same Poll Rates for both Slack and Windows.

THANKS

lah7, 2024-05-26:
What's the polling rate set to?

It looks like 1000 Hz under the Synapse 3 about box in your screenshot:
[image]

Assuming it's also 1000 Hz in OpenRazer? (Your [persistence config](https://github.com/openrazer/openrazer/issues/2132#issue-1902090556) from last year reads 500).

---

I guess to rule out DPI values having discrepancies between the two is to try:

1. In Slackware, disable persistence in `~/.config/openrazer/razer.conf` and [disable/comment out that start-up script.](https://github.com/openrazer/openrazer/issues/2132#issuecomment-2128481793)
   * For this test, we don't want anything "setting" values. We want to know what the hardware says.
3. With OpenRazer, set DPI to an unusual value you wouldn't use, like 1250 dpi.
4. Reboot into Windows, check Synapse 3's DPI value.
     * It might automatically apply whatever was set before in Windows, which is fine.
5. With Synapse 3, set the DPI to a different unusual value, like 1320 dpi.
6. Reboot back to Slackware, check the value is 1320 dpi.
   * Re-enable persistence and that script from step 1.

This is to check that Razer Synapse isn't applying some trickery and that e.g. 1320 dpi in Windows is the same number reported in OpenRazer (which asks the hardware directly). Let us know the results.

0pLuS0, 2024-05-26:
I have been playing in the past with the Poll Rates. Windows and Slack are now both 1000 Hz.

1. I changed in ~/.config/openrazer/razer.conf ```restore_persistence = False```

2. I moved ```persistence.conf``` out of ~/.config/openrazer

3. I have ~/.config/autostart/razer_startup.py so I moved ```razer_startup.py``` out of this directory.

4. I set the DPI in polychromatic to 1273, and the Poll is 1000.

5.  I use only 2 values for the DPI in Synapse 3, and they can only be changed in increments of Hundreds, so I made them both 1400.

6. I rebooted back into Slack and the DPI still remained 1273 and the Poll 1000. Settings from Windows aren't being applied in Slack.

Here's a screen shot of the settings in polychromatic.

[image]

```IMPORTANT NOTE```

Something I should of also tested, but since we are now changing the DPI to different speeds, I now see that all DPI speeds I set in Slackware are 2-3 times faster than in Windows.

The 1273 DPI I just tested, is like 1800-2000 DPI in Windows.

All I can tell is, I've used a lot of mice over the years from Logitech, HyperX, Razer, and DPI speeds were all consistently the same. I've never used a mouse where DPI speeds varied, and if there was any variation, it was not something you could see. 400, 800, 1000 across all the various mice I've ever used were the same speeds.

So, the problem isn't just that 400 DPI is faster, all the speeds are, and 2-3 times faster is a huge difference. I'm not sure what is cause such an extreme amount of inaccuracy here in Slackware.

THANKS

lah7, 2024-05-27:
I don't think OpenRazer is doing anything wrong. The driver is talking to the firmware fine - DPI set, DPI read OK - and it remembers across reboots. Firmware behaviour is a mystery. No doubt Razer Synapse ignores the hardware and sets its own thing anyway.

In the end, DPI is just a hardware setting of how many dots per inch is tracked by the sensor. Software/OS is responsible for the rest -- mouse acceleration and how it moves (flat vs adaptive). Poll rate impacts how often it _checks in_ with the CPU, which makes a difference too.

[image]

I personally think the discrepancy comes from software acceleration. As you're using OpenBox under X11, I'd assume there might not be any, so it relies more on the hardware, or it's flat movement. Windows has acceleration (by default, "Enhance cursor precision"), so the same hand movement results in a faster cursor.

In X11, there might be something in the logs:

    cat /var/log/Xorg.0.log | grep Baslisk | grep accel

For me:
```
[     7.756] (II) libinput: Logitech G203 LIGHTSYNC Gaming Mouse: Step value 0 was provided, libinput Fallback acceleration function is used.
[     7.757] (**) Logitech G203 LIGHTSYNC Gaming Mouse: (accel) selected scheme none/0
[     7.757] (**) Logitech G203 LIGHTSYNC Gaming Mouse: (accel) acceleration factor: 2.000
[     7.757] (**) Logitech G203 LIGHTSYNC Gaming Mouse: (accel) acceleration threshold: 4
```

I use a Logitech mouse now. Even on Windows, as soon as G Hub starts up, it applies some settings and the cursor feels _marginally_ faster, even though it's _supposed_ to be the same DPI value.

If you don't _need_ Razer Synapse, you _could_ uninstall that so both OSes rely _entirely_ on the hardware (as set by OpenRazer).

Another thing to experiment - if possible - is to try Wayland instead of X11, and/or another desktop environment with mouse acceleration settings (KDE, XFCE, MATE). I find there is a stark difference under Wayland vs X11 - even in KDE the mouse settings interfaces are slightly different.

0pLuS0, 2024-05-27:
I don't use ```Enhance pointer precision``` in Windows. Even with it on or off the DPI is still faster in Slack.

Here's  ```pts/1 root ~ >cat /var/log/Xorg.0.log | grep accel```

```[    25.531] (**) NVIDIA(0): Enabling 2D acceleration
[    27.007] (**) Razer Razer Basilisk Essential: (accel) keeping acceleration scheme 1
[    27.007] (**) Razer Razer Basilisk Essential: (accel) acceleration profile 0
[    27.007] (**) Razer Razer Basilisk Essential: (accel) acceleration factor: 2.000
[    27.007] (**) Razer Razer Basilisk Essential: (accel) acceleration threshold: 4
[    27.008] (**) Razer Razer Basilisk Essential: (accel) keeping acceleration scheme 1
[    27.008] (**) Razer Razer Basilisk Essential: (accel) acceleration profile 0
[    27.008] (**) Razer Razer Basilisk Essential: (accel) acceleration factor: 2.000
[    27.008] (**) Razer Razer Basilisk Essential: (accel) acceleration threshold: 4
[    27.011] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) keeping acceleration scheme 1
[    27.011] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration profile 0
[    27.011] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration factor: 2.000
[    27.011] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration threshold: 4
[    27.063] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) keeping acceleration scheme 1
[    27.063] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration profile 0
[    27.063] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration factor: 2.000```

Looking back now, I realized this issue isn't specific to only this mouse for me in Slack. When I originally started using openrazer I had a Baslisk v2, when that died I replaced it with another, and after that one, I ended up buying the v3. But I had problems with several Baslisk v3 and actually had 3 of them I went through, before I bought this Essential.

So that makes two Baslisk v2 and three Baslisk v3, and all of them had the same issue of faster DPI.

I won't be able to try another Distro under Wayland...

Something I never mentioned, I compile my own kernels, is there anything if I'm missing from the kernel that could cause this?

I actually still have a few of the Baslisk v2 and I just plugged one in and tested it, same 400 DPI was like 1000 DPI.

Here's a few screen shots of the Baslisk v2 from polychromatic.

THANKS

![DeviceBaslisk](https://github.com/openrazer/openrazer/assets/36190223/663bc176-ada4-4fda-a1f1-e2a24c40e6d9)

![Baslisk_DeviceInfo](https://github.com/openrazer/openrazer/assets/36190223/7cd0fcc1-01ed-4aed-8737-7c2845331842)

0pLuS0, 2024-05-27:
OH boy, I never ever stopped to consider Mouse Acceleration in Slack. I've been digging around and I created ```50-mouse-acceleration.conf``` with this contents;
```
Section "InputClass"
    Identifier "My Mouse"
    MatchIsPointer "yes"
    Option "AccelerationProfile" "-1"
    Option "AccelerationScheme" "none"
    Option "AccelSpeed" "-1"
EndSection
```
I logged out of X and back in, and the mouse is twice as slow, so I'm assuming this has disabled the mouse acceleration.

The output of ```cat /var/log/Xorg.0.log | grep accel``` now shows these two lines I hightlighted.

[    25.676] (**) NVIDIA(0): Enabling 2D acceleration
```[    27.145] (**) Razer Razer Basilisk Essential: (accel) selected scheme none/0```
[    27.145] (**) Razer Razer Basilisk Essential: (accel) acceleration factor: 2.000
[    27.145] (**) Razer Razer Basilisk Essential: (accel) acceleration threshold: 4
[    27.146] (**) Razer Razer Basilisk Essential: (accel) keeping acceleration scheme 1
[    27.146] (**) Razer Razer Basilisk Essential: (accel) acceleration profile 0
[    27.146] (**) Razer Razer Basilisk Essential: (accel) acceleration factor: 2.000
[    27.146] (**) Razer Razer Basilisk Essential: (accel) acceleration threshold: 4
[    27.149] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) keeping acceleration scheme 1
[    27.149] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration profile 0
[    27.149] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration factor: 2.000
[    27.149] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration threshold: 4
```[    27.201] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) selected scheme none/0```
[    27.201] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration factor: 2.000
[    27.201] (**) Razer Razer Huntsman V2 Tenkeyless: (accel) acceleration threshold: 4

There is one problem with the settings in ```50-mouse-acceleration.conf``` if I set the Poll Rate to 1000 Hz, when I log out of X and back in, the Poll Rate changes to 500 Hz. I don't know if there's something I can change ```50-mouse-acceleration.conf``` , or I have to set this in my razer startup script ```razer_startup.py````
```
#!/usr/bin/python3
import time
import openrazer.client
devman = openrazer.client.DeviceManager()

while len(devman.devices) < 2:
    time.sleep(0.5)
    devman = openrazer.client.DeviceManager()

for device in devman.devices:
    if "Basilisk" in device.name:
        device.dpi = (800,800)
        device.fx.misc.logo.none()

    if "Huntsman" in device.name:
        device.fx.static(0, 255, 0)
```

THANKS

lah7, 2024-05-28:
> I logged out of X and back in, and the mouse is twice as slow, so I'm assuming this has disabled the mouse acceleration.

In theory, since both Windows and Xorg have mouse acceleration disabled ("flat"), is your preferred DPI setting _more_ similiar now between the two OSes (assuming the same polling rate)?

There is also the **pointer speed** to consider, in which I have no idea how that can be configured under raw Xorg.

Your start-up script can ensure the poll rate is set like so:

```diff
    if "Basilisk" in device.name:
        device.dpi = (800,800)
+       device.poll_rate = 1000
        device.fx.misc.logo.none()
```

0pLuS0, 2024-05-28:
I thought with the first output I pasted of the ```pts/1 root ~ >cat /var/log/Xorg.0.log | grep accel``` this showed that Mouse Accleration was enabled?

libinput list-devices showed;  ```Accel profiles:   flat *adaptive```

If the mouse accleration in Slack is disabled, then why is the mouse cursor moving 2-3 times faster? I assumed that was because the acceleration was enabled.

THANKS

lah7, 2024-05-28:
I've confused DPI numbers and which OS has the "faster" cursor, sorry. This thread has become exhausting for me.

Your system is a unique setup (raw Xorg + OpenBox). You're right, a start-up script _shouldn't_ be needed. For persistence to work properly, you need  something _(like a systemd user service)_ to auto start `openrazer-daemon` and for the daemon to gracefully quit on exit. Your script uses the Python library which summons the daemon (via D-Bus) if it wasn't running.

I recall in https://github.com/openrazer/openrazer/issues/2066#issuecomment-1694507194 that your keyboard took a while to initialize under Slackware, hence the script. Since it looks by name, not serial, it's guaranteed to work. Some (newer) devices don't return their serial number correctly (#2122), so persistence would fail to do its job - I'm not sure if your hardware has the same randomness issue.

For why the mouse goes back to 500 Hz, it _could_ be Razer firmware being quirky when it switches [device mode.](https://github.com/openrazer/openrazer/wiki/Using-the-mouse-driver#device_mode---set-device-mode)  I can't see how/why X11 would mess with a Razer hardware setting, but starting X11 presumably causes `openrazer-daemon` to start. Generally, the newer the Razer hardware, the more likely it _needs_ software to tell it again what to do.  You could see if the act of starting/quitting `openrazer-daemon` causes this, or by replugging the device and checking Polychromatic before/after.

0pLuS0, 2024-05-28:
I'm sorry I didn't think about Mouse Acceleration, it would of certainly created less posting here. :(

Slackware has the faster cursor speeds...

What's nice about Slackware, and how most people use it, many times, it's simply based around how the developers created their source/apps. Slackware doesn't add anything to source.

So however open-razer was created and with whatever scripts it has, this is what's being used. And from what I can see there is in openrazer-daemon-3.8.0; ```/usr/lib/udev/rules.d/99-razer.rules```

And then I added for  /etc/rc.d/rc.modules.local ```/sbin/modprobe razerkbd```

By the way, looking over ```lsmod``` I see ```razermouse``` is loaded. I didn't think I needed to do ```/sbin/modprobe  razermouse``` at bootup to load it. Or is it better to define ```/sbin/modprobe  razermouse``` in ```/etc/rc.d/rc.modules.local```?

Here's ```/usr/lib/udev/rules.d/99-razer.rules```, I never made any changes to this.

[99-razer.rules.txt](https://github.com/openrazer/openrazer/files/15476663/99-razer.rules.txt)

I didn't see any mention of that for the Build Script you can see here; https://slackbuilds.org/repository/15.0/system/openrazer-daemon/

THANKS a lot for your time @lah7 and sorry if this turned into a headache for you. I greatly appreciate all you do here for Linux, openrazer is amazing and you help make Linux better!

Be well! 👍

closed 2024-10-27 by github-actions[bot]