openrazer/openrazer #2134: Support for Viper V3 Hyperspeed

marietta, 2023-09-22:
### Device Info

* Name: Razer Viper V3 HyperSpeed
* Product Number: RZ01-0491
* Model Number: 

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 013: ID 1532:00b8 Razer USA, Ltd Razer Viper V3 HyperSpeed
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00b8 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Viper V3 HyperSpeed
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          4 Default configuration
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     314
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

```

```
Please post the output of
    ls -lah /dev/input/by-id/

lrwxrwxrwx 1 root root   9 Sep 22 13:03 usb-Razer_Razer_Viper_V3_HyperSpeed_000000000000-event-if01 -> ../event9
lrwxrwxrwx 1 root root   9 Sep 22 13:03 usb-Razer_Razer_Viper_V3_HyperSpeed_000000000000-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Sep 22 13:03 usb-Razer_Razer_Viper_V3_HyperSpeed_000000000000-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Sep 22 13:03 usb-Razer_Razer_Viper_V3_HyperSpeed_000000000000-if01-event-mouse -> ../event8
lrwxrwxrwx 1 root root   9 Sep 22 13:03 usb-Razer_Razer_Viper_V3_HyperSpeed_000000000000-if01-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Sep 22 13:03 usb-Razer_Razer_Viper_V3_HyperSpeed_000000000000-if02-event-kbd -> ../event10
lrwxrwxrwx 1 root root   9 Sep 22 13:03 usb-Razer_Razer_Viper_V3_HyperSpeed_000000000000-mouse -> ../mouse0
```

```
Please post the output of
    usbhid-dump -m 1532 -ed

001:013:002:DESCRIPTOR         1695382596.792334
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

001:013:001:DESCRIPTOR         1695382596.793501
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 75 01 95 08 81 02 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 0E 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0 05 01 09
 02 A1 01 85 0A 09 01 A1 00 05 09 19 01 29 06 15
 00 25 01 75 01 95 06 81 02 75 01 95 02 81 03 06
 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05 01
 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09 31
 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0 05 0C
 09 01 A1 01 85 02 19 00 2A 3C 02 15 00 26 3C 02
 95 01 75 10 81 00 75 08 95 0D 81 01 C0 05 01 09
 80 A1 01 85 03 19 81 29 83 15 00 25 01 75 01 95
 03 81 02 95 05 81 01 75 08 95 0E 81 01 C0 05 01
 09 00 A1 01 85 04 09 03 15 00 26 FF 00 35 00 46
 FF 00 75 08 95 0F 81 00 C0 05 01 09 00 A1 01 85
 05 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 08 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0 05
 01 09 00 A1 01 85 09 09 03 15 00 26 FF 00 35 00
 46 FF 00 75 08 95 0F 81 00 C0

001:013:000:DESCRIPTOR         1695382596.794837
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 0B 81 03
 05 0C 0A 38 02 15 81 25 7F 75 08 95 01 81 06 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

001:004:000:DESCRIPTOR         1695382596.799991
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 07
 15 00 25 01 95 07 75 01 81 02 95 01 75 01 81 03
 06 00 FF 09 40 95 02 75 08 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 C0
```

z3ntu, 2023-09-22:
Please take a look at https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md , if you like tinkering I'm sure you can get this device supported yourself!

z3ntu, 2023-10-24:
A pull request for this device has been opened: #2149

See https://github.com/openrazer/openrazer/wiki/Building-a-package how to build from source.

commit 14b487e5 (2023-10-28): Add support for Razer Viper V3 HyperSpeed

Fixes #2134
PR #2149 (2023-10-28): Add support for Razer Viper V3 HyperSpeed
Fixes #2134

All basic functions are implemented, but synapse supports specifying the specific battery type to have a more accurate battery percentage. Other mice that run on AA batteries might also support that feature, but the mouse stores that option for those who care. I tested DPI, polling rate, charge and they all work. I didn't test DPI stages, but it seems identical to other mice so it should also work.
closed 2023-10-28 by z3ntu