openrazer/openrazer #2135: Support for [Razer Unify Dongle]

NathanielH-snek, 2023-09-22:
### Device Info

* Name: <See under device>
* Product Number: <See under device>
* Model Number: <See under device>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
```
Bus 001 Device 006: ID 1532:00fd Razer USA, Ltd Razer Unify Dongle
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x00fd 
  bcdDevice            1.00
  iManufacturer           1 Razer
  iProduct                2 Razer Unify Dongle
  iSerial                 3 000000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      97
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               1
[... 247 lines cut ...]
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:008:000:DESCRIPTOR         1695425227.436229
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0
```
</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2023-09-23:
Can you give some details what this thing is? What can you do with it / what is it used for?

NathanielH-snek, 2023-09-23:
> Can you give some details what this thing is? What can you do with it / what is it used for?

Yeah. I got two of these as an RMA for my viper v2 pro dongle failing. I can't seem to find these on the razer website anywhere so I'm not sure exactly what they are. They are supposed to be dongles that I believe can pair to any of the wireless v2/v3 mice and perhaps some of the keyboards as well. I'm just trying to get it to work with my v2 pro, but I think I need windows to get dongle pairing to work properly. Either way this dongle shows up as an unknown whereas my v2 pro dongle used to actually show up properly.

z3ntu, 2023-09-24:
I'm aware of some pairing functionality on other dongles (e.g. hyperpolling dongle thing). Would be great if you could capture USB packet capture on Windows when you're doing things in Razer Synapse there.

NathanielH-snek, 2023-09-24:
Once I can get my hands on a windows machine I'll update this thread. I have an old MacBook Air I got from someone I can mess with, turns out boot camp is a pita tho. I already tried getting the pairing software to work in a vm but I couldn't get it to see the dongle.

Merisal, 2023-11-08:
Hi, I have the same problem with my Razer Deathstalker v2 TKL. The keyboard itself is recognized if I use BT or the wired connection. The USB dongle with the ID 1532:00FD is part of the scope of delivery for the keyboard.
The keyboard has 100% functionality so the 2.4GHz connection is working via the dongle