openrazer/openrazer #2139: Support for PiD 1532.028D with Polychromatic Blackwidow v4 Pro with green clicky keys

CoryTakigawa, 2023-09-30:
### Device Info

* Name: <Blackwidow v4 Proe>
* Product Number: <RZ03-04680200-R3U1>
* Model Number: <RZ03-0468>

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 002: ID 1532:028d Razer USA, Ltd Razer BlackWidow V4 Pro
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x028d 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow V4 Pro
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0086
    bNumInterfaces          5
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
[... 186 lines cut ...]
 10 81 00 75 08 95 13 81 01 C0 05 01 09 80 A1 01
 85 03 19 81 29 83 15 00 25 01 75 01 95 03 81 02
 95 05 81 01 75 08 95 14 81 01 C0 05 01 09 00 A1
 01 85 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75
 08 95 15 81 00 C0 05 01 09 00 A1 01 85 05 09 03
 15 00 26 FF 00 35 00 46 FF 00 75 08 95 15 81 00
 C0

003:002:000:DESCRIPTOR         1696111067.236115
 05 01 09 06 A1 01 05 07 19 E0 29 E7 15 00 25 01
 75 01 95 08 81 02 81 01 19 00 2A FF 00 15 00 26
 FF 00 75 08 95 06 81 00 05 08 19 01 29 03 15 00
 25 01 75 01 95 03 91 02 95 05 91 01 C0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

#Sorry I am non-technical and did not understand that fully.

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>
[image]
[image]

CoryTakigawa, 2023-09-30:
Sorry that I could not do better with the screenshots of the features.  I just did not know what is wanted by you and this was my first time joining github and I just learned to use a cli yesterday/today.

z3ntu, 2023-10-01:
Duplicate of #2044

closed 2023-10-01 by z3ntu