openrazer/openrazer #2140: Support for [device name here]

abeguerra, 2023-10-02:
### Device Info

* Name: Razer Ornata V3 TKL
* Product Number: RZ03-04880200-R3U1
* Model Number: RZ03-0488

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 001 Device 012: ID 1532:024b Razer USA, Ltd Gaming Laptop [Blade 15 Advanced (Late 2019)]
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x024b Gaming Laptop [Blade 15 Advanced (Late 2019)]
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0054
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 398 lines cut ...]
 02 19 00 2A 3C 02 15 00 26 3C 02 95 01 75 10 81
 00 75 08 95 0D 81 01 C0 05 01 09 80 A1 01 85 03
 19 81 29 83 15 00 25 01 75 01 95 03 81 02 95 05
 81 01 75 08 95 0E 81 01 C0 05 01 09 00 A1 01 85
 04 09 03 15 00 26 FF 00 35 00 46 FF 00 75 08 95
 0F 81 00 C0 05 01 09 00 A1 01 85 05 09 03 15 00
 26 FF 00 35 00 46 FF 00 75 08 95 0F 81 00 C0

001:009:000:DESCRIPTOR         1696209784.258305
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 05
 15 00 25 01 75 01 95 05 81 02 75 01 95 03 81 03
 06 00 FF 09 40 75 08 95 02 15 81 25 7F 81 02 05
 01 09 38 15 81 25 7F 75 08 95 01 81 06 09 30 09
 31 16 00 80 26 FF 7F 75 10 95 02 81 06 C0 06 00
 FF 09 02 15 00 25 01 75 08 95 5A B1 01 C0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>
stepping through this

### Synapse Screenshots
I'm on ubuntu. I'm not sure how to get these screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2023-10-14:
Doing anything just on the keyboard won't produce any useful USB packets. Really just Synapse on Windows will provide what's interesting. But if you can you can also directly try adding support for your device, following https://github.com/openrazer/openrazer/blob/master/DEVELOPMENT.md

AndersKlint, 2023-12-18:
I've working on supporting this keyboard, but have some problems. Synapse works fine for changing brightness and for the following effects:  wave, spectrum, static, reactive and breath. However, custom effects doesn't work (as well as starlight and ripple - I assume they're not built into the firmware).

Checking dmesg there's errors like " Response doesn't match request" etc (see logs inside attached zip). Report params are set to this:
[image]

All my changes can be seen in the latest commit of his fork:
https://github.com/AndersKlint/openrazer-add-ornata-v3-tkl

Captured packages, screenshots of Synapse and logs can be seen in attached zip.
[ornata-v3-pcap-and-logs.zip](https://github.com/openrazer/openrazer/files/13705154/ornata-v3-pcap-and-logs.zip)

I've built the deb packages and installed them using the instruction in the wiki. Would be happy with some help getting the custom led effects working :)

AndersKlint, 2023-12-18:
I figured out the issue - I had the wrong matrix dims. I though they were equivalent to the amount of keys on the keyboard, but in fact they were only 1x8

alichtman, 2024-01-21:
I tried to test this PR, but ran into trouble building the drivers on Ubuntu 23.10: https://github.com/openrazer/openrazer/issues/2199

PR #2260 (2024-07-02): Add support for Ornata V3 TKL
Fixes #2140 

I added support for the media keys on top of the code from https://github.com/openrazer/openrazer/pull/2183.

@AndersKlint's commit from that PR was rebased onto master and @z3ntu's suggestion from here was used: https://github.com/openrazer/openrazer/pull/2183#issuecomment-2038014662

cc @bakankaya for a stamp

## Test Plan

```
openrazer on  ornata-chroma-v3-tkl-support [!] via 💎 v3.2.2 took 3s
07:58:55 PM ➜ make driver
-e
:: Compiling OpenRazer kernel modules
========================================
make -C /lib/modules/6.5.0-35-generic/build M=/ [... cut]
closed 2024-07-02 by z3ntu