openrazer/openrazer #2143: Smooth scroll, precise scroll (REL_WHEEL_HI_RES) support

ad-on-is, 2023-10-13:
I'm not sure how to explain this one, but I'll try me best to do so.

I have a Logitech G903 and G502 x Plus, and solaar and a Razer Basilisk v3 pro and openrazer, all three of them have a free-spinning wheel, which I use a lot in productivity mode.

Very simplified, when I perform a free-spinning scroll action on all these mice, the scrolling event only gets triggered when I scroll the wheel 50 degrees. When I turn on "Scroll wheel resolution" in solaar, the Logitech mice trigger that event after 5 degrees of scrolling which gives a much smoother and precise experience. 

I hope the explanation makes sense and you know what I mean.

As far as I understood, this has to do with REL_WHEEL_HI_RES and something about the value 120, etc.

Now, my question: Is there a similar setting in openrazer for their mice.

z3ntu, 2023-10-14:
With openrazer we don't have anything like that and since I also don't have any of these mice, it's difficult to investigate it. Do you know if something like that is possible on Windows for example? As in does the hardware support this?

ad-on-is, 2023-10-17:
Unfortunately, I don't have Windows on bare-metal, and getting the razer mouse recognized by synapse in a windows VM is just a PITA, therefore I cannot provide any reliable tests.

But I too was wondering whether the hardware supports it. Is there any way where I could log the raw HID events, bypassing anything else that grabs the mouse?

z3ntu, 2023-10-18:
I doubt the mouse would send anything by default. And anyways all the input is actually handled by the main kernel (hid-generic), not our kernel modules. So there's nothing special really.

Perhaps somehow you could switch the mouse to provide different reports that could be used for hi-res scrolling

ad-on-is, 2023-10-18:
you mean by trying to make it work with synapse and see whether there's some setting to activate hi-res scrolling AND save it to the onboard-memory? I can give this a try, sure.

z3ntu, 2023-10-18:
I have no clue really tbh

ad-on-is, 2023-10-18:
Ok, all good. Thx anyway.
I'll just do some research and see if I can find something out.