openrazer/openrazer #2150: Support for [device name here]

hangint3n, 2023-10-25:
### Device Info

* Name: Razer Wolverine V2
* Product Number: RZ06-0360100
* Model Number: RZ06-0356

### Device Information
```
Please post the output of
    lsusb -d 1532: -v

Bus 001 Device 004: ID 1532:0a29 Razer USA, Ltd Razer Wolverine V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass        71 
  bDeviceProtocol       208 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0a29 
  bcdDevice            1.01
  iManufacturer           1 Razer
  iProduct                2 Razer Wolverine V2
  iSerial                 3 00006A5C7FAC0E96
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0060
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               4
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x03  EP 3 OUT
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            1
          Transfer Type            Isochronous
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x00e4  1x 228 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       1
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass     71 
      bInterfaceProtocol    208 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
Device Status:     0x0002
  (Bus Powered)
  Remote Wakeup Enabled

```

```
Please post the output of
    ls -lah /dev/input/by-id/

[19:23:39 hangint3n]$  ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 200 Oct 24 10:31 ./
drwxr-xr-x 4 root root 720 Oct 24 10:31 ../
lrwxrwxrwx 1 root root  10 Oct 24 06:31 usb-Logitech_USB_Receiver-if02-event-mouse -> ../event23
lrwxrwxrwx 1 root root   9 Oct 24 06:31 usb-Logitech_USB_Receiver-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root  10 Oct 24 06:31 usb-SQT_USB_Gaming_Keyboard-event-kbd -> ../event20
lrwxrwxrwx 1 root root  10 Oct 24 06:31 usb-SQT_USB_Gaming_Keyboard-if01-event-kbd -> ../event21
lrwxrwxrwx 1 root root  10 Oct 24 06:31 usb-SQT_USB_Gaming_Keyboard-if01-event-mouse -> ../event22
lrwxrwxrwx 1 root root   9 Oct 24 06:31 usb-SQT_USB_Gaming_Keyboard-if01-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Oct 24 10:31 usb-WACOM_CTE-440-U_V4.0-3-event-mouse -> ../event24
lrwxrwxrwx 1 root root   6 Oct 24 10:31 usb-WACOM_CTE-440-U_V4.0-3-mouse -> ../js0
```

<details>

```
Please post the output of
    usbhid-dump -m 1532 -ed

[19:24:48 hangint3n]$ usbhid-dump -m 1532 -ed
No matching HID interfaces
```

</details>

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

NO Packet capture, no RGB/LEDs.  Wireshark doesn't see device.

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

z3ntu, 2023-10-26:
What support are you looking for for this device? What's there to configure from OpenRazer side if there's no RGB, no mouse functions like DPI, poling rate etc?

hangint3n, 2023-10-26:
I was just told about openrazer so I'm not sure what you support you can
provide. Bottom line I just want the get the device to work. Lsusb sees the
device, but I can't get it to work.

I would have thought doing the trace and capture with wireshark as
suggested on the openrazer wiki would have shown the various buttons being
engaged, but it doesn't even see the device.

Is there anything you can do to get this devices working or is it a brick?

z3ntu, 2023-10-26:
Unfortunately OpenRazer support doesn't mean that any core functionality will work. On mice we really only handle RGB, DPI, etc. On keyboard some extra macro buttons but the keyboard itself already works beforehand, and our driver continues to let the core kernel handle input etc.

So for this controller, there's nothing in scope of OpenRazer that would make this one work.

One thing I see in the core kernel [searching for "Razer Wolverine"](https://lore.kernel.org/all/?q=razer+wolverine) is this patch: https://lore.kernel.org/all/20211113131734.46fbc9bf@rechenknecht2k11/ which actually already has your specific device - but this patch was never applied. It might be worth trying to build your own kernel with this patch applied - at least the one line specific to your device and see if that does anything.

hangint3n, 2023-10-29:
So the patch mentioned failed, but it had the info I needed so I took that and put it in the xbox.c file and complied the kernel. Once I turned off the controller feature in Steam I got the controller to work. Yeah!

z3ntu, 2023-10-29:
So you can confirm that one line patch to Linux fixes the controller for you? I can send that patch to Linux upstream then so it'll be available in upcoming versions of Linux then.

hangint3n, 2023-10-30:
I can confirm that. However I don't know how to build a patch or how to
submit it.

z3ntu, 2023-10-30:
I can submit the patch for you, no problem. I'll let you know once it's done.

z3ntu, 2023-11-25:
Sent the patch now: https://lore.kernel.org/lkml/20231125-razer-wolverine-v2-v1-1-979fe9f9288e@z3ntu.xyz/

Hopefully it gets into v6.8 kernel

hangint3n, 2023-11-25:
Thanks for doing this as it is not something I know how to do.... Yet.

z3ntu, 2023-12-13:
Patch is applied: https://git.kernel.org/pub/scm/linux/kernel/git/dtor/input.git/commit/?h=for-linus&id=c3d1610345b79cbe29ef6ca04a4780eff0d360c7

Still on track to be included in v6.8 kernel

hangint3n, 2023-12-13:
Awesome, thanks for the update!

z3ntu, 2023-12-13:
I think I'll close this issue then, if I hear anything new about the patch I'll update here but as I wrote it should probably just be integrated into v6.8 kernel. Once a release candidate or the final version is out, would be nice if you could test that explicitly and see if your controller works as expected :) But otherwise I don't think there's anything more to do!

closed 2023-12-13 by z3ntu

z3ntu, 2024-01-23:
The patch is being backported to 4.19, 5.4, 5.10, 5.15, 6.1 and 6.6 kernels so the stable kernels released in like a week or two should also contain the patch.