openrazer/openrazer #2153: Fedora 39 installation

evgnor86, 2023-11-03:
Fedora 39
Instruction for install:

On installation get error: https://openrazer.github.io/#download

```
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM: ошибка: Verifying a signature using certificate E96C496E4E77C159C2FCA053062F9FD4D6D11CE4 (hardware OBS Project <hardware@build.opensuse.org>):
RPM:   1. Certificiate 062F9FD4D6D11CE4 invalid: certificate is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
RPM:   2. Key 062F9FD4D6D11CE4 invalid: key is not alive
RPM:       because: The primary key is not live
RPM:       because: Expired on 2023-05-23T13:32:57Z
Ошибка: Ошибка при проверке транзакции:
package openrazer-kernel-modules-dkms-3.7.0-1.1.noarch does not verify: Заголовок V3 RSA/SHA256 Signature, key ID d6d11ce4: NOTTRUSTED
package openrazer-daemon-3.7.0-1.1.noarch does not verify: Заголовок V3 RSA/SHA256 Signature, key ID d6d11ce4: NOTTRUSTED
package python3-openrazer-3.7.0-1.1.noarch does not verify: Заголовок V3 RSA/SHA256 Signature, key ID d6d11ce4: NOTTRUSTED
package openrazer-meta-3.7.0-1.1.noarch does not verify: Заголовок V3 RSA/SHA256 Signature, key ID d6d11ce4: NOTTRUSTED
```

evgnor86, 2023-11-03:
Certificaes is out of date

z3ntu, 2023-11-03:
Please see #2061 if you can find some workaround there. The key on OBS should be valid until 2025, no clue how to make the system pick that up.

lah7, 2023-11-03:
I've recently seen this at https://github.com/polychromatic/polychromatic/issues/479#issuecomment-1755946943 for the same repo, but different package.

This public gist contains instructions to fix the problem: (delete the old key, then update again)
https://gist.github.com/e7d/3b786c7410ca14a5ded61eec36de9874#the-solution

closed 2024-08-01 by github-actions[bot]