openrazer/openrazer #2155: Newest update v.3.7.0 not covering all (new) devices?

ICDarki, 2023-11-04:
A week ago I installed the development build (master branch, kind of v3.6.1 dev) and those listed devices below worked flawlessly, as those new released Razer devices were not covered in the list on v3.6.1 stable.
I had to reinstall my OS (custom Arch Linux) and installed once again the dev master branch (it was updated to v3.7.0 few days ago).
In update v3.7.0 (master build) openrazer is not recognizing those devices. So I tried v3.7.0 stable (as those devices are officialy listed now) and the problem still persist.

My devices:

Razer Basilisk V3 Pro (Wired)             1532:00AA
Razer Basilisk V3 Pro (Wireless)	1532:00AB
Razer BlackWidow V4 Pro                  1532:028D

I'm not sure where the problem is, update side or my side.

lah7, 2023-11-04:
Considering that the OS was reinstalled, have you added your user to the `plugdev` group again? Seems likely to be that if none of the devices show.

There is also the troubleshooting guide in case it's a driver issue (like if the driver is installed for the wrong kernel)
https://github.com/openrazer/openrazer/wiki/Troubleshooting

ICDarki, 2023-11-05:
@lah7 Thank you!

Such a simple command. I thought I did .. but nope, I missed. It's working now. ✅

closed 2023-11-05 by ICDarki