openrazer/openrazer #2157: Support for Kraken Kitty v2

bradfordtfm, 2023-11-06:
### Device Info

* Name: Kraken Kitty V2
* Product Number: RZ04-04730100-R3U1
* Model Number: RZ04-0473

### Device Information
```
Please post the output of
    lsusb -d 1532: -v
```
Bus 003 Device 002: ID 1532:0560 Razer USA, Ltd Razer Kraken Kitty V2
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0560 
  bcdDevice           10.06
  iManufacturer           1 Razer
  iProduct                2 Razer Kraken Kitty V2
  iSerial                 3 00000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0123
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           0
      bInterfaceClass         1 Audio
      bInterfaceSubClass      1 Control Device
      bInterfaceProtocol      0 
      iInterface              0 
      AudioControl Interface Descriptor:
        bLength                10
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdADC               1.00
        wTotalLength       0x0072
        bInCollection           2
        baInterfaceNr(0)        1
        baInterfaceNr(1)        2
      AudioControl Interface Descriptor:
        bLength                12
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID            10
        wTerminalType      0x0201 Microphone
        bAssocTerminal          0
        bNrChannels             2
        wChannelConfig     0x0003
          Left Front (L)
          Right Front (R)
        iChannelNames           0 
        iTerminal               6 Sidetone
      AudioControl Interface Descriptor:
[... 283 lines cut ...]
003:002:003:DESCRIPTOR         1699275784.741288
 05 0C 09 01 A1 01 85 01 15 00 25 01 75 01 09 E9
 09 EA 95 02 81 06 95 06 81 01 85 02 05 0C 09 00
 95 10 81 02 85 04 09 00 75 08 95 24 91 02 85 05
 09 00 95 20 81 02 85 06 09 00 95 24 91 02 85 30
 09 00 95 0F 81 02 85 31 09 00 95 10 91 02 85 32
 09 00 95 01 91 02 85 40 09 00 75 08 95 08 91 02
 85 41 09 00 75 08 95 08 81 02 85 50 09 00 75 08
 95 08 91 02 85 07 09 00 95 20 81 02 C0 05 0B 09
 05 A1 01 85 08 09 00 95 01 81 02 85 0A 15 00 25
 01 09 2F 75 01 95 01 81 06 09 20 09 21 75 01 95
 02 81 22 95 05 81 01 05 08 85 09 09 09 75 01 95
 01 91 22 95 07 91 01 85 17 09 17 75 01 95 01 91
 22 95 07 91 01 85 21 09 21 75 01 95 01 91 22 95
 07 91 01 85 1A 09 18 75 01 95 01 91 22 95 07 91
 01 C0 06 A0 FF 09 01 A1 01 85 23 09 AA 75 08 95
 28 B1 02 C0

### Packet Captures

<Please see here as we probably need packet captures>
<https://github.com/openrazer/openrazer/wiki/Reverse-Engineering-USB-Protocol>

### Synapse Screenshots
<Screenshots of all available options as seen on **Razer Synapse** in a Windows Environment>

ColinTimBarndt, 2023-12-07:
Here's the `.pcapng` download: https://colintimbarndt.de/openrazer/Kitty-V2/change_color.pcapng

In this capture, I changed the color from red (`#ff0000`) to `#c0ffee`. The Razer software blends the colors, as can be seen when filtering the packets as previously mentioned:

[image]

ColinTimBarndt, 2023-12-07:
After rebooting to Linux, I noticed that my headset is now cycling through hue. I don't know how this was configured on the headset because I set it to solid color mode on Windows before switching. Though I don't think this is important because a driver could just set an initial solid color for consistency.

z3ntu, 2023-12-28:
Should be fairly straightforward to add support, copy paste an existing device (maybe the original Kraken Kitty) and make sure your device uses the following:
* `MATRIX_DIMS = [1, 2]`
* transaction_id `0x1f`
* `razer_chroma_extended_matrix_set_custom_frame`

jlee0111, 2023-12-29:
Willing to ship someone a Kraken Kitty V2 to figure this out, if they'd even need one.

Pinkey552, 2024-05-30:
Is there any update on this?

Denendaden, 2024-07-14:
I tried adding support, but I'm afraid I'm not sure how to actually test it - [Polychromatic](https://polychromatic.app) still says it doesn't recognize the device, but that could be an issue with Polychromatic or with me just not running things correctly. I have it at https://github.com/Denendaden/openrazer/tree/kraken_kitty_v2 if someone with more knowledge than me wants to test it.

z3ntu, 2024-08-31:
@Denendaden If you want some help, feel free to join our community chat: https://github.com/openrazer/openrazer/wiki/Community-chat

Yume303, 2024-11-03:
Compiled the above branch, but as @Denendaden states, it still not works with Polychromatic.
I'll happily help in debugging, if someone instucts me what to do...

z3ntu, 2024-12-07:
@Yume303 Like I wrote above, it's easier if you join the chat, then it's a bit easier to help.

But start by checking if the kernel driver has been bound correctly, so `ls -al /sys/bus/hid/drivers/razeraccessory/0*/` should have something.

Then if that works, check the `openrazer-daemon -Fv` output and see if both the device shows up in the debug log. (stop the daemon first with `systemctl --user stop openrazer-daemon.service`)

z3ntu, 2025-02-15:
There's an open PR which adds support for this device, testing would be appreciated! #2394 

You can use https://github.com/openrazer/openrazer/wiki/Building-a-package to build from source from that branch and install it on your computer.

MaxiTornado, 2025-02-15:
I've built it on Fedora 41 using Polychromatic. It works.

ko-teiko, 2025-02-15:
The same as above. F41, polychromatic, works great.

gmongaras, 2025-02-15:
Built it on Ubuntu 24.04 using Polychromatic and it works. My cat ears now have RGB!

z3ntu, 2025-02-16:
Awesome, thanks for testing!

closed 2025-03-15 by z3ntu