openrazer/openrazer #2162: Can't individually address chroma lights

kronkinatorix, 2023-11-13:
Just bought a razer chroma since my motherboard won't play nice with openrgb and whiel I'm glad I can finally do SOMETHING with the rgb in my case, I've tried all of the gui / cli tools on the main page here and any time I select a pattern with more than one colour it will only use one. 

Additionally, I don't appear to be able to address the fans attached to it individually, and I'd understood that was something openrazer could do. 

Lastly, a quick question: the openraze-daemon appears to only run through python, and no systemd - is this generally how it's set up? I'm seeing some conflicting info on the page and attempting to run it via systemd results in errors. Nothing specifi, just error code exit status=1. Happy to provide logs if helpfu. I'm running Mint 21.

z3ntu, 2023-11-13:
What device is this about? Can you provide `lsusb | grep 1532` ?

kronkinatorix, 2023-11-16:
The Razer Chroma addressable RGB controller. Not confusing at all that it's named the same thing as all their RGB peripherals, n o, not at all.

Chroma Addressable RGB Controller
1532:0F1F 

lsusb | grep 1532 returns nothing. Odd. It's looked up the PSU and the USB slot on my mobo.

z3ntu, 2023-11-16:
> attempting to run it via systemd results in errors. Nothing specifi, just error code exit status=1

Sounds like you should check out https://github.com/openrazer/openrazer/wiki/Troubleshooting

> lsusb | grep 1532 returns nothing

If the device doesn't show up on USB, there's not much this project can do... We can only talk to the USB devices if they are connected and recognized by the kernel.

kronkinatorix, 2023-11-16:
It's showing up on usb now. No idea why but I swapped places with the Chroma / my tower's usb ports and there it is. 

 lsusb | grep 1532

'''
Bus 001 Device 007: ID 1532:0f1f Razer USA, Ltd Razer Chroma Addressable RGB Controller
'''

Still can't enable the openrazer-daemon with systemctl enable --user openrazer-daemon.

'''
Failed to connect to bus: $DBUS_SESSION_BUS_ADDRESS and $XDG_RUNTIME_DIR not defined (consider using --machine=<user>@.host --user to connect to bus of other user)
'''

I've set 
'''
export XDG_RUNTIME_DIR="/run/user/$UID"
export DBUS_SESSION_BUS_ADDRESS="unix:path=${XDG_RUNTIME_DIR}/bus"
'''
in ~/.zshrc which is my shell, and .bashrc too because why not. 

I'm added myself to plugdev with 
'''
sudo usermod -a -G plugdev $USER
'''
And confirmed it's been added. 

Openrazer-genie installed via flatpak doens't detect any dewvices. 

Polychromatic is throwing up an error: the supported devicce wasn't detected by the openrazer daemon. 

So it no longer works at all. Cool., On a fresh install of linux mint 21. btw 6.1.0-1025

kronkinatorix, 2023-11-16:
Had to reinstall the OS anyway so I went ahead and did that. 
Here's another fun little wrinkle: installing openrazer breaks audio for my pc. 
from dmesg after rebooting, after installing openrazer:
'''
[  129.954576] usb 1-1: 5:0: cannot get min/max values for control 2 (id 5)
[  135.074734] usb 1-1: 5:0: cannot get min/max values for control 2 (id 5)
[  145.314552] usb 1-1: 6:0: cannot get min/max values for control 2 (id 6)
[  145.367404] Bluetooth: RFCOMM TTY layer initialized
[  145.367408] Bluetooth: RFCOMM socket layer initialized
[  145.367411] Bluetooth: RFCOMM ver 1.11
[  150.434734] usb 1-1: 6:0: cannot get min/max values for control 2 (id 6)
[  155.554778] usb 1-1: 5:0: cannot get min/max values for control 2 (id 5)
[  160.674625] usb 1-1: 5:0: cannot get min/max values for control 2 (id 5)
[  165.794719] usb 1-1: 6:0: cannot get min/max values for control 2 (id 6)
[  170.914700] usb 1-1: 5:0: cannot get min/max values for control 2 (id 5)
[  181.154521] usb 1-1: 6:0: cannot get min/max values for control 2 (id 6)
[  186.278575] usb 1-1: 6:0: cannot get min/max values for control 2 (id 6)
[  201.634723] usb 1-1: 5:0: cannot get min/max values for control 2 (id 5)
[  206.754804] usb 1-1: 5:0: cannot get min/max values for control 2 (id 5)
[  217.249292] usb 1-1: 2:1: usb_set_interface failed (-110)
[  222.369513] usb 1-1: 2:1: usb_set_interface failed (-110)
[  227.493332] usb 1-1: 2:1: usb_set_interface failed (-110)
[  232.609218] usb 1-1: 2:1: usb_set_interface failed (-110)
[  237.729267] usb 1-1: 2:1: usb_set_interface failed (-110)
[  242.849249] usb 1-1: 2:1: usb_set_interface failed (-110)
[  247.969231] usb 1-1: 2:1: usb_set_interface failed (-110)
'''

z3ntu, 2023-11-16:
No clue why you're setting various environment variables manually.

>  Here's another fun little wrinkle: installing openrazer breaks audio for my pc.

I kinda doubt that but if you insist, please try uninstalling openrazer again and seeing if the problem persists.

kronkinatorix, 2023-11-16:
Another quick update - looks like the lights can be addressed individually, just that polychromatic isn't able to.

And Genie doesn't detect the openrazer daemon. 

So the issue likely lies elsewhere. Unless....?

Anywho, fixed my audio. For the sake of prosperity ()not sure which of these fixed it).
sudo chown -Rc $USER:$USER $HOME
apt install --reinstall alsa-base alsa-utils pulseaudio linux-sound-base libasound2
lsmod | grep snd 

and disabled fastboot in the UEFI. 

@z3ntu tested it three times. It breaks audo. It may have something to do with fastboot / adding an audio device, or may be unrelated.

lah7, 2023-11-16:
> looks like the lights can be addressed individually, just that polychromatic isn't able to.

Please could you elaborate on this? Is the device now appearing in Polychromatic but not RazerGenie?

If you're still on a "not recognised" screen, Polychromatic has a "Troubleshoot" feature, click that and it'll tell you which component isn't working (driver, daemon, etc). Can also do via Tools → OpenRazer → Troubleshoot.

Issues could be caused by Secure Boot - a UEFI feature that only allows signed drivers to load - but OpenRazer is not one of them. If that's enabled, see if it works after turning it off.

closed 2025-02-21 by github-actions[bot]